jgyates/genmon #646: Complex passwords in TankUtilty fails login

oneguynick, 2022-01-13:
### Expected Behavior

Using a complex password with TankUtility shouldn't throw an error. It looks like an issue with how the password is stored.

### Actual Behavior

Using a complex error provides this error:

```
2022-01-13 16:33:05,383 : Error in MyConfig:ReadValue: gentankutil: password: '%' must be followed by '%' or '(', found: '%k352' : myconfig.py:85
2022-01-13 16:33:05,384 : Invalid user name or password, exiting
2022-01-13 16:36:39,306 : Error in MyConfig:ReadValue: gentankutil: password: '%' must be followed by '%' or '(', found: '%k352' : myconfig.py:85
2022-01-13 16:36:39,307 : Invalid user name or password, exiting
2022-01-13 16:37:28,778 : Error in MyConfig:ReadValue: gentankutil: password: '%' must be followed by '%' or '(', found: '%k352' : myconfig.py:85
2022-01-13 16:37:28,778 : Invalid user name or password, exiting
2022-01-13 16:43:05,760 : Error in MyConfig:ReadValue: gentankutil: password: '%' must be followed by '%' or '(', found: '%k352' : myconfig.py:85
2022-01-13 16:43:05,761 : Invalid user name or password, exiting
2022-01-13 16:44:22,488 : Error in MyConfig:ReadValue: gentankutil: password: '%' must be followed by '%' or '(', found: '%k352' : myconfig.py:85
2022-01-13 16:44:22,489 : Invalid user name or password, exiting
```

### Steps to Reproduce (including precondition)

Login with a complex password to TankUtility, use same password in the add-on, watch the log. After switching from a complex password to a simple one without special characters it functions.

### Your Environment

- Generator Model: Generac 22kw
- Genmon Version: Version V1.18.04

jgyates, 2022-01-13:
I checked in a fix. This is due to the configparser library handling of strings. I added an option to make it ignore this for python 3.x or higher. for python 2.x you can just put two '%' instead of one and it should ignore the first one. Just update from the about page. the version number did not change. Let me know if you have any issues.

closed 2022-01-13 by jgyates

oneguynick, 2022-01-14:
Excellent - thank you for such a quick response. The work you have done on this project is phenomenal