jgyates/genmon #647: Determining monitor version of my Generac

xybertech, 2022-01-14:
[image]

My Generac Model: Model 04390-1
15 KW (has electrical upgrade kit) running on Propane
I installed mine in 2006 but I am not sure of the name of the monitor type. I attached an image from my repair and diagnostic manual. It is pretty basic, but has a terminal block with some monitor data. Here it is:
[image]

I could not find a name mentioned, is it one that you have supported?
If not I will probably have to start from scratch, or use your design as a starting point. It should be straight forward to read the status of the lights or defined values.

Your project is quite good and I have thought about doing this for a while, but I am glad to see you have went down that path already.

jgyates, 2022-01-14:
@xybertech, Unfortunately this generator is not supported. This project acquires data from the generator via the serial port and this unit does not appear to have a serial port.

closed 2022-01-14 by jgyates

xybertech, 2022-01-14:
I will probably just tap into the wires, terminal or use a light sensor to read the status of each of the led lights and any appropriate parameter from the terminals and a current/charge sensor for the battery so I have some data. I wont be as nice as yours, but I will have the basics.

**Thank you for taking a look and responding so fast. You did a great job on this project and the documentation is quite good as well.**