jgyates/genmon #649: Python 3 issue after Stretch->Buster upgrade

dlaporte, 2022-01-16:
After upgrading my working RPI Zero running Genmon 18.04 from Stretch to Buster, genmon no longer runs as expected.  I've tracked it back to python3, if I specify "-p 2" it loads without issue.  Below is an example command showing the "Illegal instruction" I receive.  Any idea what the issue might be?

pi@genmon:~/genmon $ ./startgenmon.sh stop
Stopping genmon python scripts
./startgenmon.sh: line 84:  6723 Illegal instruction     sudo $pythoncommand "$genmondir/genloader.py" -x $config_path
pi@genmon:~/genmon $ ./startgenmon.sh -p 2 stop
Using Python 2.x...
Stopping genmon python scripts
Stopping....
Stopping gengpio.py
Stopping genmqtt.py
Stopping genpushover.py
Stopping gensyslog.py
Stopping gentankutil.py
Stopping genserv.py
Stopping genmon.py

### Your Environment

- Generator Model: Generac G0058702
- Genmon Logs: Submit via the About page
- Genmon Version: 18.04

jgyates, 2022-01-16:
Hi @dlaporte 

Just making sure I understand your scenario, are you saying that you upgraded your pi (not reinstalled) from Stretch to Buster? My guess is that you may not have python3 installed or your python3 libraries are in an odd state. You can validate this by typing:

       python3 -V

on the pi console.

Also, has the startgenmon.sh been modified. You may try updating your repository.

      ./genmonmain.sh -u

I have not had great experience upgrading OS versions with linux in general. The one time I tried this on a pi I got very odd results.

dlaporte, 2022-01-16:
pi@genmon:~ $    python3 -V
Python 3.7.3
pi@genmon:~ $ cd genmon/
pi@genmon:~/genmon $ ./genmonmaint.sh -u
This script will update genmon to the latest version from the github repository. This script requires internet access. If you have modified any files in the genmon directory, they will be overwritten. Configuration files in the configuration directory will not be overritten.   Continue? (y/n)  y
Updating genmon...
HEAD is now at 8e4ce03 V1.18.04
Done.
pi@genmon:~/genmon $ ./startgenmon.sh -p 3 stop
Using Python 3.x...
Stopping genmon python scripts
./startgenmon.sh: line 84: 12686 Illegal instruction     sudo $pythoncommand "$genmondir/genloader.py" -x $config_path
pi@genmon:~/genmon $ ./startgenmon.sh -p 2 stop
Using Python 2.x...
Stopping genmon python scripts
Stopping....
Stopping gengpio.py
Stopping genmqtt.py
Stopping genpushover.py
Stopping gensyslog.py
Stopping gentankutil.py
Stopping genserv.py
Stopping genmon.py

jgyates, 2022-01-16:
This looks like a shell problem (i.e. the error is coming from executing the shell script, not the python command. Try typing this form the genmon directory:

       sudo python3 genloader.py -x -c /etc/genmon

This is the command that startgenmon.sh is trying to execute but with environment varibles set from the script command line. For python3 it would execute this:

       sudo python genloader.py -x -c /etc/genmon

If the above python3 command works this means that the error is in the shell script. What is the output of this command:

     echo $SHELL

it should be 

     /bin/bash

dlaporte, 2022-01-16:
Here you go!

pi@genmon:~/genmon $    sudo python3 genloader.py -x -c /etc/genmon

Illegal instruction
pi@genmon:~/genmon $ 
pi@genmon:~/genmon $    sudo python genloader.py -x -c /etc/genmon
Stopping....
pi@genmon:~/genmon $  echo $SHELL
/bin/bash
pi@genmon:~/genmon $ which python
/usr/bin/python
pi@genmon:~/genmon $ /usr/bin/python -V
Python 2.7.16

jgyates, 2022-01-16:
The "illegal instruction" is likely due to a corrupt python3 installation.

Here is some more info:

https://python-forum.io/thread-34489.html

Try typing this:

       python3 -c "print('Hello')"

If this fails then it is your python3 install, if it does not then it is one of the libraries that genmon uses. Trial and error may lead us to the problem. It may be easier to save your genmon settings (the contents of the /etc/genmon folder) and re-install the OS.

dlaporte, 2022-01-16:
That seems to work OK:

***@***.***:~ $ python3 -c "print('Hello')"

Hello

Thinking something similar yesterday, I did reinstall python3 yesterday without error:

sudo apt --reinstall install python3

I also tried to update the libraries with “genmonmaint.sh -u”, but it did seem to skip most/all since it saw they were already installed.

Dave

https://python-forum.io/thread-34489.html

jgyates, 2022-01-16:
the next step would be to execute each of these lines and see if any of them product the error. These are the libraries that genloader.py uses. Keep in mind that some of these are built into python3 and some call other libraries so this may trigger many more rounds of this trial and error to get to the core issue and you may have multiple issues with your python / python library installation.

     python3 -c "import time, sys, os, getopt, subprocess"
     python3 -c" from subprocess import PIPE, Popen, call"
     python3 -c "from shutil import copyfile"
     python3 -c "from shutil import move"
    python3 -c "from genmonlib.mysupport import MySupport"
    python3 -c "from genmonlib.mylog import SetupLogger"
    python3 -c "from genmonlib.myconfig import MyConfig"
    python3 -c "from genmonlib.program_defaults import ProgramDefaults"
    python3 -c "import importlib"

Another thing to try that is some what brute force would be to uninstall all of the libraries, then reinstall them. To uninstall type this:

    sudo pip3 uninstall crcmod  
    sudo pip3 uninstall configparser  
    sudo pip3 uninstall pyserial  
    sudo pip3 uninstall Flask  
    sudo pip3 uninstall pyowm
    sudo pip3 uninstall pytz  
    sudo pip3 uninstall pyopenssl   
    sudo pip3 uninstall twilio  
    sudo pip3 uninstall chump  
    sudo pip3 uninstall paho-mqtt  
    sudo pip3 uninstall pysnmp  
    sudo pip3 uninstall ldap3  
    sudo pip3 uninstall pyasn1  
    sudo pip3 uninstall smbus  
    sudo pip3 uninstall psutil  
    sudo pip3 uninstall pyotp  

Then run the installation script again to reinstall.

dlaporte, 2022-01-17:
Thanks for all the prompt responses!  The python commands ran without issue.  I removed the pip3 modules and ran through the genmon installation script and it fails with the same error.

Dave

jgyates, 2022-01-17:
Since you already reinstalled python3 I can only suggest two things:

1) Use python2 as long as you can, it is not supported or tested any longer by raspbian or this project

or 

2) reinstall raspbian

dlaporte, 2022-01-17:
I’ve come to the same conclusion.  Python 2 will work for now given that the Pi is 200 miles away.  Thanks again for all your help!Dave
    	—David ***@***.***

reinstall raspbian

closed 2022-01-17 by jgyates

dlaporte, 2022-01-18:
I dug in a bit more and found that it was psutil causing the crash.  Once I was able to narrow my search, I came up with this:

https://pythonissues.com/issues/2572149

Sure enough, downgrading psutil from 5.9.0 to 5.8.0 resolved the issue.

jgyates, 2022-01-18:
That is interesting. Now that we know it is psutil I can see why the above tests did not trigger it. genmon uses psutil in a static function so it is not loaded at the top of file, but only when this static function is called:

https://github.com/jgyates/genmon/blob/master/genmonlib/mysupport.py#L460

As I test I tried to duplicate this on Bullseye with python V3.9.3. It was running pstuil v5.8.0 so I upgraded to v5.9.0 and it still works for me. There must be some other combination that interacts with psutil. It is good to know. Thanks for follow up info.