jgyates/genmon #650: Confusion on features..

johndeerefarmer, 2022-01-16:
I have some questions on the features.

I want to be able to monitor the Generac at my vacation home AND if it on due to a power outage I want to be able to shed the load and then shut off the Generac to save on fuel over long periods. I understand that my WIFI must stay running so I can control the system remotely so I will have a UPS for it..
Then of course I want to be able to restart the Generac and then put the load back on it.

Finally I understand that some of Generac's software was encrypted so GenMon doesn't work?

Thanks very much

skipfire, 2022-01-16:
The Generac generator itself has no way to shed loads and genmon will not have any way to let you do that either.  Generac does have load-shed devices of questionable design and reliability, but they are automatic and not something that has any data input or output to control them.  The generator also will not shut down during an outage unless you set the physical switch to off or use the emergency shutoff, there are no commands that the Generator will obey to shut down during an outage.

Lastly, not just your wifi needs to be running (easily solved with an APC UPS - though not a Cyberpower in my experience), but your internet service needs to also stay up.  Many internet service providers have no backup power at distribution nodes and go out as soon as there is an area outage, and in my area the internet goes out 2 hours (give or take a few minutes) after an area outage (assuming the lines themselves are undamaged).  Remote management during a long area outage is at best problematic, though through no fault of the software or hardware.

Genmon does still work even after the encrypted firmware was released, though a future firmware update could always change that.

johndeerefarmer, 2022-01-16:
I am confused. Pixs of the software interface show buttons to start, stop and remove load from generator.

Also this is from GenMon's description:
Ability to start, stop, exercise and start / active the transfer switch (i.e. power your house off the generator) remotely.

My Internet service is via cellular and they have generator backup so probably not a problem there

Thanks

skipfire, 2022-01-16:
I'm not sure about the remove load from generator button, I've not seen that anywhere.  The ability to start, stop, exercise, and start with transfer are all commands the software can send to the generator, but the generator controller will ignore the STOP commands during an outage.  The buttons are great for testing and do work as expected outside of an outage.

johndeerefarmer, 2022-01-16:
I remembered wrong, the buttons are stop generator, start generator and start generator and transfer but if I can't stop the genset with a power outage that won't do me any good..
Thanks for the explanation...

I guess with Genmon I can at least know when the genset is on due to power loss. Then all I need to do is have a way to remotely monitor my propane tank's gauge. I bought the Tank Utility device last year and it's junk. You have to have an unsecure WIFI for it to work and I am not going to turn off security on my system..

.  Sounds like I need to come up with a way to have a UPS supply commercial power to the controller during a power outage. Then once I trick it into thinking commercial power has been restored I can shut it off..

jgyates, 2022-01-16:
I use the Tank Utility to monitor a propane tank and it works with my secure wifi. I did have one unit die, but they sent me another one at no charge.

johndeerefarmer, 2022-01-16:
I could never get mine to connect unless my wifi was unsecure. I tried two different routers.

grsthegreat, 2022-01-17:
Very odd, my tank utility works on my secured router

closed 2022-01-17 by jgyates