jgyates/genmon #651: No emails at exercise this week

skipfire, 2022-01-17:
### Expected Behavior

Receive emails when generator exercises.

### Actual Behavior

I had a test cycle today (10AM CST) and did not receive emails or text messages about it.  I just did the test email settings and I received the email.  I have not changed configuration in the last week.  I did receive the startup email after upgrading to 1.18.04.

### Steps to Reproduce (including precondition)

Not sure it is reproducible.

### Your Environment

- Generator Model: G0070432 - 22kW air cooled
- Genmon Logs: Submitted (at 3:28 PM EST)
- Genmon Version: V1.18.04

Is there someplace I can easily view the history of messages and outcome?  Not necessarily the content, but maybe destination, subject, and date/time?

jgyates, 2022-01-17:
The log file /var/log/mymail.log would have any errors that occurred. I am not sure what occurred. there have not been changes to email or exercise related functions in quite a while. One thing to check is that you have both a primary and secondary DNS on your pi. You can use google DNS servers as a backup.

     nameserver=192.168.1.1.  # your local router
     nameserver=8.8.8.8.         # google DNS for your backup.

Also make sure you check your spam filter. What gets caught in these is dynamic so it change change based on your email provider.

skipfire, 2022-01-17:
I'm on office365 and sending to self so all the spam filters get bypassed, but checked anyway and nothing there.  I do notice there are a lot fewer entries today than last Monday.  All the entries for 1/10 and 1/17 are below, I removed the entries that correspond to the update and restart.  If this logs every time it tries, then this looks like it tried 1 email (maybe the first?) this morning instead of the normal 4, and on previous Mondays I see 4 each time.

```2022-01-10 10:00:41,686 : Genmon <jon@pintsize.me>
2022-01-10 10:00:51,280 : Genmon <jon@pintsize.me>
2022-01-10 10:05:51,274 : Genmon <jon@pintsize.me>
2022-01-10 10:06:02,207 : Genmon <jon@pintsize.me>
.....
2022-01-17 10:00:44,929 : Genmon <jon@pintsize.me>```

jgyates, 2022-01-17:
Does the log page show and exercise on the expected day. I want to validate that it actually ran (assuming you did not hear/see it run). Also, check your myserial.log and mymodbus.log for errors (and the Monitor page if you have not restarted the software since then). If the software missed the update of modbus register 0001 due to comm errors for the duration of the exercise, then this would cause the issue as well. Any changed in register 0001 is the event that starts the notification process.

skipfire, 2022-01-17:
I did hear it run for the 5 minute exercise.

The log does show the exercise
```
    Run Log : 
        01/17/22 10:05:48 Stopped - Auto 
        01/17/22 10:00:36 Running - Exercise 
        01/10/22 10:05:43 Stopped - Auto 
        01/10/22 10:00:31 Running - Exercise 
        01/03/22 10:05:38 Stopped - Auto 
        01/03/22 10:00:26 Running - Exercise 
```
myserial.log is empty

mymodbus.log has nothing for today, here are the last 2
```
2022-01-08 20:27:28,899 : Error: timeout receiving slave packet for register 005d Buffer: 0
2022-01-14 03:07:32,439 : Error: timeout receiving slave packet for register 0005 Buffer: 0
```

I'm open to chalking it up to a glitch and see what happens next Monday.

jgyates, 2022-01-17:
Yeah, that is odd. It is like the modbus register never changed. Is this an Evolution 2.0?

skipfire, 2022-01-17:
Yeah, Evo2 with pre-encryption firmware.  This is the first time it has missed emails aside from when I had no internet for 10 days after Ida.

jgyates, 2022-01-17:
the reason I ask is that Evo2 does throw some odd values for register 0001 from time to time. These invalid values cause emails to be sent since register 0001 changes. There is an option on the advanced settings called "Ignore Unknown Values" that will attempt to remove the unknowns registers values by filtering them out. If you have this option enabled and the Evo2 spit out an invalid value during the exercise, this would also cause and email not to be sent.  If that is the case it is probably just bad timing as the strange results for register 0001 happen only on rare occasions. Other than that I don't have any good ideas as to the cause. Let me know if you get them next week.

skipfire, 2022-01-17:
That setting is off on mine.  Next week it is.

jgyates, 2022-01-17:
if you can remember, try to have a browser open during your exercise cycle since it logs the registers as they change. I am going to close this issue for now but post your updates to this thread.

closed 2022-01-17 by jgyates