jgyates/genmon #652: SNMP MIB file available?

firestormv1, 2022-01-18:
This is a pretty basic question, so forgive me if I don't fill out the template.  I looked through the wiki but the SNMP monitoring portion is very basic (I'm willing to contribute my findings!).

My genmon install is working perfectly except for the SNMP monitoring, using the most current version as of a couple of days ago (v1.18.04).

Is there an already defined MIB file available for genmon?   I can snmpwalk it if I specify the OID as "." however having a MIB file I can throw at monitoring systems (Nagios, Observium, others) may be useful. 

I only have access to an air cooled Guardian 24kW with Evolution 2.0 controller, but I might be able to figure out MIBs for others based on gensnmp.py.

Thank you.

jgyates, 2022-01-18:
Unfortunately I have not created a MIB. The closest I have is the documentation here:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensnmppy-optional

firestormv1, 2022-01-18:
I was able to cobble together a MIB last night and it looks to work well.

Give me a couple of days to polish it up and I'll be happy to send it to
you.

firestormv1, 2022-01-18:
Here's what I have so far (proof of work).  I need to get the OID info straightened out so snmptranslate will display correct info, but it looks to be working so far.  The below walk was done while the generator was not running, but ready to run (auto mode):

`GenMon-MIB::SwitchState = STRING: "Auto"
GenMon-MIB::EngineState = STRING: "Off - Ready"
GenMon-MIB::ActiveRelays = ""
GenMon-MIB::ActiveSensors = ""
GenMon-MIB::BatteryVoltage = STRING: "14.00 V"
GenMon-MIB::BatteryStatus = ""
GenMon-MIB::RPM = STRING: "0 "
GenMon-MIB::Frequency = STRING: "0.00 Hz"
GenMon-MIB::OutputVoltage = STRING: "0 V"
GenMon-MIB::OutputCurrent = STRING: "0.00 A"
GenMon-MIB::OutputPower = STRING: "0.00 kW"
GenMon-MIB::RotorPoles = STRING: "0 "
GenMon-MIB::UtilityVoltage = STRING: "249 V"
GenMon-MIB::UtilityMaxVoltage = STRING: "250 V"
GenMon-MIB::UtilityMinVoltage = STRING: "248 V"
GenMon-MIB::UtilityThresholdVoltage = STRING: "156 V"
GenMon-MIB::UtilityPickupVoltage = ""
GenMon-MIB::SetOutputVoltage = ""
GenMon-MIB::LastAlarmLog = STRING: "01/16/22 17:42:17 Low Battery : Alarm Code: 0000"
GenMon-MIB::LastServiceLog = ""
GenMon-MIB::LastRunLog = STRING: "01/16/22 18:26:33 Stopped - Auto "
GenMon-MIB::MonitorTime = STRING: "Tuesday January 18, 2022 02:40:09"
GenMon-MIB::GeneratorTime = STRING: "Tuesday January 18, 2022 02:40"
GenMon-MIB::Model = STRING: "Generic Air Cooled"
GenMon-MIB::Serial = STRING: "3010270275"
GenMon-MIB::DetectedController = STRING: "Evolution 2.0, Air Cooled"
GenMon-MIB::NominalRPM = STRING: "3600"
GenMon-MIB::RatedKW = STRING: "24 kW"
GenMon-MIB::NominalFrequency = STRING: "60 Hz"
GenMon-MIB::FuelType = STRING: "Natural Gas"
GenMon-MIB::FuelLevelSensor = STRING: " "
GenMon-MIB::EstimatedFuelInTank = STRING: " "
GenMon-MIB::EngineDisplacement = STRING: "999 cc"
GenMon-MIB::AmbientTemperatureSensor = STRING: "56 F"
GenMon-MIB::ThirtyDayKWH = STRING: "0.00 kWh"
GenMon-MIB::ThirtyDayFuelConsum = STRING: "2.22 cubic feet"
GenMon-MIB::TotalPowerFuelConsum = STRING: "2.22 cubic feet"
GenMon-MIB::ThirtyDayRunHours = STRING: "0.02 h"
GenMon-MIB::EstHoursFuelRemain = STRING: " "
GenMon-MIB::CurHoursFuelRemain = STRING: " "
GenMon-MIB::FuelInTank = STRING: " "
GenMon-MIB::FuelLevelState = STRING: " "
GenMon-MIB::CtrlCalCurrent1 = STRING: "1489 "
GenMon-MIB::CtrlCalCurrent2 = STRING: "1558 "
GenMon-MIB::CtrlCalVolts = STRING: "1051 "
GenMon-MIB::CtrlNominalLineVolts = STRING: "240 V"
GenMon-MIB::CtrlRatedMaxPower = STRING: "24 kW"
GenMon-MIB::CtrlParamGroup = STRING: " "
GenMon-MIB::CtrlVoltageCode = STRING: " "
GenMon-MIB::CtrlPhase = STRING: " "
GenMon-MIB::CtrlHoursProtection = STRING: " "
GenMon-MIB::CtrlVoltsPerHz = STRING: " "
GenMon-MIB::CtrlGain = STRING: " "
GenMon-MIB::CtrlTargetFreq = STRING: " "
GenMon-MIB::CtrlTargetVoltage = STRING: " "
GenMon-MIB::ExerciseTime = STRING: "Weekly Tuesday 14:00 Quiet Mode Off"
GenMon-MIB::ExerciseDuration = STRING: " "
GenMon-MIB::AirFilterServiceDue = STRING: " "
GenMon-MIB::OilServiceDue = STRING: " "
GenMon-MIB::SparkPlugServiceDue = STRING: " "
GenMon-MIB::BatteryServiceDue = STRING: " "
GenMon-MIB::ServiceADue = STRING: "199 hrs or 12/14/2023 "
GenMon-MIB::ServiceBDue = STRING: "399 hrs or 12/13/2025 "
GenMon-MIB::BatteryCheckDue = STRING: "12/14/2022 "
GenMon-MIB::TotalRunHours = STRING: "0.00 h"
GenMon-MIB::HWVersion = STRING: "V1.01"
GenMon-MIB::FirmwareVersion = STRING: "V1.13"
GenMon-MIB::GMMonHealth = STRING: "OK"
GenMon-MIB::GMUptime = STRING: "Generator Monitor running for 0:19:56."
GenMon-MIB::GMPwrLogFileSize = STRING: "0.00 MB of 15.00 MB"
GenMon-MIB::GMSWVersion = STRING: "V1.18.04"
GenMon-MIB::SerialPacketCount = STRING: "M: 37914, S: 37913"
GenMon-MIB::SerialCRCErrors = STRING: "0 "
GenMon-MIB::SerialCRCPercentErrors = STRING: "0.00%"
GenMon-MIB::SerialTimeoutErrors = STRING: "0"
GenMon-MIB::SerialTimeoutPercentErrors = STRING: "0.00%"
GenMon-MIB::SerialModbusExceptions = STRING: "4"
GenMon-MIB::SerialValidationErrors = STRING: "0"
GenMon-MIB::SerialInvalidData = STRING: "0"
GenMon-MIB::SerialDiscardedBytes = STRING: "0"
GenMon-MIB::SerialCommRestarts = STRING: "0"
GenMon-MIB::SerialPktsPerSecond = STRING: "63.39"
GenMon-MIB::SerialAvgTransactionTime = STRING: "0.0310 sec"
GenMon-MIB::PiCPUTemp = STRING: "97.70 F"
GenMon-MIB::PiHardwareModel = STRING: "Raspberry Pi 4 Model B Rev 1.2"
GenMon-MIB::PiCPUFreqThrottle = STRING: "OK"
GenMon-MIB::PiARMFreqCap = STRING: "OK"
GenMon-MIB::PiUndervolt = STRING: "OK"
GenMon-MIB::PiCPUUtil = STRING: "4.46%"
GenMon-MIB::PiOSName = STRING: "Raspbian GNU/Linux"
GenMon-MIB::PiOSVersion = STRING: "10 (buster)"
GenMon-MIB::PiSysUptime = STRING: "1 day, 8:42:05"
GenMon-MIB::PiNICUsed = STRING: "eth0"
`

So far, the only thing that might be variable is the OID that's assigned to GenMon. Right now I have the MIB written as using the default Enterprise ID of '9999', if a user wants to use SNMP and they change the value, they'll also need to change the Enterprise ID in the MIB, otherwise the MIB won't work and SNMP tools won't recognize it's genmon they're talking to.  This isn't a genmon thing per-se, but how SNMP works overall.  I don't know if applying for a PEN (Private Enterprise Number) would be something you want to pursue, but outside of the one caveat dealing with the Enterprise ID, the MIB appears to be functional at a minimum.

jgyates, 2022-01-18:
That looks great so far. Let me know when you think it is read and I can add it to the repository.

elias314, 2022-01-19:
This isn't a question about the MIB specifically, but wouldn't it be better to have the values expressed as numbers without the units so they're easier to consume and graph as numbers; e.g. "240" instead of "240 V"?  I think most of the SNMP tools I've used in the past don't have a facility to parse out the units and expect just a number.

jgyates, 2022-01-19:
@elias314 

You have the option of returning numeric values only with the "Numerics only" option under the SNMP add on.

jgyates, 2022-01-20:
You certainly appear to know your stuff on SNMP. I was leaning toward the second option, but if have want to do any of them I am open to that as well. I appreciate you doing this, it will really help the project. If you think option 3 is better we can do that as well. If the SNMP addresses are changing I would rather make them as compliant as possible in one change rather than several incremental changes. In short, what ever you are willing to do I will incorporate into the SNMP support.

also, I applied for an enterprise ID as you suggested. That is a good idea. Hopefully this will be complete in a week or two. We can make the MIB use the enterprise ID and leave the option for people to change.

firestormv1, 2022-01-20:
I'm all for helping out, and I'm enjoying the challenge!  (Seriously, SNMP is a monster of a rabbithole, lol).

Let's do this, I'll work on SMIv2 compliance while we wait for the enterprise ID.  When you get the new ID, let me know what it is and my PR will contain the SMIv2 compliant MIB and the gensnmp.py that is correct (doesn't use .0 for leaves), contains the new working MIB, and contains the new enterprise ID set as default (although users can still change it if desired).

This way, it's only one PR that contains everything necessary for SMIv2 compliance and users can use the included MIB if they want.   I'll include a README with the MIB so users can take it and run with it.

firestormv1, 2022-01-20:
Just to put my money where my mouth is, I forked genmon and applied the changes for SMIv1. It's fully operational on my Pi (now it's a test box, lol) so I know it works for Evolution controllers and now passes smilint for compliance!

I also just noticed that I failed to capture a few OIDs for the H100 controller (and possibly Nexus too).  I'll go line by line through the OIDs in gensnmp.py and make sure all OIDs are accounted for in the MIB.  Because there's generator specific OIDs, I'll make sure to expand the description labels to contain that information.  If I'm gonna do it, I'm gonna do it right. :D

I'll advise as progress is made.  

Thank you!

jgyates, 2022-01-21:
@firestormv1 
The genmon enterprise ID is 58399

Also, if you do a PR, put the MIB file in the genmon/data folder

firestormv1, 2022-01-21:
Awesome!  That didn't take long at all! I figured IANA was much slower than that.

I'll go ahead and start putting things together.   It might be a few days but I will get you a status update by next week.

bbrown-sc, 2022-01-21:
Brings back memories. I got our company's enterprise ID #45 back in 1991. Pre-Internet, if you can imagine that. Back then you could call Joyce Reynolds at ISI to get your enterprise number and pick up a couple of class B addresses at the same time! You can thank me for sysLocation OID 1.3.6.1.2.1.1.6 (rfc1213).

firestormv1, 2022-01-22:
Ok, so I've hit a bit of a snag. I can't seem to figure out how to get gensnmp.py to return the OID for the device.

I thought I could be crafty and sneak it under the uptime and the genmon descriptor but I don't know enough python to get it to work right:

self.mibData.append(MyOID((1,3,6,1,2,1,1,2,0),return_type = type(OID), description = "OID", default = ".1.3.6.1.4.1.58399", log = self.log))

I'm not sure how to get the type(OID) to pysnmp, I just know enough python to scrape by.  The OID type is important as it tells NMS applications (and snmpwalk) what the vendor specific OID is for a particular device. Some NMSes (like Observium) won't even register a device if it can't see the vendor specific OID in an snmpget/snmpwalk.  Examples of this are below.

If you can guide me on how to include the custom OID type to pysnmp, I can roll it in with the rest of the SNMP update.  I've already flattened the OIDs and confirmed operation under SMIv1.  Once this is working, I can then work on SMIv2. 

Please let me know. 

*genmon*
matt@little-black-book:~/development/genmon$ snmpwalk -On -v1 -c public generac 
.1.3.6.1.2.1.1.1.0 = STRING: Genmon Generator Monitor
.1.3.6.1.2.1.1.3.0 = Timeticks: (45473) 0:07:34.73
(missing .1.3.6.1.2.1.1.2.0 OID)

*Weathergoos-V1*
matt@little-black-book:~/development/genmon$ snmpwalk -On -v1 -c public goose | head -3
.1.3.6.1.2.1.1.1.0 = STRING: WxGoos-1
.1.3.6.1.2.1.1.2.0 = OID: .1.3.6.1.4.1.17373.2
.1.3.6.1.2.1.1.3.0 = Timeticks: (322125248) 37 days, 6:47:32.48

*HP Procurve switch*
matt@little-black-book:~/development/genmon$ snmpwalk -On -v1 -c public procurve-poe | head -3
.1.3.6.1.2.1.1.1.0 = STRING: HP J8693A Switch E3500yl-48G, revision K.15.06.0017, ROM K.15.30 (/ws/swbuildm/ec_rel_eureka_qaoff/code/build/btm(ec_rel_eureka_qaoff)) (Formerly ProCurve)
.1.3.6.1.2.1.1.2.0 = OID: .1.3.6.1.4.1.11.2.3.7.11.59
.1.3.6.1.2.1.1.3.0 = Timeticks: (270613562) 31 days, 7:42:15.62

firestormv1, 2022-01-22:
Hah!  Disregard!  I figured it out (well, partially).   It wouldn't let me give the OID definition after timeticks because it was out of order.  (.1 then .3, then .2)  

matt@little-black-book:~/development/genmon$ snmpwalk -On -v1 -c public generac 
.1.3.6.1.2.1.1.1.0 = STRING: Genmon Generator Monitor
.1.3.6.1.2.1.1.2.0 = Wrong Type (should be OBJECT IDENTIFIER): STRING: ".1.3.6.1.4.1.58299"
.1.3.6.1.2.1.1.3.0 = Timeticks: (480) 0:00:04.80

Now to see if I can correct the type.

firestormv1, 2022-01-22:
Ok, I think I spoke too soon.  I got the three OIDs to show using a "string" type for .2.0 (middle line from last post).  The problem is that it needs to be OBJECT IDENTIFIER otherwise it won't work.

I've tried defining the type in MyOID, however it doesn't appear to work without breaking someting.  If I place the new definition after TimeTicks, then the second and third OID come out as Timeticks.  If I place the new definition before TimeTicks, it breaks the third OID. 

With ObjectIdentifier after Timeticks (from MyOID), the conditional looks like this:

```
        try:
            if self.return_type == str:
                return api.protoModules[protoVer].OctetString(self.value)
            elif self.return_type == int:
                return api.protoModules[protoVer].Integer(self.value)
            elif self.return_type == type(TimeTicks):
                return api.protoModules[protoVer].TimeTicks((time.time() - self.birthday) * 100)
            elif self.return_type == type(ObjectIdentifier):
                return api.protoModules[protoVer].ObjectIdentifier(self.value)
            else:
                self.LogError("Invalid type in MyOID: " + str(self.return_type))
                return  api.protoModules[protoVer].Integer(self.value)
        except Exception as e1:
            self.LogErrorLine("Error in MyOid __call__: " + str(e1))
            return None
```
and the snmpwalk looks like this:

```
.1.3.6.1.2.1.1.1.0 = STRING: Genmon Generator Monitor
.1.3.6.1.2.1.1.2.0 = Wrong Type (should be OBJECT IDENTIFIER): Timeticks: (28448) 0:04:44.48
.1.3.6.1.2.1.1.3.0 = Timeticks: (28449) 0:04:44.49
```
(Note:  Both OID (2.0) and systemUpTime.0 (3.0) are showing as type Timeticks, this is incorrect.)

If I put ObjectIdentifier before Timeticks in the MyOID conditional, 

The MyOID conditional looks like this:
```
        try:
            if self.return_type == str:
                return api.protoModules[protoVer].OctetString(self.value)
            elif self.return_type == int:
                return api.protoModules[protoVer].Integer(self.value)
            elif self.return_type == type(ObjectIdentifier):
                return api.protoModules[protoVer].ObjectIdentifier(self.value)
            elif self.return_type == type(TimeTicks):
                return api.protoModules[protoVer].TimeTicks((time.time() - self.birthday) * 100)
            else:
                self.LogError("Invalid type in MyOID: " + str(self.return_type))
                return  api.protoModules[protoVer].Integer(self.value)
        except Exception as e1:
            self.LogErrorLine("Error in MyOid __call__: " + str(e1))
            return None
```

The snmpwalk looks like this:

```
.1.3.6.1.2.1.1.1.0 = STRING: Genmon Generator Monitor
.1.3.6.1.2.1.1.2.0 = OID: .1.3.6.1.4.1.58299
.1.3.6.1.2.1.1.3.0 = NULL
```
(Note, OID is correct type now, but systemUpTime (.3.0) is broken now)

and I get an error message in /var/log/gensnmp.log:

`2022-01-22 01:41:01,472 : Error in MyOid __call__: Malformed Object ID None at ObjectIdentifier: 'NoneType' object is not iterable : gensnmp.py:82`

Unfortunately, my python-fu is not very stronk so I need your help.  I'm not sure what to do to get type(TimeTicks) and type(ObjectIdentifier) to work together.  .2.0 has to be ObjectIdentifier type, and .3.0 has to be TimeTicks type in order for SNMP to be functionally correct.

Any suggestions?

jgyates, 2022-01-22:
you can send me what you have so far and I can take a look tomorrw and see if I can figure out the source of the error. I will look at it tomorrow. Has this line:

       self.mibData.append(MyOID((1,3,6,1,2,1,1,3,0),return_type = type(TimeTicks), description = "Uptime",log = self.log))

in SetupSNMP changed?

jgyates, 2022-01-22:
you can also email me directly at generatormonitor.software@gmail.com

firestormv1, 2022-01-24:
Just a quick update to the thread, it's in active development.  jgyates has gotten me unstuck (thank you!) and I am proceeding with refactoring for SMIv2 compliance.  We're almost there!

jgyates, 2022-01-28:
Version 1.18.05 has an updated gensnmp add on, a MIB file and is SMIv2 compliant thanks to @firestormv1 

Please note that the OIDs have changed to better comply with the standard and it is now recommended to use the genmon enterprise ID 58399 instead of the older 9999 default. 

If there are any questions on this please post them to this thread. See the [wiki](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensnmppy-optional) and the [change log](https://github.com/jgyates/genmon/blob/master/changelog.md) for additional details.

closed 2022-01-28 by jgyates