jgyates/genmon #655: Overcrank message

moisedo, 2022-01-21:
Question: Is there a way to clear the overcrank message from the Web UI? - if not, How difficult will it be to add that functionality.

Thanks!

jgyates, 2022-01-21:
I don't have a lot of of info to go on with your question so I am assuming you are running an Evolution air cooled generator as that is the most popular with this project. Are you referring to the section on the main Status page titled "Last Alarm Entries"? If you are then this is not indicating an alarm condition, but rather the last entries in the three alarm logs kept by the controller. If you are referring to an active alarm on the controller then you must clear the alarm at the controller. You can clear an alarm at the controller by pressing OFF then ENTER on an Evolution controller.

This software supports several different generator controllers so my answer may not be suitable for your question since you did not include those details.

jgyates, 2022-01-25:
I am going to close this issue since I have not heard back I assume you resolved this question. If you still have questions feel free to post to this thread.

closed 2022-01-25 by jgyates