jgyates/genmon #656: Tank Utility connectivity

mikeodea, 2022-01-21:
First off: Genmon is amazing, thank you for creating it

TankUtility integration was working and then stopped. Unsure why. 

- Access to the TU UI works fine 
- Auth is correct
- TU UI shows correct fuel level 
- Genmon is latest version 
- Genmon TU tile shows 0 fuel 
- With TU enabled, I'll get notifications from Genmon about the tank being less than 10% capacity, which is not correct. 

At some point (unfortunately I am not sure when, but at least since september of '21) I lost the ability to read the tank level via the TU plugin in GenMon. A manual login to both the mobile app and web app for TU work fine, the auth is correct and I am getting real time readings. 

I have tried to re-install Genmon after wiping out the install directory and reconnecting new to TU, but I cant seem to figure out. The errors printed in the tankutil log is: 

2022-01-21 14:59:27,273 : Error logging in in TankCheckThread, retrying 

my Genmon install is running on a dedicated Raspi 4, powered via PintSize Generator Hat, using WiFI for connectivity 

### Your Environment

- Generator Model: Generac     Model : G0070432 22kw LP 
- Genmon Logs: Just submitted 
- Genmon Version: V1.18.04

jgyates, 2022-01-21:
Can you validate the "Tank Name" parameter in the genmon add on is the same as the "Device Name" in the  TU web interface. If these values are different it would cause this error.

mikeodea, 2022-01-21:
sure thing. I just checked that the "Tank Name" field of the TU Add On matches the same value in the TU UI "Device Name" - and then restarted. It did not change the status and the log line remains the same as I posted. One thing to note, the device name on the TU side is in the format (PII scrubbed): "XXXXXX 330, XXXXXXXXXX" - Could the coma would be  a factor?

jgyates, 2022-01-21:
it is possible, there was a change in V1.18.04 in the reading of the config files that allowed for parameters (i.e. passwords, tank names) to be read from a file with special characters. That could be related to the issue. I just checked in an updated version of /genmon/genmonlib/mytankutil.py that has some additional logging added. Edit the file /etc/genmon/gentankutil.py and add this line:

      debug = True

Then save the file and restart genmon. Wait 30 seconds and send your logs again. This should include additional debug info in /var/log/gentankutil.log

Let me know if you have any questions.

jgyates, 2022-01-21:
Also, once we are finished with this process do not forget to remove the debug = True line in the config file as this will add wear to the SD card if this option is left enabled for a long time.

mikeodea, 2022-01-22:
Got it. Added the line to  /etc/genmon/gentankutil.py,  restarted & sent logs. The format of gentankutil.log appears the same, FWIW.

jgyates, 2022-01-22:
I got your logs. Did up update your software on the about page first? I did not explicitly say that but the update will not be on your system until you do that so I will not see any additional debug info.

jgyates, 2022-01-22:
and it the file to change is /etc/genmon/gentankutil.conf, not /etc/genmon/gentankutil.py, sorry about the confusion on this. typo on my earlier reply.

mikeodea, 2022-01-22:
No worries - Yep, I did not update before updating the conf. Re-ran update, restarted and now I have debug. Sending logs now

jgyates, 2022-01-22:
I got your logs, it looks like there is a mismatch in the Tank Name and the Device Name. 
[image]

In the above the "Device Name" is "Propane Tank"

[image]

In the above the "Tank Name" is also "Propane Tank"

The "Device Name" on TU online and "Tank Name" in genmon must be the same. Yours do not appear to match. The logs indicate your genmon add on Tank Name is "Fxxxxxxx 330, xxxxxxxxxe" but your Device Name online is a long sequence of numbers "0025xxxxxxxxxxxxxxxxx832"

gentankutil.py is not finding a tank with the name you specify. It does log in OK but the log shows the tank name specified is not the Device Name given from TU on line. 

If you make these two match it should work for you. Let me know of you have any questions.

mikeodea, 2022-01-22:
Ah ha! I'd probably have missed that - but that seemed to do the trick. Whats funny is I seem to recall putting a long ID in there when I first set it up (which I got from the TU UI) but the LP utility maybe changed it at some point and when I reinstalled this I copied what was in the TU UI at install time, resulting in the mismatch. Thank you for the help.

jgyates, 2022-01-22:
No problem. I am glad it is working for you.

closed 2022-01-22 by jgyates