jgyates/genmon #658: Warranty

aric8456, 2022-01-30:
Is this expected to have an impact on the warranty for a Generac? Should I plan to remove the pi every time my dealer comes for service, or is this innoculous in the eyes of Generac?

jgyates, 2022-01-31:
I removed mine the first time, but after that I left it attached. In my experience the service folks are there to make a living on service, not creating problems with their customers. Of course I can't speak for everyone, just my experience.  They did not ask questions other than "is that a monitor?" (i said yes) and that was only when my controller had an issue, for general service they did not even ask questions.

Considering that this project does not alter the functionality of the controller or generator in anyway (it only adds external features based on existing functionality of the controller) I would guess that they would have a hard time invalidating a warranty by using this project. If you connected power to your controller and your actions caused your controller to blow, then they might have an issue, but I have yet to run into anyone who has done that.

grsthegreat, 2022-01-31:
i have no idea if a certain service guy will complain or not. when i do service on units i could care less if there is a 3rd person or after market item on generator, unless thats the reason unit was damaged. Genmon causes no issues with the controller, so i dont see any issue.

closed 2022-02-06 by jgyates