jgyates/genmon #659: Implementing gentankdiy with h-100 Panel

j45p3r2, 2022-01-31:
I have a Generac QT025 with 2.4L engine powered by propane and H-100 panel. I was hoping to utilize gentankdiy to monitor fuel level but I see it is not supported with H100 panel. Is this still correct?

jgyates, 2022-01-31:
I took a look the coder to refresh my memory and it looks like the support is already there. You just need to disable the "Use Sensor for Fuel Gauge" on the settings page and enable the "DIY Fuel Tank Gauge Sensor" add on. Let me know if you have any issues or if the documentation needs to be updated.

jgyates, 2022-02-06:
I am going to close this issue for now but feel free to post here for followup questions or status.

closed 2022-02-06 by jgyates

j45p3r2, 2022-02-16:
Hi I just wanted to report back that the DIY Fuel Sensor add on works perfectly with the H-100 panel.

jgyates, 2022-02-16:
Great! Thanks for the feedback.