jgyates/genmon #662: Estimated Fuel still shown even when accurate fuel amount is available

lps-rocks, 2022-02-05:
### Expected Behavior

Estimated Fuel should be disabled when a sensor is detected or plugin is used to monitor fuel source

### Actual Behavior

Estimated Fuel amount remains on maintenance page

### Steps to Reproduce (including precondition)

1. Configure Tank Utility Add-On to pull tank details.
2. Configure total tank size in settings
3. Go to Maintenance page, estimated fuel is still shown despite an accurate number being available.

### Screenshot or Pictures relating to the problem (if possible)

[image]
[image]

### Your Environment

- Generator Model: 22kW Home Guardian w/Evo 2 Controller
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.04

jgyates, 2022-02-05:
@lps-rocks 

I got your PRs, thanks. 

The fuel estimation can be disable by setting the tank size to zero in the settings. Let me know if that does not solve the issue for you.

For some, like myself who have a diesel tank the built in fuel sensor is accurate most of the time but fuel percentages over 92% and below 10% are not accurate so having both is handy. Have the ability to show both was intentional.

lps-rocks, 2022-02-05:
Unfortunately, disabling the tank size also makes it stop showing the measured amount as well, it also disables the estimated runtime remaining calculations.

[image]

jgyates, 2022-02-05:
I will look into this. Currently genmon does not use the tank size parameter from tank utility since it already had that data from local settings. I am hesitant to make this change until I can look at it closely as this removes functionality. In the interim, If you want to compensate for a non full tank you can either use a lower tank size or use the "Subtract Fuel" parameter on the settings page.

lps-rocks, 2022-02-05:
Could we make it an advanced option to disable the display of estimated tank fuel? It does look like the estimated runtime remaining accurately uses the sensor data, so nothing needs to be changed there.

lps-rocks, 2022-02-05:
I created a PR with the logic I changed on my installation so it functions properly now without having to grab the capacity from tank utility. It looks like tank utility is already grabbing the capacity but just isn't *using* it. 

It may show what I'm trying to do a little bit better.

PR #664 (2022-02-05): Don't show estimated levels for propane tanks w/sensors
# Pull Request Template

## Description

Propane tanks are never filled to 100% and refills are anywhere from 80-85% of total tank size, so tank monitoring utilities and sensors never represent 100% filled. So hide the Estimated value (which assumes full = 100%) whenever a more accurate value is available.

Fixes #662 

## Type of change

Please delete options that are not relevant.

- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## How Has This Been Tested?

Please describe the tes [... cut]
closed 2022-02-05 by jgyates