jgyates/genmon #668: External CTs not showing when no generator connected

skipfire, 2022-02-05:
I'm working on some improvements to the CTs, but the data is not showing with the current version.

### Expected Behavior

When the External CT add-on is enabled and Strict is OFF, the gauges and CT data would show with or without a generator present.

### Actual Behavior

Not seeing the External CT gauges or data in a fresh genmon install that has never been connected to a generator even with the add-on enabled, strict off, and the Sensor HAT connected.

### Steps to Reproduce (including precondition)

Enable the add-on, turn strict off, wait for genmon to restart, view page.  This happened with a fresh install as well as an install that I've previously been using where the CT data was visible.

### Screenshot or Pictures relating to the problem (if possible)

[image]
[image]

### Your Environment

- Generator Model: none connected
- Genmon Logs: submitted from 2 different installs
- Genmon Version: v1.18.05 (installed around 2022-02-05 11:00 AM EST)

jgyates, 2022-02-05:
I can not duplicate this. Also, I have not received any logs on this, do you have outbound email enabled? Looking at /var/log/gencthat.log may provide some insights. Have you edited /boot/config.txt?

jgyates, 2022-02-05:
I can duplicate it now. i will let you now what I find.

skipfire, 2022-02-05:
I had edited /boot/config.txt and tested with my stand alone script so I know that it was reading from the HAT.  After that I installed GenMon and my setting is gone and the line  where it was (immediately above the [CM4] which was the first type tag) is where I think genmon install added the top dtoverlay=pi3-disable-bt.  After adding the spi1-1cs line back in I got the text for the legs but not the gauges for the CTs, CPU, or WiFi, but I'm not worried about the gauges since I can see the legs.

jgyates, 2022-02-05:
I had a problem in my test setup. I can not duplicate this issue. I don't see a change that could impact this recently and I can not duplicate. Can you provide logs? Mine shows up with and without a generator connection.

skipfire, 2022-02-05:
I have no log entiries in gencthat.log after re-adding the dtoverlay line, so I think I was just seeing the impact of the lost config.txt line.  Thanks.

closed 2022-02-05 by skipfire