jgyates/genmon #670: Informational- found the best box for 3b with hat

Don331, 2022-02-05:
After testing 6 different boxes, I’ve found one the holds the 3b/hat kit from pint size.me and allows for and external antenna. 
The box is Amazon LMioEtool ABS Plastic Dustproof Waterproof IP65 Junction Box Hinged Shell Universal Electrical Project Enclosure Gray 5.9 x 5.9 x 3.5 inch (150 x 150 x 90mm)

I made a backer plate from a piece of roof strapping, so the magnets on the pi hold it in place
[image]
[image]
[image]

jgyates, 2022-02-06:
This one?
https://smile.amazon.com/LMioEtool-Dustproof-Waterproof-Electrical-150mmx150mmx90m/dp/B07KXSPRGC/?th=1

Don331, 2022-02-06:
Yes, that’s it!

Don Barry

danorum, 2022-03-26:
How did you get the connecter through that hole in the box?

Don331, 2022-03-26:
You have to release the pins out of the connector.  There’s a tool you can buy but it doesn’t work well. 
What i used was a staple. Worked great. 

https://youtu.be/IFR9Wnslruk

Don Barry

ssmoss, 2023-09-12:
@Don331 do you mind sharing which wifi adaptor you used?

Don331, 2023-09-12:
EDUP AC600M USB WiFi Adapter for PC, Wireless USB Network Adapters Dual Band 2.4G/5.8Ghz Wi-Fi Dongle with Antenna for Laptop Desktop Compatible Windows 10/11/8.1/8/7/XP/Vista/Mac OS X 10.6~10.15.3https://a.co/d/0BZryIkDon Barry

ssmoss, 2023-09-12:
@Don331 Thanks for your quick reply. Did you have to build drivers for it or does it work out of the box on a Pi 3A+?

Don331, 2023-09-12:
Worked out of boxDon Barry

ssmoss, 2023-09-12:
@Don331 One last question, which antenna and adapter did you use to get though enclosure?

ssmoss, 2023-09-12:
Just to update, the [linked](https://a.co/d/0BZryIk) adapter seems to come with a different chipset. It is using the `RTL8811CU` chipset which does not seem to be supported out the box using the 64bit headless version of Raspberry Pi OS. That said `fastoe` has created Pi version of the drives located [here](https://github.com/fastoe/RTL8811CU_for_Raspbian). There is one catch if you are using the 64bit version you need to change one step in the instructions per this [issue](https://github.com/fastoe/RTL8811CU_for_Raspbian/issues/11) and use  `make ARCH="arm64”` instead of just `make`. The build does take a while will need to be updated if your kernel changes per this [issue](https://github.com/fastoe/RTL8811CU_for_Raspbian/issues/13). Once it completes and is installed a second `wlan` interface will appear and it will connect to the network that onboard wifi connects to. I haven't personally done this but it seems like you can set it up to connect to two different networks if wanted per this [guide](https://www.jeffgeerling.com/blog/2021/working-multiple-wifi-interfaces-on-raspberry-pi). My plan is to have both connected to the same network and use an external antenna on the second one. The only downside is that it takes up the only USB slot on the Pi 3A+ so a cheap USB 2.0 Hub might be in order.