jgyates/genmon #673: Best way to get data into Grafana?

TopBanana123, 2022-02-12:
Hopefully this is the right place to post. I'm looking to get some basic information into Grafana, either through InfluxDB or any other method

All I need is the Generator Status, battery Voltage, Charger Status and RPM. I'm making a power dashboard and I'd like to get stats from Genmon in there

I've seen some posts on going from MQTT to HA to InfluxDB to Grafana. Not being an HA user, I'm trying to find out if there is a way without getting that setup too?

Screenshots of my progress (IoTaWatt + InfluxDB + Grafana)

[image]

[image]

skipfire, 2022-02-12:
Grafana has a couple of ways to directly interface without going through any middlemen, apparently there is a specific JSON format that is natively supported by grafana for any system that implements it.  I am working on one, but I haven't had the free time I'd like to just crank it out.

TopBanana123, 2022-02-12:
I'll email you some of my free time if that helps

skipfire, 2022-02-12:
I wish, I've got some time blocked tomorrow to work on it, so I will see how straightforward it is then.

TopBanana123, 2022-02-12:
Just to put it out there, I would pay for the feature

skipfire, 2022-02-12:
Are you using a local or cloud Grafana?  I'm looking at local and cloud I expect would work basically the same, but of course would have router/firewall concerns.

TopBanana123, 2022-02-12:
I'm using local, my goal is for all of my monitoring to work during a 2 week outage with no internet

skipfire, 2022-02-13:
There is a plugin that can read from genmon; if it has no security enabled.  I'm not sure if there is a way to handle the security with what the plug and genmon already do or if something custom has to be written.  Maybe an option in genmon to allow 

https://marcus.se.net/grafana-json-datasource/installation/

Once that is done you can configure the datasource to the genmon status_json route, http://{genmonip}/cmd/status_json and save/test.  With that done you can add panels using the "JSON API" data source and a field path like "$.Status[*].Engine[*].Battery Voltage" and the configure the rest as needed.

TopBanana123, 2022-02-14:
Thanks Skipfire, I am going to try that tonight. I've also been modifying the Generator Dashboard. Currently the information on the top right is non-functional

I think I've got it in a pretty good place. Would like better Min/Max displays for Voltage and Frequency though

[image]