jgyates/genmon #674: cts info

andrewvanlagen, 2022-02-14:
how many cts can i run with Genmom

jgyates, 2022-02-14:
I assume you are talking about CTs (current transformers), let me know if that is not the case. With the addon and Hat from pintsize.me you can run 2 CTs.

andrewvanlagen, 2022-02-14:
That chip set can handle. 8 current cts. 
Does you software handle that

andrewvanlagen, 2022-02-14:
Ok Thanks

skipfire, 2022-02-14:
The HAT only supports 2, and the additional circuitry is not present to support more, but there is space that it would be possible. It is planned that a channels could be used for a fuel monitor.  The pHAT and future versions of the HAT are going to use the 4 channel version of the chip unless other requirements are present as will the combo board when I have it ready next month (I hope), but I am also looking at using a different ADC that supports greater resolution and uses I2C instead of SPI, but is also 4 channels.  What would the extra CTs monitor?

TopBanana123, 2022-02-14:
If you are looking for more CT's for other circuits, look into an IoTaWatt