jgyates/genmon #675: SMS/Twilio not sending after power outage

dieseld23, 2022-02-16:
### Expected Behavior

After power loss, (and loss of internet until generator powers up and router boots up) Twilio SMS messages should continue to attempt sending until successful

### Actual Behavior

After power loss, Twilio seems to only try twice to send and stops. 

### Steps to Reproduce (including precondition)

Cut power, generator turns on. After 8 minutes no SMS messages telling generator is on/running/power loss (power cut so generator saw an actual power loss). 

Output of log

2022-02-16 08:19:40,298 : Error: HTTPSConnectionPool(host='api.twilio.com', port=443): Max retries exceeded with url: /2010-04-01/Accounts/XXXXXXXXXXXXXXXXXXXXXXXXXXX/Messages.json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x74dcdd90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
2022-02-16 08:20:05,451 : Error: HTTPSConnectionPool(host='api.twilio.com', port=443): Max retries exceeded with url: /2010-04-01/Accounts/XXXXXXXXXXXXXXXXX/Messages.json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x74d15470>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

### Your Environment

- Generator Model: 0065510 (22kW Evolution, Air Cooled)
- Genmon Logs: See above
- Genmon Version: 1.18.05

Does the system continue to try (or should it)?  My log for wifi shows it didn't reconnect with the router until 08:22.

kjwilliams1974, 2022-02-16:
Guys -

Looking at installing the genmon project - curious if I will see charger status, battery, line voltage, leg voltages on generator side, etc.  I am not real clear on what my 38kw generac  water cooled unit’s controller interface will show me for data points….  Evolution 1 I believe - as the unit is 6 years old….  

Regards,

Jeff Williams
***@***.***
***@***.*** (backup)
1-281-380-1714 (cell)

jgyates, 2022-02-16:
@dieseld23 

I have added retries to the following add ons: gensms, genslack and genpushover. Let me know if you have any issues. There are two new parameters on the add on page for each of the above mentioned add ons that allow you to adjust the duration of the retry and the maximum duration before dropping the message. Let me know if you have any questions,

closed 2022-02-17 by jgyates