jgyates/genmon #676: Long time to update & Genmon does not restart after update.

RoyRomeo, 2022-02-22:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon updates and restarts.

### Actual Behavior

Genmon takes long time to update and will not restart.

### Steps to Reproduce (including precondition)

After each update I have to Putty in and type "./startgenmon.sh -p 2 restart" for it to restart.

### Screenshot or Pictures relating to the problem (if possible)

I saw on one of your previous posts that you recommended the following to a Genmon user:

    ./startgenmon.sh stop
    ./genmonmaint.sh -C
    ./genmonmaint.sh -u -n 
    ./startgenmon.sh restart

Could this work in my case?

### Your Environment

- Generator Model:  G0062440
- Genmon Logs: Submitted.
- Genmon Version: V1.18.07

jgyates, 2022-02-22:
This is a result of one line of code. If I change to to make this work for you then everyone else who tries to upgrade from 1.17.x to 1.18.x will not upgrade to the python 3 version until a reboot.

What is the reason you are not using python 3?

jgyates, 2022-02-22:
your logs state you are running butster (raspbian V10) so there is no reason that i am aware of that python3 should not work for you.

jgyates, 2022-02-22:
What happens if you type this on the command line of your pi:

    sudo apt-get update
    sudo apt-get install python3-pip

If asked for a confirmation type 'y' then enter.

RoyRomeo, 2022-02-22:
I am a novice.  Do you do this from the pi login or cd genmon?

RoyRomeo, 2022-02-22:
I purchased my Genmon from Jared VanFleet (liltux), years ago.

skipfire, 2022-02-22:
@RoyRomeo it doesn't matter what directory you execute apt commands in.

RoyRomeo, 2022-02-22:
This is what happened:
pi@genpi [~] sudo apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Hit:3 https://deb.nodesource.com/node_10.x buster InRelease
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' change                                                                                                                     d its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be ap                                                                                                                     plied. See apt-secure(8) manpage for details.
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed i                                                                                                                     ts 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be ap                                                                                                                     plied. See apt-secure(8) manpage for details.
pi@genpi [~] sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  dh-python libpython3-dev libpython3.7-dev python3-asn1crypto
  python3-cffi-backend python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-lib2to3 python3-secretstorage
  python3-setuptools python3-wheel python3-xdg python3.7-dev
Suggested packages:
  python-crypto-doc python-cryptography-doc python3-cryptography-vectors
  python-dbus-doc python3-dbus-dbg gnome-keyring libkf5wallet-bin
  gir1.2-gnomekeyring-1.0 python-secretstorage-doc python-setuptools-doc
The following NEW packages will be installed:
  dh-python libpython3-dev libpython3.7-dev python3-asn1crypto
  python3-cffi-backend python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-lib2to3 python3-pip python3-secretstorage
  python3-setuptools python3-wheel python3-xdg python3.7-dev
0 upgraded, 21 newly installed, 0 to remove and 0 not upgraded.
Need to get 47.7 MB/49.5 MB of archives.
After this operation, 80.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpython3.7-de                                                                                                                     v armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3.7-dev a                                                                                                                     rmhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.                                                                                                                     7/libpython3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193                                                                                                                      80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.                                                                                                                     7/python3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-mis                                                                                                                     sing?
pi@genpi [~] pi@genpi [~] sudo apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Hit:3 https://deb.nodesource.com/node_10.x buster InRelease
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' change                                                                                                                     d its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be ap                                                                                                                     plied. See apt-secure(8) manpage for details.
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed i                                                                                                                     ts 'Suite' value from 'testing' to 'oldstable'
N: This must be acc-bash: pi@genpi: command not found
epted explicitly before updates for this repository can be ap                                                                                                                     plied. See apt-secure(8) manpage for details.
pi@genpi [~] sudo apt-get install python3-pip
[... cut ...]
pi@genpi [~] E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.                                                                                                                     7/python3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
-bash: E:: command not found
pi@genpi [~] E: Unable to fetch some archives, maybe run apt-get update or try with --fix-mis                                                                                                                     sing?
-bash: E:: command not found

jgyates, 2022-02-22:
Now type this:

    sudo apt-get update --allow-releaseinfo-change

Then type this:
     sudo apt-get update
     sudo apt-get install python3-pip

RoyRomeo, 2022-02-22:
pi@genpi [~] sudo apt-get update
Hit:1 https://deb.nodesource.com/node_10.x buster InRelease
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
pi@genpi [~] sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  dh-python libpython3-dev libpython3.7-dev python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography python3-dbus python3-dev python3-distutils python3-entrypoints python3-gi
  python3-keyring python3-keyrings.alt python3-lib2to3 python3-secretstorage python3-setuptools python3-wheel python3-xdg python3.7-dev
Suggested packages:
  python-crypto-doc python-cryptography-doc python3-cryptography-vectors python-dbus-doc python3-dbus-dbg gnome-keyring libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-secretstorage-doc
  python-setuptools-doc
The following NEW packages will be installed:
  dh-python libpython3-dev libpython3.7-dev python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography python3-dbus python3-dev python3-distutils python3-entrypoints python3-gi
  python3-keyring python3-keyrings.alt python3-lib2to3 python3-pip python3-secretstorage python3-setuptools python3-wheel python3-xdg python3.7-dev
0 upgraded, 21 newly installed, 0 to remove and 0 not upgraded.
Need to get 47.7 MB/49.5 MB of archives.
After this operation, 80.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libpython3.7-dev armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3.7-dev armhf 3.7.3-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/libpython3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/python3.7/python3.7-dev_3.7.3-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

jgyates, 2022-02-22:
I don't see where you ran this command:

     sudo apt-get update --allow-releaseinfo-change

RoyRomeo, 2022-02-22:
genpi [~] sudo apt-get update --allow-releaseinfo-change
Hit:1 https://deb.nodesource.com/node_10.x buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]
Get:7 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.6 MB in 35s (388 kB/s)
Reading package lists... Done
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
pi@genpi [~] sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://deb.nodesource.com/node_10.x buster InRelease
Hit:3 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
pi@genpi [~] sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  dh-python libpython3-dev libpython3.7 libpython3.7-dev libpython3.7-minimal libpython3.7-stdlib python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring python3-keyrings.alt python3-lib2to3 python3-secretstorage python3-setuptools python3-wheel python3-xdg python3.7
  python3.7-dev python3.7-minimal
Suggested packages:
  python-crypto-doc python-cryptography-doc python3-cryptography-vectors python-dbus-doc python3-dbus-dbg gnome-keyring libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-secretstorage-doc
  python-setuptools-doc python3.7-venv python3.7-doc binfmt-support
The following NEW packages will be installed:
  dh-python libpython3-dev libpython3.7-dev python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography python3-dbus python3-dev python3-distutils python3-entrypoints python3-gi
  python3-keyring python3-keyrings.alt python3-lib2to3 python3-pip python3-secretstorage python3-setuptools python3-wheel python3-xdg python3.7-dev
The following packages will be upgraded:
  libpython3.7 libpython3.7-minimal libpython3.7-stdlib python3.7 python3.7-minimal
5 upgraded, 21 newly installed, 0 to remove and 131 not upgraded.
Need to get 53.0 MB/54.8 MB of archives.
After this operation, 80.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian.mirror.constant.com/raspbian buster/main armhf libpython3.7 armhf 3.7.3-2+deb10u3 [1,253 kB]
Get:2 http://raspbian.mirror.constant.com/raspbian buster/main armhf python3.7 armhf 3.7.3-2+deb10u3 [330 kB]
Get:3 http://raspbian.mirror.constant.com/raspbian buster/main armhf libpython3.7-stdlib armhf 3.7.3-2+deb10u3 [1,664 kB]
Get:4 http://raspbian.mirror.constant.com/raspbian buster/main armhf python3.7-minimal armhf 3.7.3-2+deb10u3 [1,464 kB]
Get:5 http://raspbian.mirror.constant.com/raspbian buster/main armhf libpython3.7-minimal armhf 3.7.3-2+deb10u3 [583 kB]
Get:6 http://raspbian.mirror.constant.com/raspbian buster/main armhf libpython3.7-dev armhf 3.7.3-2+deb10u3 [47.2 MB]
Get:7 http://raspbian.mirror.constant.com/raspbian buster/main armhf python3.7-dev armhf 3.7.3-2+deb10u3 [513 kB]
Fetched 53.0 MB in 39s (1,374 kB/s)
Reading changelogs... Done
(Reading database ... 99680 files and directories currently installed.)
Preparing to unpack .../00-libpython3.7_3.7.3-2+deb10u3_armhf.deb ...
Unpacking libpython3.7:armhf (3.7.3-2+deb10u3) over (3.7.3-2+deb10u2) ...
Preparing to unpack .../01-python3.7_3.7.3-2+deb10u3_armhf.deb ...
Unpacking python3.7 (3.7.3-2+deb10u3) over (3.7.3-2+deb10u2) ...
Preparing to unpack .../02-libpython3.7-stdlib_3.7.3-2+deb10u3_armhf.deb ...
Unpacking libpython3.7-stdlib:armhf (3.7.3-2+deb10u3) over (3.7.3-2+deb10u2) ...
Preparing to unpack .../03-python3.7-minimal_3.7.3-2+deb10u3_armhf.deb ...
Unpacking python3.7-minimal (3.7.3-2+deb10u3) over (3.7.3-2+deb10u2) ...
Preparing to unpack .../04-libpython3.7-minimal_3.7.3-2+deb10u3_armhf.deb ...
Unpacking libpython3.7-minimal:armhf (3.7.3-2+deb10u3) over (3.7.3-2+deb10u2) ...
Selecting previously unselected package python3-lib2to3.
Preparing to unpack .../05-python3-lib2to3_3.7.3-1_all.deb ...
Unpacking python3-lib2to3 (3.7.3-1) ...
Selecting previously unselected package python3-distutils.
Preparing to unpack .../06-python3-distutils_3.7.3-1_all.deb ...
Unpacking python3-distutils (3.7.3-1) ...
Selecting previously unselected package dh-python.
Preparing to unpack .../07-dh-python_3.20190308_all.deb ...
Unpacking dh-python (3.20190308) ...
Selecting previously unselected package libpython3.7-dev:armhf.
[... 57 lines cut ...]
Setting up python3-wheel (0.32.3-2) ...
Setting up python3-gi (3.30.4-1) ...
Setting up python3.7-minimal (3.7.3-2+deb10u3) ...
Setting up python3-crypto (2.6.1-9+b1) ...
Setting up python3-lib2to3 (3.7.3-1) ...
Setting up python3-asn1crypto (0.24.0-1) ...
Setting up python3-cffi-backend (1.12.2-1) ...
[... cut ...]
Setting up python3-cryptography (2.6.1-3+deb10u2) ...
Setting up python3-pip (18.1-5+rpt1) ...
Setting up python3-keyrings.alt (3.1.1-1) ...
Setting up python3.7 (3.7.3-2+deb10u3) ...
Setting up libpython3-dev:armhf (3.7.3-1) ...
Setting up python3.7-dev (3.7.3-2+deb10u3) ...
Setting up python3-secretstorage (2.3.1-2) ...
Setting up python3-dev (3.7.3-1) ...
Setting up python3-keyring (17.1.1-1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for mime-support (3.62) ...

jgyates, 2022-02-22:
perfect. Now type this:

     python3 -V

It should show the version number of python. As long as you don't get a 'command not found' error you are doing great.

Next type this:

     cd genmon
     ./startgenmon.sh restart

This should start using python3. You may see some messages to the screen regarding installing libaries for python3 but that is OK. It may take a couple of minute to install everything.

Did you ever change your crontab to start python2  (-p 2) on startup? If you did then let me know and we can fix that as well.

RoyRomeo, 2022-02-22:
Genmon is working.  How can I tell if the crontab was changed?

jgyates, 2022-02-22:
type this:

sudo crontab -l

it should show this:

@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start

if it shows this:

@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh -p 2 start

or some variant with "-p 2" in it then we need to change it.

RoyRomeo, 2022-02-22:
i@genpi [~] sudo crontab -l
@reboot sleep 30 && /bin/bash /home/pi/genmon/startgenmon.sh start
*/5 * * * * /bin/bash /home/pi/WiFi_Check.sh

jgyates, 2022-02-22:
ok. you should be set. Try to click the upgrade button on the About page and see if it works as expected now.

RoyRomeo, 2022-02-22:
Works like a champ!  Thanks very much for all your help and assistance.

closed 2022-02-22 by jgyates