jgyates/genmon #677: genmon updates are incredibly easy and drama free (informational)

aa777888-2, 2022-02-22:
There seems to be no other place for this sort of thing so...I just wanted to thank the author for his incredible skill and generosity. I've only been a part of the genmon community for a short time, but my genmon installation has already been through three updates and each one has been a completely trivial, trouble-free exercise. Thank you for what you do!

jgyates, 2022-02-22:
Thanks for the kind words.

CHawks78, 2022-02-23:
I too would like to express my thanks to jgyates.  I'd never even heard of a raspberry pi before I stumbled across this project.  The wiki tutorial was all I needed to get my genmon running great.  Awesome software development, and support....I did look over some "closed" issues to help with some of my questions along the way.

DC072012, 2022-02-24:
> I too would like to express my thanks to jgyates. I'd never even heard of a raspberry pi before I stumbled across this project. The wiki tutorial was all I needed to get my genmon running great. [...]

If you haven’t already and are inclined to do so, you can support the project at this link:
https://github.com/jgyates/genmon

Towards the middle of the page there is a donate button.  

I am not involved in this other than being a happy customer and having previously donated.

jgyates, 2022-02-25:
Thanks for the feedback everyone!

closed 2022-02-25 by jgyates