jgyates/genmon #678: Enhanced exercise dispaly problem

UPS99, 2022-02-24:
FYI only - a small settings oddity - I noticed today that the Enhanced Exercise function was showing "off", despite the fact that it is actually enabled and has been working OK - running for the 10 minutes I have set.  Selecting "on" prompted with those previous settings still filled out.  Genmon then restarted and appears normal, and Enhanced Exercise now shows "on".  I don't know when this changed - just happened to notice it today.

jgyates, 2022-02-25:
Thanks, I will keep an eye out for the cause of this. At the moment it is not obvious to me what the issue may be.  I am going to close this for now as I don't have any actions I can take, but I will update this if I have a resolution.

closed 2022-02-25 by jgyates

UPS99, 2022-02-25:
That’s fine.  I just wanted to provide this limited info in case it would be useful.  Unfortunately I didn’t notice which update it followed.