jgyates/genmon #679: No Power Graph info using PintSize.me HAT on a Nexus controller

smeisner, 2022-02-24:
In a nutshell, after adding a Pintsize.me HAT to my Nexus controlled generator, I expected to see power output in the KW Output gauge and the Power Output graph. I am seeing data in the External Current and External Power gauges, but I really wanted the historical data in the graph.

Since I have a Nexus controller, I initially did not change the currentoffset or the currentdivider, but I did try setting currentdivider to 1, but that did nothing. After scanning the source code, I believe I understand why; this is only used for non-Nexus controllers. Also while reading the code, I think the collected values will only be displayed in the “External …” gauges, not the KW Output graph, which is what I really want.

Is there any way to get this info to the graph on the front screen to show historical data? Or am I misunderstanding how the code works?

### Expected Behavior

Data would be shown in KW Output graph and Power Output gauge on dashboard.

### Actual Behavior

Both appear empty (no data), but External Current and External Power show data.

### Steps to Reproduce (including precondition)

Installed Pintsize.me HAT to add 2 CTs to the Nexus controlled setup. After modifying the /boot/config.txt file as directed, I then started the Pi and via the Genmon Add-Ons screen, added the External Current Transformer (CT) Sensors. I then rebooted and started the generator.

I did notice issue #168 is very similar, but with a different genset and different controller. So the code used will be very different.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0060331
- Genmon Logs: Submit via the About page .. done. Also sent the registers.
- Genmon Version: V1.18.07

jgyates, 2022-02-24:
I have not received any registers or logs today. You may want to check that outbound email is working. I have tested this and it is working for me when I simulate a Nexus controller. Having your logs would help a lot.

jgyates, 2022-02-24:
also, have you tried to refresh your browser since enabling the CT HAT plug in? A refresh is needed to get the updated list of gauges / graphs.

jgyates, 2022-02-24:
also check that "Disable Power / Current Display" on the settings page is not enabled.

smeisner, 2022-02-24:
> I have not received any registers or logs today. You may want to check that outbound email is working. I have tested this and it is working for me when I simulate a Nexus controller. Having your logs would help a lot.

Can you check again? All the test emails worked before sending the registers & logs, so I just changed to gmail for my SMTP server.

smeisner, 2022-02-24:
> 

yes, I did check that already.

[image]

smeisner, 2022-02-24:
> also, have you tried to refresh your browser since enabling the CT HAT plug in? A refresh is needed to get the updated list of gauges / graphs.

Yep ... many times!  :D

skipfire, 2022-02-24:
This may or may not be related, but in issue #645 there is a picture from dennis48755 showing it as well, it has a displayed voltage of 0 which appears to result in a kw of 0.  I don't know if there is a voltage number used somewhere behind the scenes.
https://github.com/jgyates/genmon/issues/645#issuecomment-1025015655

jgyates, 2022-02-24:
No log files yet.

you can download your logs with this url

http://192.168.1.10:8000/cmd/get_logs

And copy and paste your registers from this link:

http://192.168.1.10:8000/cmd/registers_json

into an email and send it to generatormonitor.software@gmail.com

jgyates, 2022-02-24:
Just sending your logs files via the web interface 

OR

this into in an email:

http://192.168.11.22:8000/cmd/support_data_json

and then attached archive file from 

http://192.168.1.10:8000/cmd/get_logs

an email and send it to [generatormonitor.software@gmail.com](mailto:generatormonitor.software@gmail.com)

The above two are almost equivalent. The web interface also sends comm stats but I don't really need that.

smeisner, 2022-02-24:
Done! Just sent the email. The registers are without the generator running. Is that OK?

jgyates, 2022-02-24:
that is fine.

jgyates, 2022-02-24:
your outbound email is not working due to this error:

Error SMTP sendmail: (534, b'5.7.9 Application-specific password required. Learn more at\n5.7.9  https://support.google.com/mail/?p=InvalidSecondFactor 

You need to setup an app specific password which lowers the security of the account. I recommend creating a dedicated email to only be used for sending so lowering the security will not impede your normal email account.

Still looking at the main issue....

smeisner, 2022-02-24:
Yeah, when I said earlier "I switched to Gmail" this is what I had just completed. The account I am using is almost a "throw away" account, so I don't care too much.

Earlier I was using the SMTP server I am running locally that does send test emails no problem. When I tried to submit the diag info, it was complaining about an improperly formed address.

jgyates, 2022-02-24:
Send me the output of this url

http://192.168.1.10:8000/cmd/start_info_json

also, can you show the output of /var/log/gencthat.log

I updated the script on the repository a day or so ago to include this update so you don't have that update. Alternatively you can update the repository on the about page and then send a new set of logs with the archive from this command.

http://192.168.1.10:8000/cmd/get_logs

smeisner, 2022-02-24:
ok, output from /cmd/start_info_json sent.

Here;s the gencthat.log:

> root@generator:/var/log# cat gencthat.log
> 2022-02-18 10:30:42,743 : Copied /home/pi/genmon/conf/gencthat.conf to /etc/genmon/gencthat.conf
> 2022-02-18 10:43:21,604 : Error: Connect : [Errno 111] Connection refused : myclient.py:47
> 2022-02-18 10:59:40,680 : Error: RX:[Errno 104] Connection reset by peer : myclient.py:78
> 2022-02-18 10:59:49,698 : Error: Connect : [Errno 111] Connection refused : myclient.py:47
> 

...definitely not what I expected!

I will do an update and send the logs from /get_logs next.

jgyates, 2022-02-24:
OK. I have looked at all the data you sent, here is what I see:

genmon looks like it is reporting back to the web interface that the power graph is supported.

search /cmd/start_info_json  for powergraph, you should see two instances 

What browser are you using? Can you duplicate this on another computer / browser?

If you are using Chrome can you go to the main menu then to "More Tools - > Developer Tools" . This will bring up a broswer developer tools window. In the menu for the developer tool window click on "Console". Then refresh the web page to genmon. Do you see any errors?

You should see something like this:
[image]

jgyates, 2022-02-25:
I checked in a small update (the versions number did not change). I do not think this will help your problem, but there may be a a small chance that it does. This change allows the power graph to show up if the controller does not define one by default and the CT add on is enabled. For Nexus controllers this means that it will try to be defined a 2nd time if it is not detected the first time. I don't think this will solve your problem because based on the output mentioned above the power graph is being detected and enabled by genmon but not being displayed by the web app (browser). This should help anyone with a custom controller without a modbus register to define power.

jgyates, 2022-02-25:
if you do update to the latest send me another output of /cmd/start_info_json if you can

smeisner, 2022-02-25:
I already did the update and sent the info to the email address you mentioned (this was ~30 mins ago).

So far, I was using Chrome on Windows. I tried a completely different computer (Linux) running a different browser (Firefox) and I got the exact same result.

While running on gen power, I did search the output of  /cmd/start_info_json for powergraph and only found 1 instance. 

Going back to my Windows / Chrome setup, I went into dev tools and refreshed and got this:

> Mixed Content: The page at 'https://generator.meisners.net/' was loaded over HTTPS, but requested an insecure favicon 'http://generac.com/favicon.ico'. This request has been blocked; the content must be served over HTTPS.

This error is because I used Generac's favicon for my Genmon site. Please don't tell me this could cause a problem!!!  :D

smeisner, 2022-02-25:
Let me be clear; the output from /cmd/start_info_json had the following instances of the string "powergraph":

> "PowerGraph": true,
>  "subtype": "powergraph", 

Is that what you meant by 2 instances?

jgyates, 2022-02-25:
yes, that is what I mean by two instances.

     "While running on gen power, I did search the output of /cmd/start_info_json for powergraph and only found 1 instance."

IF you are not seeing two instances of powergraph (i.e. one powergraph and one PowerGraph) then send me the output of /cmd/start_info_json for this along with your registers when it is running.

smeisner, 2022-02-25:
Just sent the output from /cmd/start_info_json and on generator power. Also here's screenshot of genmon after a few minutes on gen power:

[image]

jgyates, 2022-02-25:
your output while running contains two powergraph instances.
I tried this on windows chrome and it is working for me. Let me think about what else to try.

jgyates, 2022-02-25:
ok. so having the software fetch the favicon from a http site while using https will make the power graph fail? that is strange, but good to know.

smeisner, 2022-02-25:
> ok. so having the software fetch the favicon from a http site while using https will make the power graph fail? that is strange, but good to know.

no..no..I removed the favicon and it still acts the same way. the only diff was in the Google Chrome console I am not receiving the "Mixed Content: " error message i posted before. I was hopeful it would make a differ with Genmon, but it didn't.

jgyates, 2022-02-25:
I have been thinking that your issue was the graph was not visible, but after looking at your picture it appears your problem is the graph is visible but not showing any data, correct? This picture does clarify that for me. That is a completely different issue so I will start over on this.

jgyates, 2022-02-25:
What is your settings for the "Strict Usage of CT data" parameter on the CT Hat add on?

Also, where physically are the CTs connected? Are they monitoring the two legs directly from the generator or are they after the transfer switch?

smeisner, 2022-02-25:
> What is your settings for the "Strict Usage of CT data" parameter on the CT Hat add on?

[image]

> 
> Also, where physically are the CTs connected? Are they monitoring the two legs directly from the generator or are they after the transfer switch?

Two legs directly from the generator. Actually, inside the generator compartment. On my unit, the two red wires that exit the generator enclosure.

smeisner, 2022-02-25:
> I have been thinking that your issue was the graph was not visible, but after looking at your picture it appears your problem is the graph is visible but not showing any data, correct? 

Correct.

jgyates, 2022-02-25:
all you need to do is to set "Strict Use of CT data" to Off. This setting will force the software to only use the CT data in a calculation (i.e. power) if the system is in an outage. 

If the CTs were after the transfer switch then this setting would need to be enabled as the utility current would be reported if the transfer switch was not activated.

Let me know how it goes.

smeisner, 2022-02-25:
cripes...well, I feel like an idiot for wasting your time! Thank you for your help. Was this documented somewhere?

[image]

jgyates, 2022-02-25:
No worries, I should have read your original email more carefully. 

The software [overview page](https://github.com/jgyates/genmon/wiki/1----Software-Overview#gencthatpy-optional) has a section on this add on and the pop up bubble for the setting contains some info too. I also just checked in a change that will make new installations default to off for this setting.

Anytime a new feature is rolled out there is typically a learning curve on how to structure the documentation and approach the support so thanks for helping me figure out the weak spots.

closed 2022-02-25 by jgyates