jgyates/genmon #681: GPIO Output Add-on not working

sambeetm, 2022-03-02:
Hi..Is the GPIO Output add-on broken ? Its not working anymore. Not sure since when though. It only lights-up READY Led and does not change with other statuses (like exercising, running, off etc). It used to work perfectly. I looked into the processes and /home/pi/genmon/gengpio.py is up and running. In the /var/log/gengpio.log is 0 byte. Any other way to debug ?

I tested all the GPIO pins triggering low-high from PI(using a simple .py program) and all are responding. So, no wiring issues.

Thanks

jgyates, 2022-03-03:
Hi @sambeetm 

Apparently yes. Thanks for pointing this out. I checked in a fix (version number did not change) so just update your software from the about page. Let me know if you have any issues.

Thanks

sambeetm, 2022-03-03:
Thanks for the fix. Working as expected now.....!!

jgyates, 2022-03-03:
Nice setup with your pi and LEDs

closed 2022-03-03 by jgyates

sambeetm, 2022-03-04:
@jgyates what is the condition for "Internet Connection GPIO 3" to stay high ? After a day of working fine, this light is turning on. I checked  1) genserv is up and accessible from outside 2) PI can access internet.

jgyates, 2022-03-04:
The software attempts to make a HTTP connection to google.

https://github.com/jgyates/genmon/blob/768d5e4c01a01bffeb88bfe080b0463dfc55cbc1/genmonlib/myplatform.py#L424

jgyates, 2022-03-04:
I found one more bug that probably effects this. I checked in the change so just update from the about page. Sorry about that.

sambeetm, 2022-03-04:
Thanks !! its gone for now. I will monitor for few days and let you know if there are any more findings...