jgyates/genmon #682: Google Gmail Less Secure Apps Access Going Away 5/30/2022

alphaa211, 2022-03-04:
Any chance there will be a Genmon workaround to still be able to easily use a gmail address for notifications?  As soon as I turn off the less secure apps setiing Genmon can no longer send me notifications via my Gmail account.  Gmail being one of the most popular out there, I thought I'd ask.  Thanks!  

"To help keep your account secure, starting May 30, 2022, ​​Google will no longer support the use of third-party apps or devices which ask you to sign in to your Google Account using only your username and password."

https://support.google.com/accounts/answer/6010255?rfn=1646361360139&anexp=nret-fa&pli=1

jgyates, 2022-03-04:
I got an email on that this morning as well. I will put this on the priority list to add Oauth support.

jgyates, 2022-03-05:
After reviewing this issue, the thing to do is to enable 2 factor authentication (2FA) for your google account, then use an "App Password":

https://support.google.com/accounts/answer/185833?hl=en

An App Password is a password generated by google (Microsoft and Apple has the same options) than can be used by applications like genmon that does not support Oauth or 2FA.

The problem with using Oauth for authentication in genmon is that it requires a secure client ID and google does not recommend storing a secure client ID in open source software. There are other ways for the user to generate keys and use the GMail API instead of SMTP and an app password, but those are much more complicated for the end users than just using two factors authentication and an app password (it involves using the google developer console and is not a very straight froward process).

This is something that everyone using gmail with genmon will need to address by May 30, 2022.

jgyates, 2022-03-06:
FYI, this thread has instruction on enabling App Passwords for gmail

https://www.reddit.com/r/GMail/comments/t6hhfj/you_may_lose_access_to_some_of_your_thirdparty/

I am using this now on my production and test system and can validate it works well.

liltux, 2022-03-22:
I just saw this.  This is exactly what I do and is easy and works.  I have a single gmail account for all of my little IOT projects and use the same account just different App Password.  Also provides a history of when the device last accessed the account.  Works like a charm.

grsthegreat, 2022-06-09:
hmmm, i set up 2 factor authorization on my gmail accounts, and my computer is able to send test messages on all accounts. however last Wednesday my generator exercised but i received no messages on email.

grsthegreat, 2022-06-10:
I just re-entered verification password, and now it sent test email. So i guess i just mis typed it originally. Seems all good now.

alphaa211, 2022-06-10:
I have had repeated issues with similar failures to send actual notifications via email but Genmon test emails work fine.  The only solution I have found thus far is a reboot of the pi.  Now I do it every 2 weeks when I check the oil.  It started several weeks after I enabled the gmail 2 FA and App Password.

jgyates, 2022-06-11:
your /var/log/mymail.log file should have specific errors. 

Just a guess, if your pi is using DHCP and your router may of been restarted then it is possible that your router is blocking any access from the pi if it is using an IP that is has not been allocated by the router since it's reboot.

RoyRomeo, 2022-06-22:
I enabled 2 FA and App password.  When I "Test Email Settings" from the Genmon settings page, the email goes through, however I no longer receive any notifications from Genmon.  Any help will be greatly appreciated.  I submitted logs, not sure if they are of any value for this situation.

grsthegreat, 2022-06-22:
> I enabled 2 FA and App password. When I "Test Email Settings" from the Genmon settings page, the email goes through, however I no longer receive any notifications from Genmon. Any help will be greatly appreciated. I submitted logs, not sure if they are of any value for this situation.

Did you change password in genmon to match the new secure password

RoyRomeo, 2022-06-22:
That's pretty much what I did, but when I turned off the generator I did not get a notification.

jgyates, 2022-06-22:
If you update your email password on the settings page and click test, it will test the new password, however if you do not click the save button at the bottom of the settings page then genmon will not start using the new password.

RoyRomeo, 2022-06-22:
I did click the save button at the bottom of the settings page.

jgyates, 2022-06-22:
@RoyRomeo I have not received any logs. Log sent form the About page are sent via email so if you email is not working I will not get your logs. 

The first thing I would suggest is that you try validating all of your information you have for your email settings. If you log into your pi you can view your settings in this file:   /etc/genmon/mymail.conf 

If your password in this file does not match your account password it will not work.

To view your logs you can log into your pi via ssh and type this command to view the log:

        cat /var/log/mymail.log

You can copy and paste the log data to this thread.

Or, for another way to get your logs to me, you can issue this command in a browser to download all the logs to your computer.

        https://192.168.1.10:8000/cmd/get_logs

NOTE: change the pi address in the above command to match the IP address of your pi. This will download your log files in a compressed archive file named genmon_logs.tar.gz. You can attach this file to this thread on the github web site. If you try to send it as an email attachment it will get stripped off by github.

RoyRomeo, 2022-06-22:
@jgyates When I log into my pi, I'm in the genpi directory.  When I type in file: /etc/genmon/mymail.conf, I get command file not found.

skipfire, 2022-06-22:
Use `cat /etc/genmon/mymail.conf` to have it print the file content.

jgyates, 2022-06-22:
This is not a command it is a file path. To view the file type this command followed by the file path:

     cat /etc/genmon/mymail.conf

RoyRomeo, 2022-06-22:
@jgyates The app password is listed correctly.

RoyRomeo, 2022-06-22:
@jgyates 

[MyMail]

# optionally disable email use by the program. set to true to disable all email functionality
disableemail = false

# optionally disable receiving mail
disableimap = true

# optionally disable sending mail
disablesmtp = false

# The password for the email account used for sending and receiving email
email_pw = lbnnqjclwiuivfdy

# the email account
email_account = Joes.Generator@gmail.com

# address that email are sent to
email_recipient = 2155195441@messaging.sprintpcs.com,Joe.Pryor@Verizon.net

# the email from which the emails are sent (sender email address)
sender_account = Joes.Generator@gmail.com

# optional, name of the sender, i.e. Genmon or John Doe
sender_name = Joe's Generator

# the SMTP mail server used for sending email, leave empty to disable SMTP, i.e. "smtp_server ="
smtp_server = smtp.gmail.com

# the IMAP server address used for receiving email, leave empty to disable IMAP, i.e. "imap_server ="
imap_server = imap.gmail.com

# the SMTP port used on the SMTP server for sending email
# Note about ports and email security: typically port 587 is used for
#  TLS and port 465 is used for SSL
smtp_port =

# use SSL encryption to send emails
ssl_enabled = false

# do not use TLS encryption if true, the default value is false (use TLS).
# This is parameter ignored if ssl_enable is true. Enabling this and disabling
# SSL encryption will result in no SMTP encryption.
tls_disable = false

# use BCC instead of To for sending mail
usebcc = False
smtpauth_disable = false
pi@genpi [~] cat /var/log/mymail.log
2022-06-22 10:01:37,133 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:05:48,013 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:05:48,015 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:09:58,893 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:09:58,894 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:14:09,775 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:14:09,779 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:18:20,653 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:18:20,654 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:22:31,533 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:22:31,534 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:26:42,413 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:26:42,414 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:30:53,293 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:30:53,294 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:35:04,173 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:35:04,175 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 10:39:15,055 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 10:39:15,056 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
[... 222 lines cut ...]
2022-06-22 18:09:53,773 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:09:53,774 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:14:04,653 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:14:04,654 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:18:15,533 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:18:15,534 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:22:26,413 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:22:26,414 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:26:37,293 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:26:37,294 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:30:48,172 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:30:48,173 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:34:59,053 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:34:59,054 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:39:09,932 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:39:09,933 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:43:20,813 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:43:20,814 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:47:31,703 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:47:31,703 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:51:42,573 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:51:42,574 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2022-06-22 18:55:53,453 : Error SMTP Init : SSL:<False>: [Errno 101] Network is unreachable : mymail.py:501
2022-06-22 18:55:53,454 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
pi@g

jgyates, 2022-06-22:
@RoyRomeo 

you smtp port is blank. It should be 587 for gmail.

if that does not work, then reboot your pi. You appear to be having network issues with your pi, but that could be due to the port being unspecified. 

I edited the above to remove your email password. You posted the contents of /etc/genmon/mymail.conf and you did not remove your password. You may consider changing your password as this thread gets emailed out to a lot of people. It is just an app specific password so it is probably only good for reading and sending email on that account.

RoyRomeo, 2022-06-22:
@jgyates 

I had 587 in the smtp server port.  I changed it to 465 and switched on "Use SSL Encryption" and now it appears to work.  Does that make sense?

RoyRomeo, 2022-06-23:
@jgyates 

I changed smtp server port to 587, switched off "Use SSL Encryption" and it does appear to be working.

Thanks very much for your help.

closed 2022-09-11 by jgyates