jgyates/genmon #683: Not Receiving Data generates an SMS (from carrier email addin) message (but not an email)

danorum, 2022-03-09:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I would expect to get the same SMS messages that I get in email when I have the same config selected. While I am troubleshooting the issues I am having with my serial connection, it would be great to not be getting all these text messages.

### Actual Behavior

I don't get an email for "Not Receiving Data" and OK, but I do get an SMS  

### Steps to Reproduce (including precondition)

Use Mobile Carrier Email to SMS with Errors, Warning, Information and Outages selected
Email set to Outage Error Warn, Info 

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 22kW Evolution
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.08

jgyates, 2022-03-10:
Thanks for the feedback. The short messages used in all of the add on modules works differently than the email messages that are built into genmon.py. 

The short message used for communications just monitors the state of the Monitor-> Generator Monitor Stats -> Monitor Health and if the state changes it sends a short message notice. This "Monitor Health" will show a few different outputs depending on the state of the generator monitor software. It also always sends a short message for the initial state when the system is first loaded and additional messages each time the state changes.

Email messages are implemented differently and are more an integrated part of genmon as they are sent a very specific events. Email notices for communications failure are sent but after a period of time, that period of time varies based on the type of communication used (serial or serial over TCP) and the type of controller. These email messages have the title "Generator Monitor Communication WARNING at <SITENAME>".

A short message about system health is always sent when a given add on is started. If your comms are working the system health is "OK"

If your comms are not working when genmon is started you will see and immediate short message about the comms not working, but in a few minutes you will see and email stating the comms problem as well. The email sent only sent after the initialization sequence of genmon is completed. The SMS is sending the default state of the comm status, which is comms are not active before the initialization  sequence is complete. 

Short version: if you wait a min or two you should see an email about failed comms within a few minutes, but the short messages send a current snapshot of the communications status on start up with no period of waiting. 

Let me know if you still think there is an issue. I am going to close this issue but let me know if you have questions.

closed 2022-03-10 by jgyates

danorum, 2022-03-10:
Thank you for the really detailed answer to my short question. I thought the add-in just picked up messages with the same "types". I had both notifications and the sms email add-in with the same settings. I am currently troubleshooting why my genmon won't work with my new generator and in addition to pulling my hair out with it, I was getting tons of texts and couldn't figure out why the texts didn't match the SMS messages. With the timeouts, my genmon would go to "Not receiving communication" and then go back to OK. I would get a flood of these texts.

jgyates, 2022-03-10:
Is your communications still intermittent? or is this just on startup that you are seeing issues? The best judge of your communications status is on the Monitor page of the web interface. The logs you submitted has this for communications status:

Comm Stats : 
   Packet Count : M: 1866, S: 1815
   CRC Errors : 0 
   CRC Percent Errors : 0.00%
   Timeout Errors : 50
   Timeout Percent Errors : 2.68%
   Modbus Exceptions : 3
   Validation Errors : 0
   Sync Errors : 0
   Invalid Data : 0
   Discarded Bytes : 0
   Comm Restarts : 0
   Packets Per Second : 13.01
   Average Transaction Time : 0.0367 sec

The 50 timeout errors are not a show stopper but not normal. the modbus exceptions for Evolution 2.0 are expected due to the encryption of the the serial comms. Are your timeout errors continuing to increase?

jgyates, 2022-03-10:
as a point of reference, here is an example of normal comms on an Evolution 2.0

Comm Stats : 
   Packet Count : M: 1312590, S: 1312589
   CRC Errors : 0 
   CRC Percent Errors : 0.00%
   Timeout Errors : 0
   Timeout Percent Errors : 0.00%
   Modbus Exceptions : 86
   Validation Errors : 0
   Sync Errors : 0
   Invalid Data : 0
   Discarded Bytes : 0
   Comm Restarts : 0
   Packets Per Second : 63.28
   Average Transaction Time : 0.0310 sec

danorum, 2022-03-10:
Yes, still having tons of timeouts. First test was to add a ground wire to the molex plug. The monitor was working rock solid on my old 11 year old generac, but there were so many timeouts that it took forever to come up and then dropped in and out. Unfortunately, when I was going to add the ground from the generator to the RS232 ground on my ttl-232 chip, a pad for one of the other wires came loose. I grabbed a backup genmon setup I had and added the ground from pin 4 and continued to get timout errors. I have one last ttl<->rs232 chip and am going to fix my first genmon and try again. It may be a bit before I can sit down and redo this. I expected to just plug into the new generator and be off and running, and that was not the case.

jgyates, 2022-03-10:
ok. Keep me posted. This is the first generator I have come across with the 1.13 firmware so I am wanting to make sure this is not related to the firmware. Just to confirm the timeouts only started appearing on the new generator?

danorum, 2022-03-10:
Yes. Genmon was rock solid on the old one. Even if it is just for test, I may pick up one of those HATs, even if I go back to my zero custom build.

jgyates, 2022-03-10:
I just checked in a minor update that will add a little more detail to the modbus error log. When you have a chance update your software on the About page then let it run for a few min then send me the logs on the about page. This change adds the tx packet count to the error log for timeouts. This will let me know if the timeouts have any pattern to them or if they are intermittent.

jgyates, 2022-03-10:
I have no idea if this is related in any way, but , a new version of the firmware for Evolution 2.0 was released (v1.14), I am not sure if it starting to roll out for over the air updates or not.

jgyates, 2022-03-11:
FYI, I searched through some of the other error logs that I have received and it appears that I have had several Evo2 firmware V1.13 logs submitted and I don't see an issue with serial comms in these logs. I have gotten V1.13 logs as far back as December 28, 2021. From my view it does not look like this is a comm problem specific to the all Evo2 models with V1.13 firmware. Let me know if you have any other questions to help resolve this issue.

danorum, 2022-03-26:
Wanted to follow up. It is something in my cable with the serial connection. I set up a Pi 3 with a HAT and it started working right away. I am going to redo with a different serial converter board I have and make sure it is all wired up right and try again with my zero. If I am powering from the battery, I shouldn't need a ground, right? If I decided to add a ground wire anyway, should it be pin 4 or 2?

skipfire, 2022-03-26:
@danorum on the molex pin 2, some controllers do not seem to have pin 4 tied to ground.  Just make sure you don't get the pins backwards left to right or you will hit the 12v.  You are correct that if you are tied to ground on the battery that it should not be needed, but some power adapters can pin the high-side (5v on the USB is 12v on the battery, ground on the USB would be 7v on the battery), in that case you aren't getting ground from the battery.

danorum, 2022-03-26:
This is very odd. The loopback test at the molex works great and this is the same setup I had on the last generator. I didn't connect ground yet, but I am getting about 50% timeout errors. I am going to order pin headers so I can test the zero with a pHAT. This worked reliably for multiple years so I am pulling my hair out for sure. I have at least confirmed that genmon should work with my generator. The HAT I tested with using the Pi 3 had no timeout errors.

danorum, 2022-03-26:
One thing I noticed is that the wiring diagrams all show connecting the power on the TTL side of the TTL/RS232 to 5V, but I had mine connected to 3.3V. Based on the text below on the making a cable page, 3.3V may be the best. Would this possibly be causing my timeout errors with the controller on the new generator?