jgyates/genmon #684: No graph UI when you go to the home screen...  Help

kd5bwp, 2022-03-12:
GenMon will not show graph...  New to this system...  Power cycled the device tried checking Disable power/ current display setting was working then stopped...

Console shows this.
[image]

[image]

[image]

[image]

[image]

jgyates, 2022-03-13:
There is a lot of information missing so I can not really provide much but some guesses. 

What model generator?
Is genmon communicating with the controller? Add ons will not function if communications are not active.
Log files were not submitted.

The errors on the console you pictured above appear to be from a chrome extension, not from genmon.

If your serial comms are not working then it would be expected that the graph would not show up. The gauges will not show not show up if genmon has not finished initializing, which will take a lot longer if serial comms are not working. 

check the Monitor page, you should see something like this 
[image]

Also, when you say "New to this system" I do not know what you mean. Are you saying that you are new to the raspberry pi or are you saying that genmon is newly installed on this system.

kd5bwp, 2022-03-13:
Thank you for the fast response I'm new to this Genmon setup.  The graphs came back when I plugged in the phat to the evo controller.  I was just setting up the device and see that I had graphs and then they disappeared and when I first plugged it into the gen the graphs on the home page didn't show up for a bit.  But now they do.  
The model of the generator is a RD02023 with Firmware 1.  It's a 20kw Diesel.

The only other question is how does the fuel gauge work?  Is this just a estimate?  It does have a fuel gauge and over limit alarm with low cut off built into the unit but currently only have a little under a half and not full.

[image]

On the email I tired a few setting and wasn't able to get my email to work...  Tried a few combinations with ssl on and off.

[image]

jgyates, 2022-03-13:
There is a fuel estimate gauge however for Evolution Diesel units there is a fuel gauge sensor built in. To enable it turn on the "Use Sensor for Fuel Gauge" on the settings page (this option only shows up on Evolution Liquid cooled models with diesel fuel).

Regarding you email question, did you install via the genmonmaint.sh script? The "WRONG_VERSION_NUMBER" indicates you may have an issue with your installation.

For email username and passwords, it depends on the type of email account you are using (gmail, Microsoft, etc). This section in the wiki has info related to email setup: https://github.com/jgyates/genmon/wiki/1----Software-Overview#configuring-mymailpy

All of the settings can be controlled by the web interface, however you either need to have either a "app specific" password or use "less secure app" settings with your email provider. The link above has more details on this.

Let me know if you have additional questions.

jgyates, 2022-03-13:
Also, you show the CT Hat add on in your above pictures. Do you have the CT Hat installed?

kd5bwp, 2022-03-13:
Yes I have the CT hat installed.

jgyates, 2022-03-13:
Did you install via the genmonmaint.sh script?

kd5bwp, 2022-03-13:
Not sure. Can I run it?

jgyates, 2022-03-13:
I am a little concerned about your last answer. How did you come to have genmon installed on this raspberry pi? The answer to this is dependent on what we do next. If you are not sure then I would suggest that you start over with the instructions on this page:

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

jgyates, 2022-03-13:
in addition, there is an additional step described on this link:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gencthatpy-optional

that is needed for setting up the CT Hat. If you have not modified the file /boot/config.txt as described then the CT Hat will not work. This would need to be done after the install process.

closed 2022-03-15 by jgyates