jgyates/genmon #685: Trouble installing chump

tmorse305, 2022-03-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I used sudo pip3 install chump

### Actual Behavior

and got this response:
pi@raspberrypi:/ $ sudo pip3 install chump
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: chump in /usr/local/lib/python3.9/dist-packages (1.6.0)
pi@raspberrypi:/ $
Is there another step?  I can't file the .conf file so I suspect it really didn't install

### Steps to Reproduce (including precondition)

NA
### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.09

jgyates, 2022-03-13:
The chump library is installed by genmonmaint.sh with the '-i' (install) option. The message you are receiving "Requirement already satisfied" means that the library is already installed. If you want to use the pushover add on you should just enable it on the add on page and it should work if you installed with the genmonmaintsh script, assuming your pushover credentials are good. If you are still having issues let me know.

jgyates, 2022-03-13:
The conf file for genpushover is in /etc/genmon/genpushover.conf

closed 2022-03-13 by jgyates

tmorse305, 2022-03-13:
Thank you jgyates.  I searched for the .conf file but didn't find it, must not have searched that file tree as it was right where you said it would be.  It had never been personalized for my pushover.  No clue when I installed it, don't get old.... :)
Thanks again for a great product and great support!