jgyates/genmon #686: Genmon not showing 3-phase utility and generator voltage information correctly

efrenmartin, 2022-03-14:
Great monitoring system! Thank you so much. Already a supporter!

### Expected Behavior
I expect Genmon to be aware of my gen being 3-phased 220/380V and correctly display (utility and gen) voltage for each phase or some consistent with utility voltage information.

### Actual Behavior
For some reason Genmon show model as unknown and does not show 3-phase information correctly. It seems to show the addition of 2 single phase voltages (220+220=440). Hint: Under Maintenance, Model shows as unknown.

### Steps to Reproduce (including precondition)
I may be hard to reproduce, due to my country’s voltage. 

### Your Environment
I own a 3-phase 220/380V 20kW 50hz air-cooled Generac generator with an Evolution controller.  Genmon is running on a preloaded unit I purchased from you 2 months ago. The generator is installed at home in Buenos Aires, Argentina.

- Generator Model: G0072890
- Genmon Logs: submitted 3/16/22 16:35 approx
- Genmon Version: v1.18.09
[image]
[image]
[image]
[image]
[image]
[image]

efrenmartin, 2022-03-14:
Added info on Genmon with gen is runnin
[image]
g

jgyates, 2022-03-14:
Hi @efrenmartin 

You can change the nominal voltage of the utility voltage and output voltage gauges by going to the advanced page (double click the gear icon in the upper right corner of the web interface) and then set "Use Nominal Volts Override" to On and set "Override nominal line voltage in UI" to the voltage desired (something 440 would do well in your case. Making this change will allow the two voltage gauges to scale to the larger voltage output you specify.

An attempt to detect the model number is only made when the software version changes and it requries the internet to look up the model number. To force a check you can edit the /etc/genmon/genmon.conf file and remove the lines that define "model" and "version".

      sudo nano /etc/genmon/genmon.conf

Go to the bottom of the file, you should see something like these lines in the last 10 or so lines:

      version = V1.18.09
      model = Unknown

Delete each line then save the file by typing Ctrl+X, then Y then Enter. Then restart genmon by typing:

       cd genmon
      ./startgenmon.sh restart

I validated with the registers you sent that the model lookup works with the serial number in your registers using my simulator.

Regarding fully supporting 3 phase, this issue is two fold. 1) I don't have enough data to know if there are additional modbus registers that could be beneficial in displaying voltage info on each line for 3 phase Evolution controllers. and 2) it is possible that this type of data does not exist. The software is limited by what info Generac exports via modbus.

On industrial generators there are three modbus registers for each line voltage and one for the total voltage output. I only now of one modbus register to read for the utility voltage and one for the output voltage.  If you are willing to perform some tests it is possible we could find these additional registers. Of course there is also a chance that we will find that the info we want is not exported by the controller over modbus.

Let me know if you have any additional questions or if you want to perform some tests to try to determine the additional modbus registers. These tests will requires enabling some debugging features in genmon and then running your generator.

efrenmartin, 2022-03-15:
> Hi @efrenmartin
>
> You can change the nominal voltage of the utility voltage and output voltage gauges by going to the advanced page (double click the gear icon in the upper right corner of the web [...]

Done, but only affected the output voltage gauge range, no changes to the nominal.

[image]

> An attempt to detect the model number is only made when the software version changes and it requries the internet to look up the model number. To force a check you can edit the /etc/genmon/genmon.conf file and remove the lines that define "model" and "version".

Done. Worked. Now I have the model displayed.

> Regarding fully supporting 3 phase, this issue is two fold. 1) I don't have enough data to know if there are additional modbus registers that could be beneficial in displaying voltage info on each [...]

Not really needing this. Each phase is externally monitored in terms of sequence order (not sure if this is called like that in English -E1,E2,E3 or R, S, T naming for each phase) and voltage range so I do not need that level of detail. Just a nice to have.

> Let me know if you have any additional questions or if you want to perform some tests to try to determine the additional modbus registers. These tests will requires enabling some debugging features in genmon and then running your generator.

Of course I can do some testing in order to help improve the software. Happy to help.

Finally, the Outage tab shows that there never was an outage when I simulated 3 or 4 during the day in order to test the setup. The voltages on the same tab marked in yellow are also wrong.
[image]

jgyates, 2022-03-15:
How are you simulating an outage? Are you cutting E1, E2 and E3? The pickup and threshold voltages are read from the controller (assuming they are the same modbus register for 3 phase). An outage detected when the utility voltage drops below the threshold voltage. Of course if your utility voltage is not correct this would be an issue. 

Can you tell me the expected output of each leg and the total expected voltage of the utility and generator output voltage?

efrenmartin, 2022-03-15:
[image]

jgyates, 2022-03-15:
Since the outage detection is being handled by your transfer switch then you may want to try these two settings located on the settings page:

Do Not Check for Outages - this disables the software checking for outages

and 

Smart Transfer Switch - This removes some UI elements that would not be applicable (like weekly exercise)

I am curious, if the controller is reading the line voltage as the sum of two RMS line voltages, does this value of 440V utility voltage go to zero if you disconnect the power via the transfer switch? Are all four utility lines (three lines and and a neutral) connected to the controller? I am trying to make sense out of the 440V value.

Does the transfer switch initiate a weekly or bi-weekly exercise or is this performed by the controller?

I have seen genmon used with one model like this before in Australia however the line to line voltage was 400V in that instance, not 380V.

efrenmartin, 2022-03-16:
>Do Not Check for Outages - this disables the software checking for outages

Just did this

>Smart Transfer Switch - This removes some UI elements that would not be applicable (like weekly exercise)

This was already set this way... seems like now more Line information is being populated...
[image]

>I am curious, if the controller is reading the line voltage as the sum of two RMS line voltages, does this value of 440V utility voltage go to zero if you disconnect the power via the transfer switch?

The power does not go directly to the transfer switch. Instead, the external 3-phase monitoring devices sends 220V through N1 and N2 control wires to the generator (see  attached manual page)

>Are all four utility lines (three lines and and a neutral) connected to the controller?

No, they are not.

>I am trying to make sense out of the 440V value.
Yeah... me too.

>Does the transfer switch initiate a weekly or bi-weekly exercise or is this performed by the controller?

Before installing Genmon this was controlled by the Mobile Link software and worked as expected. Now I am not sure. Seems that it will do the standard bi-weekly test on Saturdays at 6pm and the Enhanced Genmon exercise on Fridays at 0530pm. Once I confirm I will disable one of them....

>I have seen genmon used with one model like this before in Australia however the line to line voltage was 400V in that instance, not 380V.

Seems like a lattitude-geo thing... the Aussies and us use also the same kind of wall plugs... :-)

efrenmartin, 2022-03-16:
The generator was delivered to me and I installed it myself (I do not trust poorly trained technicians, afraid to say) . Is it possible that no calibration has been done and that some of the values require calibration/setting to make some sense?

jgyates, 2022-03-16:
I did not the an attachment. If you respond in email to this thread all attachments are stripped by github. You can attach a file from the web interface by just drag and drop the file into the thread edit box on the github site.

If your transfer switch does not initiate the weekly exercise then you can disable the "Smart Transfer Switch" option. This option just disables the controls from showing up on the Maintenance page in the case where the transfer switch (not the controller) handles the weekly exercise. It should like your switch is not handling the weekly exercise. The controller is still initiating the exercise in all cases (genmon and mobile link). 

The "Do No Check for Outages" option disables the checking of outages by the genmon software.  When outages are checked by the software it simply checks that the utility voltage is above the threshold voltage. If the utility voltage drops below the threshold voltage then an outage is logged and notification is sent to the user. In your case an outage would be detected if the combined voltage on N1 and N2 is below the threshold voltage. the threshold voltage is settable in the controller but the default for this model is 132V.

The ~440V on the utility voltage makes a lot more sense now. The controller is just taking the N1 and N2 voltages and adding them. This values (the sum of N1 and N2) is read over modbus by the software and displayed as the utility voltage.

Regarding Mobile link, what functionality is present with mobile link that is not present in genmon. In theory, since mobile link is getting data over modbus, just like genmon, we should be able to support that functionality as well.

efrenmartin, 2022-03-16:
N1 and N2 are live and neutral output from the monitoring device...
[image]

jgyates, 2022-03-16:
I just checked in an update that will adjust the utility voltage if an Evolution Air Cooled 3 phase controller is detected. I did not update the version number so just update the software from the About page. As as result of this change you will likely be able to uncheck the "Use Nominal Volts Override" on the Advanced settings page. I tested this with my simulator and outage checking works as expected. With this update and 220V across N1 and N2 you should not need  to use "Use Nominal Volts Override" or "Do Not Check for Outages" and as long as you set the exercise time in the controller and not with your transfer switch you will not need "Smart Transfer Switch" either.

 Let me know if this does not work for you or if you have any other issues.

jgyates, 2022-03-19:
@efrenmartin 

I am going to close this issue. Let me know if you still have any issues.

Thanks
Jason

closed 2022-03-19 by jgyates

efrenmartin, 2022-03-20:
There are still some minor issues… but I won’t be able to post since I am traveling.  Be back on Monday.