jgyates/genmon #687: Any support for PWRcell?

bdruth, 2022-03-20:
Question: can this be used in some way with Generac's PWRcell line of solar products (inverter, battery)? The info sheet on them says they support CANbus, RS4854, and Ethernet. There's a footnote that also mentions "Modbus", but I can't find what the footnote is for. Any pointers to any communities that have folks looking into this?

oriolism, 2022-03-20:
I was looking to integrate my PWRcell into Home Assistant and i found that Pika Energy(Bought by Generac) had a local sensor for this inverters, but it looks like they deprecated the local access. So far i found that they use a raspberry pi for communications(top left black box) and ssh is enabled. The beacon port(usb b) is sending the information via serial (baudrate 19200) to the raspberry pi. I connected my laptop to that port and the sensors values are received every time you hit enter. I think that the RJ11 service port is the modbus connection to the inverter but i dont have the cable to test. Pika Energy had the comm terminals exposed but they where replaced by the RJ45(Cts) Transfer Sw and rapid Shutdown Sw.  

https://api-docs.neur.io/#sensor-local-access

jgyates, 2022-03-20:
This project does not support any solar inverters or battery.

Typically solar inverters support the sunspec method to export data. Github has many projects that support sunspec in one form or another.  https://github.com/search?q=sunspec

Here is a repository that appears to support PWRCell by getting data from generac: https://github.com/edalquist/pwrcell-aggregator

closed 2022-03-20 by jgyates