jgyates/genmon #688: Generator Output Frequency Doubled

nah2323, 2022-03-22:
We have two monitors on liquid cooled generators, with NEXUS panels. 
 rPi3B+ pintsize hat with 12vdc powering through the hat. 
I updated to the most recent genmon version v1.18.09

For some reason the rpm reading is correct, but it's calculating a 4 pole rotor on a 2 pole rotor machine.  
For reference (with a very very few rare exceptions,) all 3600rpm units are 2 pole rotors, and all 1800rpm units are 4 pole rotors.

Is there a setting somewhere that I can correct that value?  In reality I know the machine is operating properly as 3600rpm is correct, and the machine would fault out with a frequency that high in a split second.

[image]

I apologize for deleting the template to convey this issue.

jgyates, 2022-03-23:
@nah2323 

Can you send your registers for testing?

jgyates, 2022-03-29:
I never received your registers so I can't say that I have a good data point to test this fix. I checked in an update that, based on the info in this thread will divide the frequency read from the controller by 2 if the "Use Nexus Legacy Frequency" is option is enabled on the Advanced setting page. You can access the advanced settings page by double clicking the gear icon in the upper right on the main web interface.

I am closing this issue due to lack of feedback but if possible please let me know if this works or send your logs so I can validate.

closed 2022-03-29 by jgyates

jgyates, 2022-04-05:
I am removing this code as it introduced additional complexity. If you want to revisit this issue you will need to response to this thread and provide register data.