jgyates/genmon #690: Log files for H-100 controller

wb9rro, 2022-03-26:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Create a logfile in csv or excel with various parameters such as start time, stop time, Run time, Output voltage (each phase), output current (Each phase), oil pressure, coolant temp and so forth. Append this info to a Google doc or CSV file via wifi. I'm not enough of a programmer to do this myself (yet). 

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2022-03-26:
@wb9rro I actually just did this not long ago.  I did it using NodeRed and some bash scripts to work with a SQLite3 database.  I would use the web command access to pole json status,maintenance,monitor etc. data from the Genmon server.  Then store that data in the database and export from there to a csv file.  More than one way to skin a cat type of thing.  Figured it all out through online tutorials.

jgyates, 2022-03-26:
@wb9rro 

This is outside the scope of the project however you could do what @liltux described manally by entering the URL commands in your browser and copy and past them into a document to reformat them as you like then paste into a spreadsheet. These are the URLs you would use (adjust the IP address based on your pi's IP address)

http://192.168.1.10:8000/cmd/status_json
http://192.168.1.10:8000/cmd/maint_json
http://192.168.1.10:8000/cmd/logs_json

closed 2022-03-26 by jgyates