jgyates/genmon #691: Low Voltage warning Pi 4

philruel, 2022-03-27:
[image]

jgyates, 2022-03-27:
Hi @philruel 

I don't see a question but I assume you want to ask what you should do about this? I did not received a copy of your log files, which you can submit on the about page if you have outbound email configured.

Is any other software loaded (besides genmon) on the pi?

you may want to to a visual check of your pi to make sure nothing is obstructing heat from dissipating from the pi (i.e. bugs, not sufficient air flow, etc). Also, are you using heat sinks on your pi. Your CPU temp is not out of the ordinary but that can vary based on the CPU load and ambient temperature.

Another thing you can try is to make sure your power supply is not defective. This could cause this issue.

skipfire, 2022-03-27:
Unfortunately, your post is just an image, no text.  You state this has happened several times over the last few weeks, but the picture shows an uptime of almost 8 weeks, and it only resets upon reboot.

Ignoring that, you've got 4 potential causes.

1. A fluke since it hasn't been rebooted in 8 weeks to reset
2. Your power converter is going bad
3. Your Pi has an onboard converter that is having problems (I have seen this recently on a 3 A+)
4. Your battery/charger is having problems

The best way to troubleshoot this is reboot to see if it recurs, if it does then swapping out whatever components you can one by one and see if it persists.

philruel, 2022-03-28:
Sorry you couldn't see the text:

 
I'm getting a low voltage warning on my Raspberry GUI.  Below is information taken from Genmon which also sees the undervoltage and CPU frequency throttling.  This is not a new issue.  It's been occurring for the past few weeks.  I was getting no issue when I first build it 6 months ago.
I'm using this power converter as suggested in the Wiki for powering my Pi.
https://www.amazon.ca/gp/product/B086KTGRH1/ref=ppx_yo_dt_b_asin_title_o03_s01?ie=UTF8&psc=1

- Generator Model: 065520
- Genmon Version: V1.18.09
 
Thank you!

jgyates, 2022-03-28:
I received your logs. Nothing unusual there. I would take the suggestions from @skipfire

philruel, 2022-03-28:
@jgyates 
No software is loaded on the pi apart from Genmon.
Yes, there's a heatsink on the chip.

@skipfire 
I'll check for the 4 causes.

Thank you!

skipfire, 2022-03-28:
@philruel if you have the instructions for your power converter you may want to also check the operational temperature ranges.

philruel, 2022-03-28:
@skipfire it didn't came with instructions.  I looked at Amazon web page and it didn't gave me more info.
Winters here go down to -35 C.  Could be the cause...
Do I have an alternative to power my Pi 4B?

skipfire, 2022-03-28:
A lot of ICs bottom out at -20C, chips that do -40C/F are about double the price of chips that do -20C, and sadly some chips are only rated for 0C.  The documentation for the Pi doesn't list a range for the board, but the chip is only rated to -40C.

jgyates, 2022-04-01:
You can power the pi from an outlet or PoE with an adapter however you would need to take the extra step of connecting the pi's ground on the GPIO to the controller ground on the molex.

one thing to consider would be to run the RS-232 lines from the generator inside and run the pi with a temp controlled environment. This would allow you run a pi from a standard wall plug. Ground would need to be shared between the pi and the generator.

jgyates, 2022-04-04:
I am going to close this thread for now but feel free to post here with any follow up status or additional questions.

closed 2022-04-04 by jgyates