jgyates/genmon #692: No communication with controller

renobles, 2022-03-30:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

### Expected Behavior

Genmon shows connectivity to generator controller, and myserial.log doesn't show errors.

### Actual Behavior

Genmon says "Monitor Health : System Initializing. Not receiving data."
Myserial.log shows: Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:129

### Steps to Reproduce (including precondition)

1- Setup Raspberry Pi OS lite on a Raspberry Pi 3b+
2- Install genmon
3 - run serialconfig.py and verify the serial setup
4 - Run serialtest.py at each step of the connection (verified all the way to the end of the cable)
5 - Review status in the webui or myserial.log

### Screenshot or Pictures relating to the problem (if possible)

See actual behavior above. 

### Your Environment

- Generator Model: Generac 6250/ Controller: J8371A (Evolution 1.0)
- Genmon Logs: Submit via the About page - Submitted
- Genmon Version: Version V1.18.09

jgyates, 2022-03-30:
@renobles 

A few questions / comments:

1) does you myserial.log file only have one instance of this error. I would try restarting genmon and look to see if the timestamp on this message occurs when you try load genmon or is this occurring many times? does the file mymodbus.log have errors as well?
2) I have not received you logs. You have to have outbound email working with genmon before the software can send your error logs.
3) Were changes made when running serialconfig.py? if so then a reboot is needed before the changes take effect.
4) I assume you are using /dev/serial0 and not /dev/serial1 or a USB serial device, correct?
5) what is the output of this command on the console:

        ls -la /dev/ser*

6) This wiki entry may have some data that is relevant. It contains the most common serial troubleshooting approaches.

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

jgyates, 2022-03-30:
Also, pictures of your cabling can help.

renobles, 2022-03-31:
1. The error pops up once on restart of genmon or reboot of the rpi. Mymodbus has errors:
-Mymodbus also has errors: 
`2022-03-31 09:56:28,399 : Error: timeout receiving slave packet for register 0243 Buffer: 0, sequence 21882 `
-These happen repeatedly. Like every few seconds.

2.  I went to setup outbound email but got an error when I tested that:
```
An error occurred: Error Initializing SMTP library: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1123)
Please try again.
```

3. The unit has been reboot multiple times. Same errors persist

4.  That is correct.

5.
```
lrwxrwxrwx 1 root root 7 Mar 30 15:24 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Mar 30 15:24 /dev/serial1 -> ttyS0
```

6. I went through that wiki entry and all of the tests worked. I do see that error under "Other issues", but don't have anything that I know of using the serial port other than genmon

Pics are attached of the cabling. The red and black cables go to the positive and negative terminals of the battery. The block is in place to facilitate swapping the tx and rx connections for testing.
[image]
[image]
[image]
[image]

jgyates, 2022-03-31:
above you mentioned that you "Install genmon". Did you use this command to install genmon?

      ./genmonmaint.sh -i

The error regarding incorrect SSL version is a possible issue with the install process. Are you running rasbpian debian lite? I see you are using the PiHat, are you using a preloaded SD card or did you install this from scratch?

Did you make this cable? It is difficult to tell from your pictures, however if you have TX and RX swapped you will see the behavior you are seeing (loopback passes but can not communicate with the controller). You may try swapping the tx and rx connections as a test.

renobles, 2022-03-31:
That is the command I used to install genmon. I set the rpi up from scratch using raspberry pi os/raspbian lite. Is there a way to verify/repair the install of genmon?

The cable is the one I purchased with the PiHat. I did test the communication with the tx and rx how it came in the cable and swapped. Same result both times.

skipfire, 2022-03-31:
That looks like the cable has been modified from what was shipped, what all has been changed?

jgyates, 2022-03-31:
one thing to do is to make sure that when you run the loopback test is to make sure that genmon is not running in the background. The genmonmaint.sh install has an option to start genmon on boot. To run the loopback test you must stop genmon with this command:

      cd genmon
      ./startgenmon.sh stop

you can restart genmon (from the genmon folder) with this command:

      ./startgenmon.sh restart

You can run the installation again with this command:

       ./genmonmaint.sh -i

The install will prompt you to copy the config files and to start genmon on startup. If you select "yes" to copy the config files it will overwrite any other settings you have made to your config files. If you already ran the installation script and selected "start on boot" then there is no need to do that again.

renobles, 2022-03-31:
Thanks. I did make sure to stop genmon before running the serial tests each time. And restart it after. I will give the install a go again and report back.

renobles, 2022-03-31:
> That looks like the cable has been modified from what was shipped, what all has been changed?

All that I did was insert a block inline in the cable to facilitate swapping the tx and rx connection. Oh, and put twist wire connectors on the power connections to make it easier to remove for bench testing.

renobles, 2022-03-31:
After running the installation again and rebooting, same result for both the myserial/mymodbus logs and the smtp setup.

jgyates, 2022-04-01:
This command will download the logs into a tar.gz file. You can send that as an attachment to this thread by dropping into the text box on the gitbhub web interface. If you try to send it as an email attachment it will get stripped off. Change the IP address in the commands below that match your pi's IP address.

       http://192.168.1.10:8000/cmd/get_logs

You can also issue this URL:

     http://192.168.1.10:8000/cmd/support_data_json

Copy and past this output into this thread.

This is the same info that is sent when sending registers/ log files from the about page.

If you have went though the info on this page:

     https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

I can take a look at your logs and see if anything lends a clue as to what is occurring.

renobles, 2022-04-01:
Thanks for the extra info. Github didn't like the .tar, so I zipped it up and attached it. Here is the JSON output:
```
{
  "Program Run Time": "Generator Monitor running for 18:17:22.",
  "Monitor Health": "System Initializing. Not receiving data. ",
  "Controller Selected": "generac_evo_nexus",
  "StartInfo": {
    "version": "V1.18.09",
    "sitename": "SiteName",
    "python": "3.9.2",
    "platform": "linux",
    "os_bits": "64",
    "zone": "EDT",
    "fueltype": "Propane",
    "model": "Generic Air Cooled",
    "nominalKW": "22",
    "nominalRPM": "3600",
    "nominalfrequency": "60",
    "phase": "Unknown",
    "Controller": "Evolution, Air Cooled",
    "Actual": "",
    "NominalBatteryVolts": "12",
    "PowerGraph": true,
    "FuelCalculation": false,
    "FuelSensor": false,
    "FuelConsumption": true,
    "UtilityVoltage": true,
    "RemoteCommands": true,
    "ResetAlarms": false,
    "AckAlarms": false,
    "RemoteTransfer": true,
    "RemoteButtons": false,
    "ExerciseControls": true,
    "WriteQuietMode": false,
    "Firmware": "",
    "Hardware": "",
    "Linux": true,
    "RaspbeerryPi": true,
    "SetGenTime": true
  },
  "PlatformStats": [
    {
      "CPU Temperature": "99.68 F"
    },
    {
      "Pi Model": "Raspberry Pi 3 Model B Plus Rev 1.3"
    },
    {
      "Pi CPU Frequency Throttling": "OK"
    },
    {
      "Pi ARM Frequency Cap": "OK"
    },
    {
      "Pi Undervoltage": "OK"
    },
    {
      "CPU Utilization": "0.56%"
    },
    {
      "OS Name": "Debian GNU/Linux"
    },
    {
      "OS Version": "11 (bullseye)"
    },
    {
      "System Uptime": "18:55:44"
    },
    {
      "Network Interface Used": "wlan0"
    },
    {
      "WLAN Signal Level": "-64 dBm"
    },
    {
      "WLAN Signal Quality": "46/70"
    },
    {
      "WLAN Signal Noise": "-256 dBm"
    },
    {
      "WLAN ESSID": "Baer"
    },
    {
      "System Time": "Friday April 1, 2022 07:53:00"
    }
  ],
  "Data": {
    "Registers": {
      "Num Regs": "0",
      "Not Changed": "0",
      "Changed": "0",
      "Total Changed": "0.00",
      "Base Registers": [],
      "Log Registers": {
        "Logs": {
          "Alarm Log": [],
          "Service Log": [],
          "Run Log": []
        }
      }
    }
  },
  "Registers": {},
  "Strings": {},
  "FileData": {}
}
```
[genmon_logs-2.tar.zip](https://github.com/jgyates/genmon/files/8397400/genmon_logs-2.tar.zip)

jgyates, 2022-04-01:
@renobles 

I looked at your logs and the above info and I want to validate a few things with your system.

1) Can you show the output of this command:

       cat /etc/os-release 

in the above output the "OS Name": "Debian GNU/Linux" should be "Raspbian GNU/Linux". This command will give the full OS info. This makes me think you have regular Debian for Raspberry Pi (https://raspi.debian.net/) instead of Raspbian (https://www.raspberrypi.com/software/).

2) can you show the output of this command:

      pip3 freeze

This will show all of the versions of the python libraries used.

renobles, 2022-04-01:
1 - 
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Which is quite interesting. I used the raspberry pi os imager at: https://www.raspberrypi.com/software/ to create the image. It looks like in late 2021 they changed from "raspbian" to "raspberry pi os". The latter being Debian bullseye. Could that be an incompatibility with genmon?

2 - 
```
certifi==2020.6.20
cffi==1.15.0
chardet==4.0.0
chump==1.6.0
click==8.0.4
colorzero==1.1
configparser==5.2.0
crcmod==1.7
cryptography==36.0.1
distro==1.5.0
Flask==2.0.3
geojson==2.5.0
gpiozero==1.6.2
idna==2.10
itsdangerous==2.1.1
Jinja2==3.0.3
ldap3==2.9.1
MarkupSafe==2.1.0
paho-mqtt==1.6.1
ply==3.11
psutil==5.9.0
pyasn1==0.4.8
pycparser==2.21
pycryptodomex==3.14.1
PyJWT==2.3.0
pyOpenSSL==22.0.0
pyotp==2.6.0
pyowm==2.10.0
pyserial==3.5
pysmi==0.3.4
pysnmp==4.4.12
python-apt==2.2.1
pytz==2021.3
requests==2.25.1
RPi.GPIO==0.7.0
six==1.16.0
smbus==1.1.post2
spidev==3.5
ssh-import-id==5.10
twilio==7.7.1
urllib3==1.26.5
Werkzeug==2.0.3
```

jgyates, 2022-04-01:
Raspbian and Raspberry Pi OS are the same thing. Raspbian (Raspberry Pi OS) based on the Debian Linux distribution (debian.org). Debian.org has their own version of a Raspberry pi compatible OS, which is what you appear to have installed.

You can install several operating sytesms using Raspberry pi imager. I would suggest using "Raspberry Pi OS Lite (32)" which is Raspbian. If you run the "cat /etc/os-release" command it should show this:

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi OS was once called Raspbian but they renamed it, but it is still a port of Debian. It was ported by the Raspberry Pi Foundation so it is more compatible with the hardware than any other linux distribution for the raspberry pi. 

I have never tested a raspberry pi with plain debian. Since you are seeing hardware related errors in the serial log I would suggest using Raspberry Pi OS / Rasbpian. 

Also, you are running the 64 bit version of Debian. I have tested the 64 bit version of  Raspberry Pi OS and it works however there is not a real advantage of running the 64 bit version on a pi and the 32 bit version has a lot more milage so it likely has a lot more any operation system issues worked out.

Assuming you are confident with your cabling, I would recommend starting over with Raspberry Pi OS Lite (32) as I can see no other reason why you are having serial issues.

renobles, 2022-04-04:
Thanks for the help on this. A reinstall with the 32-bit version worked for me. Many thanks!

closed 2022-04-04 by renobles