jgyates/genmon #693: Can't cancel a print command - Can only save

davisgoodman, 2022-04-04:
### Expected Behavior

For any of the printing options whether printing the logs, registers or else should be able to cancel

### Actual Behavior

The cancel option works on Edge, Opera but on Chrome, you can't cancel.

### Steps to Reproduce (including precondition)

Try printing any journal, logs or register and cancel the operation.

### Screenshot or Pictures relating to the problem (if possible)

[image]

### Your Environment

- Generator Model: Genrac 20k
- Genmon Logs: Submit via the About page
- Genmon Version:  V1.18.09

jgyates, 2022-04-04:
@davisgoodman 

I am unable to duplicate this on Chrome 99 or Chrome 100 on Mac or Windows 10. There is a new version of Chrome out (V100) have you tried updating and restarting Chrome? From the softwares perspective there is not way to tell the print engine to allow a cancel or disallow a cancel so I am not sure there is anything I can do about this considering I can not duplicate. 

also, just to validate I assume you mean you can not press the cancel button in the picture above and you do *not* mean cancel the print job from the print queue.

I can successfully cancel the print preview screen (by pressing the cancel button) when printing to a printer and saving to a PDF on both Chrome 99 and 100 on Windows 10 and Mac OS.

jgyates, 2022-04-04:
you may also want to validate you can duplicate this (or not) by printing from a pager other than the genmon interface.

davisgoodman, 2022-04-04:
@jgyates  Well seems either my Chrome updated or else. I can no longer replicate the issue. Sorry for the noise

jgyates, 2022-04-04:
no worries

closed 2022-04-04 by jgyates