jgyates/genmon #695: 30hz displayed should be 60hz 20Kw generac air cooled running at 3600 rpm with no load. Looking for a solution.

601dan, 2022-04-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2022-04-04:
I would need more info to give you any insight.

Generator Model?
Genmon Version?
Is this a new installation? If yes then submit your logs on the about page.
Is this a Nexus or Evolution controller?
If it is a Nexus do you have the setting name "Use Nexus Legacy Frequency" enabled?

jgyates, 2022-04-04:
"Use Nexus Legacy Frequency" is on the Advanced settings (double click the gear icon in the top right to access the advanced settings). If you have a Nexus air cooled and this setting is enabled it will divide the frequency by 2. Early versions of Nexus firmware require this, but later versions do not.

601dan, 2022-04-04:
Generac 6250
Genmon version V1.18.09
New install of genmon, generator installed 2013
Evolution 1 controller
Sent logs and registers

601dan, 2022-04-04:
The software is fantastic for my needs head and shoulders above generac hardware. I was using the RF remote 

Dan Nobles
PS; patience is needed (75)

jgyates, 2022-04-04:
Hi @601dan 

I assume you mean the G006250. I have not received your registers. In order to send them on the about page you have to have outbound email configured and working. I can get a look a the register for the frequency by doing this while the generator is running:

      1)  Start the generator
      2) Open the genmon web interface and click on the gear icon in the upper right one time, this will bring up the registers that genmon is monitoring. The register we want to look at is marked in the picture below:

[image]

Let me know the value of this register. If the generator is running it should be around 3C hexadecimal or 60 decimal. If the value is around 1E hex or 30 decimal then the halfed  (30Hz instead of 60Hz) value is being sent from your controller. If it is 3C / 60 Hz but being displayed as a 30Hz then we will need to look at your settings as you may have an incorrect setting in your config file.

601dan, 2022-04-04:
Hex 3c is displaying in the engine frequency hz register
My outgoing email is set and tested

601dan, 2022-04-04:
Generator model is what you noted. Generally generac only goes by 6250 for parts

601dan, 2022-04-04:
Resent register data with generator running

601dan, 2022-04-04:
3b not 3c is showing

601dan, 2022-04-04:
Actually it goes between 3b and3c probably matching the reported 29.5 and 30hz on the gui

jgyates, 2022-04-04:
3b hex is 59 and 3c is 60, so the controller is reporting the data correctly.

One the Maintenance page of the web interface, what does is say for "Controller Detected" ?
One the Monitor page of the web interface, what does it say for "Controller"?

Are these values the same or different and what is the exact value?

I have not received your register data, my guess is that you outbound email is not setup correctly. You can test your email settings on the Settings page. Outbound email must work before you can submit your registers or log data successfully.

601dan, 2022-04-04:
Evolution, Air Cooled
Evolution, Air Cooled   Both are the same

email password reverted to an old one probably something I did. Resent the registers

jgyates, 2022-04-04:
Are you saving your settings at the bottom of the page when you make a change in the email settings? I still have not received your registers or logs.

Have you edited any of the config files in /etc/genmon manually or did you setup everything using the web interface?

601dan, 2022-04-04:
I did not save the setting, just did and will send register
My son the IT wiz set the hardware and software up so I don’t know the answer to your last question. I have messaged him to find out.

jgyates, 2022-04-04:
I got your registers. Thanks. That helped a lot. I figured out that I introduced a bug in the last update. I checked in an update that should fix your issue. Just go to the about page and click the Update button. Let me know how it goes.

601dan, 2022-04-04:
Works like a champ. You are a master! Can’t believe the assume support. Thanks so much. Mark this closed!

closed 2022-04-04 by jgyates