jgyates/genmon #696: Error being logged: invalid literal for int() with base 10

skipfire, 2022-04-04:
Error being logged in mymail.log, but does not appear to be impacting anything.  I found this while troubleshooting emails not being sent but it appears to be unrelated as I fixed the email send and the error is still being logged (an issue in December fixed by updating pyserial from a beta version fixed it, did not confirm the pyserial version first).

Error in MyConfig:ReadValue: MyMail: extend_wait: invalid literal for int() with base 10: '' : myconfig.py:94

`# extend wait time on errors
extend_wait =`

I'm thinking 2 lines of code would fix it (2nd & 3rd below), but I don't know if it would cause any other problems.

`            if self.config.has_option(self.Section, Entry):
                if self.config.get(self.Section, Entry) == "":
                    return default
                if return_type == str:
                    return self.config.get(self.Section, Entry)`

### Expected Behavior

No error logged

### Actual Behavior

Error logged

### Steps to Reproduce (including precondition)

extend_wait with no value, error logged on-start

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070432 (22kW air cooled Evo2)
- Genmon Logs: Submitted
- Genmon Version: 1.18.10 (and at least some previous versions)

skipfire, 2022-04-04:
I tried making that change locally and it threw several errors, so probably a bad idea.

jgyates, 2022-04-04:
This error message is informational the code in myconfig.py will print this when it is expecting an integer but receives no value. In that case the default value is used. The easy way to get rid of this error is to set "Extend email retry" to zero on the advanced settings page. 

Earlier versions of the code did not have an entry in mymail.conf for this entry which insured that the default value (zero) was used. If you ever assigned a value then removed it then this error will show up, but the empty value will be replaced by zero.

closed 2022-04-04 by jgyates

skipfire, 2022-04-04:
Great, thanks!