jgyates/genmon #697: Email not being sent when the site name contains an accentuated character

davisgoodman, 2022-04-09:
### Expected Behavior
Receiving email notification when events happens

### Actual Behavior

Not receiving emails, only SMS which has been configured

### Steps to Reproduce (including precondition)

When the Site name contains an accentuated character such as "é" the mail log contains the following: 
2022-04-08 19:07:14,431 : Error in SendMailThread, retrying (2): 'ascii' codec can't encode character '\xe9' in position 83: ordinal not in range(128) : mymail.py:511

Having accentuated characters in the Email config doesn't have any effects.

[image]
 I had to remove the "é" in "Gard(e)nias for the mailer to stop errors in mymail.log

I can still have the "é" in the Email Settings
[image]

### Your Environment

- Generator Model:  G0070380
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.10

jgyates, 2022-04-09:
@davisgoodman 

I am assuming that the "Email Account" field, which is not shown in your picture above has the "é" in the domain name based on the error. This error is occurring when the "Email Account" field and "Email Password" field are passed to the python [smtplib](https://docs.python.org/3/library/smtplib.html) library. This library is used as a helper to handle the interactions to the email server. It is included with python. I have tried a few experiments to try to work around this issue, but at the moment I don't have a good answer other than to use a different email account. You *might* be able to send to an account with an accented name (I have not tested but the code looks like this is possible) however the login code within the smtp library is not handling unicode and I am not aware how to work around this or if I even can. I will try to look into this a bit more but for now the only thing I can suggest is to use another email account for sending.

jgyates, 2022-04-09:
@davisgoodman 

I checked in a change that ***might*** help. I can't say for sure since I don't have and email account with an accented domain to test with but with this change it does appear to get passed to the mail server. This change will only allow for accented characters in the domain. It passes the user part of the email address with out any parsing or modifications. Let me know if this change works for you.

To test just go to the About page and click "Update To the Latest Version" This will pull the update and restart genmon. The version number has not been updated.

davisgoodman, 2022-04-09:
@jgyates Just to point out, there are NO accentuated character in the email domain. Anyway this is not supported. I blanked out the email domain just for security purpose. The issue is when there is an accentuated character in the Site Name as shown above. 
Any accentuated character in the email settings such as the Sender name has no effect and doesn't prevent emails of being sent. 

It's actually working now since I removed the "é" in the Site name

jgyates, 2022-04-09:
Thanks for the clarification. I can't duplicate this issue. Added an accented character to the sender name and I do not get any email errors. I am doing this on a Mac and I add the accented charter by long pressing the "e" on the keyboard then selecting the desired accent.

Anyway, I am glad it is working for you now. If you want to explore the issue with the sender name just let me know.

closed 2022-04-09 by jgyates

davisgoodman, 2022-04-09:
Ok the issue is not in the sender’s name it’s when you set one in the “Site name” at the top

 

It should read “ Generator Monitor at 105 Des Gardénias” but I had to remove the é in it to have emails go out. You can cut and paste the one here to test it or even better make it “105 Des Gardénias”

jgyates, 2022-04-09:
Ah, OK. I was getting thrown by the picture. I keep looking for the issue in your picture.

I copied and pasted "105 Des Gardénias" from you reply above and put it in my "Site Name" setting and I got this email.
[image]

I am not sure what is different between you setup and mine.

davisgoodman, 2022-04-09:
Hum interesting.... let me see if I can reproduce it again. Hopefully it won't have me  lie! I tried a few times before open the issue

davisgoodman, 2022-04-09:
Ok so was able to reproduce. Setting my site name as mentioned, getting these errors in mymail.log and not getting any notifications.
2022-04-08 19:07:14,431 : Error in SendMailThread, retrying (2): 'ascii' codec can't encode character '\xe9' in position 83: ordinal not in range(128) : mymail.py:511
2022-04-09 15:16:50,444 : Error in SendMailThread, retrying (2): 'ascii' codec can't encode character '\xe9' in position 83: ordinal not in range(128) : mymail.py:511
2022-04-09 15:18:50,451 : Error in SendMailThread, retrying (2): 'ascii' codec can't encode character '\xe9' in position 83: ordinal not in range(128) : mymail.py:511
2022-04-09 15:20:50,460 : Error in SendMailThread, retrying (2): 'ascii' codec can't encode character '\xe9' in position 83: ordinal not in range(128) : mymail.py:511
2022-04-09 15:22:50,464 : Error in SendMailThread, retrying (2): 'ascii' codec can't encode character '\xe9' in position 83: ordinal not in range(128) : mymail.py:511

As soon as I remove it, then all emails flow in without issues.

Let me know if you'd like other kind of logs...

jgyates, 2022-04-09:
I think the issue may be the local settings. Try this:

from a console in the pi type:

       sudo raspi-config

This will bring up the raspbian config tool. From the menu select "Localization Options", then select "Local". This will bring up a menu to select the language and local settings. For your language and region make sure that UTF-8 option is selected, and not any other options. For example for US English select "en_US.UTF-8 UTF-8". I think the default is "en_UK.UTF-8 UTF-8". 

If you do not have UTF-8 selected then you likely have another one selected that only represents the first 256 characters in the Unicode character set so when python is trying to convert strings it does not have access to the needed character. The UTF-8 has 65535 characters instead of 256.

Let me know if that helps or not.

lmamakos, 2022-04-10:
Note that email headers are not directly UNICODE capable; they are specified for ASCII characters.  There's an encoding scheme by which non-ASCII character sets are handled.  For example, see [RFC-2047](http://www.faqs.org/rfcs/rfc2047.html) as one particular reference for this.

I'd imagine that, ideally, some python library used to construct an email message (including header) would take care of these details for you.  I've not ventured into investigating python libraries for this purpose, however, so con't have a specific suggestion.

As background, this restriction originally dates back to the time where RFC-822 -style email message format were transmitted over a variety of channels and "network", not just SMTP over the Internet.   (_e.g.,_ UUCP, BITNET, USENET news and other ancient things that have likely been mostly forgotten by now.)

jgyates, 2022-04-10:
That is good info, but the issue here is that a string with unicode for the subject generating an error within the smtplib code. The conversion works on my system but not his, which leads me to believe that the issue is related to the installed locale settings on the pi. If your system does not support UTF-8 then you can not convert from unicode to an encoded ASCII.

davisgoodman, 2022-04-11:
Maybe the issue is because I have two locales installed but both are UTF8. I’ll remove the En-Ca-UTF8 one keeping only the US one and try again.

jgyates, 2022-04-11:
what do you get if you type this command:

     locale

My output is this:

locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

jgyates, 2022-04-11:
The other thing I noticed from the logs you submitted is that you do not appear to be running the "Raspberry Pi OS" formally known as Raspbian. Instead you appear to be running debian. They are very similar but not exactly the same. This could be the source if the issue. I have noticed one other person who has installed Debian in stead of Raspbian but they had another issue. How did you install the OS on the pi? If you used the Raspbperry pi imager, what OS did you select?

davisgoodman, 2022-04-11:
Here is the output of locale:

 

***@***.***:/var/log $ locale

LANG=en_US.UTF-8

LANGUAGE=

LC_CTYPE="en_US.UTF-8"

LC_NUMERIC="en_US.UTF-8"

LC_TIME="en_US.UTF-8"

LC_COLLATE="en_US.UTF-8"

LC_MONETARY="en_US.UTF-8"

LC_MESSAGES="en_US.UTF-8"

LC_PAPER="en_US.UTF-8"

LC_NAME="en_US.UTF-8"

LC_ADDRESS="en_US.UTF-8"

LC_TELEPHONE="en_US.UTF-8"

LC_MEASUREMENT="en_US.UTF-8"

LC_IDENTIFICATION="en_US.UTF-8"

LC_ALL=

 

and I just did the test. I reset the Site Name to “ 105 Des Gardénias” and again email notification are not going out. SMS does but not emails. 

 

As for the install I believe I installed Raspberry PI OS LITE (64bit) would that make sense?

jgyates, 2022-04-11:
I noticed you have the 64 bit version of the OS which outputs this value:

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

I have a test system setup with the 64 bit version. I can duplicate this on the 64 bit version. This may be a bug in the OS locale code. Not that I can duplicate the issue I can look into it more closely.

jgyates, 2022-04-11:
I think I fixed it. This turned out to be an issue that only shows up with python3 on 64 bit systems. In short now you can use the special characters in the email. The only caveat is that this will not work with pythons 2.7 which is not really supported anymore anyway.

I checked in an update so just update from the About page then test.

Let me know if you have any issues or questions.

davisgoodman, 2022-04-11:
Hum.. I don’t see the update in the About page, should I or it will still say 1.18.10

jgyates, 2022-04-11:
I did not update the version.

davisgoodman, 2022-04-11:
All good, case closed.

 

Thanks for taking the time. This is an awesome project. I wish I would have found about it before. Already building one more box for a friend!

davisgoodman, 2022-10-11:
This is the only ones selected: