jgyates/genmon #698: Genalexa.py  process dying - Remote commands not supported

davisgoodman, 2022-04-09:
Trying to get Alexa to discover Genmon

### Actual Behavior

When trying to start the plugin, the process dies and these are logged in genalexa.log

I'm quite sure the Evolution 2.0 Controller supports remote commands as I'm able to sue the ClientInterface.py with generator:remote=start

### Steps to Reproduce (including precondition)
This is what my genalexa.conf file contains:
pi@generac:~/genmon $ cat /etc/genmon/genalexa.conf
[genalexa]
# The name for which the Echo speaker will respond.
name=generator

# The callback port for the echo speaker. Needs to be available
port=52004
debug = True

Log file has the entry every time I restart genmon
2022-04-09 15:30:53,181 : Key word: generator; Port: 52004; Debug: False
2022-04-09 15:30:53,406 : Generator does not support remote commands. So you cannot use this addon. Exiting....
2022-04-09 15:41:17,365 : Key word: generator; Port: 52004; Debug: False
2022-04-09 15:41:17,535 : Generator does not support remote commands. So you cannot use this addon. Exiting....
2022-04-09 15:44:23,465 : Key word: generator; Port: 52004; Debug: True
2022-04-09 15:44:23,686 : Generator does not support remote commands. So you cannot use this addon. Exiting....
2022-04-09 15:46:19,093 : Key word: generator; Port: 52004; Debug: True
2022-04-09 15:46:19,140 : Generator does not support remote commands. So you cannot use this addon. Exiting....
2022-04-09 15:47:45,591 : Key word: generator; Port: 52004; Debug: True
2022-04-09 15:47:45,637 : Generator does not support remote commands. So you cannot use this addon. Exiting....

### Your Environment

- Generator Model: Generac 20kw G0070380
- Genmon Logs: Submit via the About page
- Genmon Version: 1.8.10

jgyates, 2022-04-09:
This error is occurring because the add on is asking genmon if remote commands are supported by the connected controller but it is not getting the answer it expects. This could be due to the a few issues. 

Is genmon connected to the controller and communicating?
Are there any serial comms issues?
Is the pi running slow for any reason?

Does the following URL return any data (change the IP, http/https and port number to match your systems)? 

     http://192.168.1.10:8000/cmd/start_info_json

If it does the genalexa.py program is getting this data and looking for the "RemoteCommands" value to be "true". If it does not have this value or if it is "false" then you get this error.
 
   
I may be able to get more info if you submit your logs on the About page.

davisgoodman, 2022-04-09:
> This error is occurring because the add on is asking genmon if remote commands are supported by the connected controller but it is not getting the answer it expects. This could be due to the a few [...]

Genmon is working flawlessly. I actually had a 12 hours power outage yesterday and no serial errors at all. Rebooted a few times today and still no errors.
[image]
I was even able to start the generator with the ClientInterface.py.

Which logs would you want. All of /var/log or only the gen...log ones?

jgyates, 2022-04-09:
If your outbound email is working just go to the about page and click "Submit Logs". This will send all of the logs to me via email.  

Do you have the "Start", "Stop" and "Start and Transfer" buttons on the maintenance page?

The logs sent by clicking are the same as this URL http://192.168.1.10:8000/cmd/get_logs plus this info:

http://192.168.1.10:8000/cmd/support_data_json

davisgoodman, 2022-04-09:
They have been submitted

jgyates, 2022-04-09:
I received them. 

Do you have "Smart Transfer Switch" enabled on the Settings page? 

This option is for installations that use a transfer switch that monitors the utility line to determine if an outage occurred. In this situation the transfer switch (not the controller) would notify the generator when an outage occurred. In most installations with Evolution controllers a "dumb" switch is used and the controller monitors the utility line. 

This option is off by default. If enabled then remote commands are not available to add ons as the generator controller would be in Two Wire Start mode so the smart transfer switch could start the generator.

davisgoodman, 2022-04-09:
Yes I do as  I actually have a Smart Transfer Switch.

jgyates, 2022-04-09:
ok. then everything is working as expected. Genmon is reporting that remote commands should not be sent by add ons. 

Did you say that you used a remote command from ClientInterface.py and it worked? Is your controller setup for Two Wire Operation? The reason I ask is that if remote commands still work then in theory the code could be changed to allow this to work. The question remains is this a good idea to have two different sources enabling the generator.

jgyates, 2022-04-09:
also, is the ATS setting the exercise time? What ATS are you using?

davisgoodman, 2022-04-09:
Not sure if it is actually a Smart Transfer switch. Maybe you can confirm

jgyates, 2022-04-09:
If you had an attachment in your last replay, github will strip it off. To send an attachement you have to drop / drag it into the web interface.

davisgoodman, 2022-04-09:
Hum not sure the pictures made it through.
[image]
[image]

davisgoodman, 2022-04-09:
It might also explain that there is no records from MQTT as when the exercise is setup in the controller, right?

jgyates, 2022-04-09:
yes, that would explain that as well. Genmon assumes that the ATS is handling the exercise for a smart switch if that setting is enabled.  I can't really tell from the picture but if your exercise is set from the controller then I would assume that this is not a smart transfer switch. I don't see an interface to make any settings like threshold voltages or exercise time. You can try disabling "Smart Transfer Switch" in the settings. All this setting really does is it disables some of the functionality of genmon so it will not get in the way of the smart transfer switch. In my experience things like setting the exercise do not work with a smart transfer switch. The few I have seen used with this project on the Evolution did not even have the utility voltage attached to the controller as that is not needed if your switch is monitoring the utility line for outages.

davisgoodman, 2022-04-09:
Now that you mention this, I remember changing myself the exercise time and day on the controller itself before I had genmon installed then, I guess it wouldn't be a Smart Transfer Switch. But, between the generator and the transfer switch you can see that blue cable that comes out of the generator to the transfer switch. Could it be that this allows the Evolution Controller to manage the Smart Transfer Switch?

davisgoodman, 2022-04-09:
Ok now that I have turned of Smart Transfer Switch I do see with MQTT explorer the Exercise Time Settings
[image]

jgyates, 2022-04-09:
The connections between the controller and the ATS are similar in both a smart and dumb switch (only a couple of wires in both cases). The wires are connected differently in each scenario. With a dumb switch the controller uses a relay to signal to the switch to close the switch which cuts the electrical path to the utility ensuring the generator power does not go to the grid. With a smart switch the ATS signals the controller to start and stop via a two wire interface, everything else is up to the ATS.

davisgoodman, 2022-04-09:
OK but you seem to say that with a Smart Switch the settings for the Exercise time would be done on the Smart Switch, not the controller so if that is the case, it may not be a Smart Switch as I did change the exercise time on the controller last year.

If so then all is good and of course now the genalexa process do start!

jgyates, 2022-04-09:
OK. I am closing this thread but feel free to post here with any follow up data or questions.

closed 2022-04-09 by jgyates

davisgoodman, 2022-10-11:
No but…. I have gpioin enabled because I have the buttons on my box could that be the issue?

 

http://192.168.1.10:8000/cmd/support_data_json

jgyates, 2022-10-11:
No