jgyates/genmon #700: Evo 2 controller powering Pi 4 and HAT appears to have Utility Voltage measurement interference - use battery power instead

rainieryoung, 2022-04-10:
_I suspect this issue is more of an "inform" about using battery power source instead of the generator's controller..._

### Expected Behavior
Connecting Raspberry Pi 4, HAT, and molex cable for data + power without impacting the Generator Controller's monitoring of Utility Voltage or Output Voltage(?).

### Actual Behavior
**Problems** Utility Voltage measured by Evo 2 Controller is significantly impacted, enough to cause the generator to start-up repeatedly due to incorrect undervoltage readings with hardware powered off controller instead of battery.  When "off," the generator will not start unexpectedly, but the controller (reported through Genmon) will continue to show erratic Utility Voltage.  During an outage, the generator would start and run, but not transfer.  Unclear if this was further mis-reading of utility voltage or a separate mis-reading of demand or output voltage.  

**Solution** Disconnecting the 5v power to the HAT/Pi and using a [battery to USB power supply ](https://smile.amazon.com/HOMREE-Step-Down-Converter-Module-Charger/dp/B07ZQB6S3L/ref=sr_1_5?dchild=1&keywords=12v+to+5v+dc+converter+3a+usbc&qid=1601328672&sr=8-5) to the Pi 4' usb power eliminates the problems.

**Takeaways** If anyone is having trouble with Utility Voltage readings and is using the controller for power, try using the battery as the power source for your Pi hardware instead.

### Steps to Reproduce (including precondition)
I can't say if this would be consistent for everyone, but here's what was used:

Hardware:
- Raspberry Pi 4
- [PintSize.me HAT (standard)](https://pintsize.me/store/ols/products/generator-hat)
- 8 pin molex provided by PintSize.me (1meter) OR a repinned [EPS 8](https://www.amazon.com/gp/product/B000M802RG) (f / f instead of f / m)

Generator: 
- Guardian Series 20kW
- Evo 2 controller 

### Screenshot or Pictures relating to the problem (if possible)

#### Image 1
Utility Voltage recorded in Home Assistant.  I first connected Genmon on 03/31 and the change on 04/04 is the day I replaced the battery and the voltage readings were impacted enough that the generator was auto-starting then shutting down. 

I had a working theory of "interference" given that: 1) The generator behaved normally if the Pi was disconnected 2) My replacement of the battery was an unlikely root cause and 1) indicated the new battery was also not the problem 3) On 04/05 with the generator "off" and genmon connected I observed a general downward slide in the Utility Voltage reading (and no power outage at the time).
[image]

#### Image 2
The power outage occurred on 04/07.  You can see in this image the voltage reading is more like the readings from 03/31 than 04/05.  I can't say for sure why the transfer to generator power was impacted, but as I look at this the jumps to ~220 may have led the transfer to stay in standby.  It's also possible there's interference in an area I didn't measure (e.g. Generator Output Voltage)

The flat line at ~11;30PM is when I shutdown the Pi.  This particular graph in Home Assistant just carries the last value forward rather than showing a gap (which can be done, just not in this default view I'm utilizing)
[image]

#### Image 3
This as as-of 04/10, using the exact same hardware (and battery installed on 04/04) with the ONLY difference of cutting the power from the controller to the Pi and instead using battery power to USB.

Using full 0 to 250 scale for ease of comparing:
[image]

Same data set, filtered to only active readings (which auto-set the scale to 243 to 246):
[image]

Other notes:  When I first connected Genmon, it was not clear to me if my utility power was _that bad_ or if the reading that Genmon could see wasn't great.  I believe I now understand that the [Known Issue]( https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues) is limited or only expected for an Evo 2's Output Voltage (from the generator) and it is unexpected to have inaccurate Utility Voltage.

### Your Environment

- Generator Model: G0070391
- Genmon Logs: Submit via the About page -- I did this on 04/10/22 at the time of writing this issue.  The Pi was offline for several days prior to this morning, so I'm not sure if the logs will include data back to the events described above.
- Genmon Version: V1.18.10

skipfire, 2022-04-10:
@rainieryoung can you contact me through the Contact Us at the bottom of the https://PintSize.Me site?  I would like to get some additional details as this is the first I've heard of this being an issue.  I have an Evolution 2 controller as well and have not noticed this behavior so I'd like to rule out any sort of an issue with the HAT.

rainieryoung, 2022-04-10:
✅

closed 2022-04-10 by jgyates