jgyates/genmon #701: Custom buttons that do not change with update?

cuncil1, 2022-04-17:
I have my generator on my roof, and I would like to avoid going up every time I need to turn off the generator to change to my inverter to save money, since I live in Puerto Rico, electricity is unstable, and gas is getting really expensive.
Therefore, I am going to hardwire three relays to the ON, MANUAL, and OFF buttons. It is really easy for what I have seen. In order to hardwire the ON button, I need to short-circuit the 1 and 2 pins in the flat ribbon FPC that connects to the board. The MANUAL button requires to short-circuit the 1 and 3 pins in the same ribbon. The OFF switch is different but easier, there is a black and red 2-wire connector that need to be short-circuited. I will turn on the relays for 0.5 seconds simulating the button press. It is really easy using ssh with JuiceSSH app in Android. However, I would like to add 3 buttons to the Genmon software but I do not know how to add them in a place that avoids going away with an update.
Any recommendation? If it is impossible, I will do it with the JuiceSSH snippet, but it would be cooler with Genmon.

Thanks!

jgyates, 2022-04-17:
Hi @cuncil1 

I have tried to design the core of genmon to not rely on anything platform specific like GPIO. Any use to GPIO are handled in add on programs. Due to this design your request would not be supported in genmon.

closed 2022-04-17 by jgyates

cuncil1, 2022-04-17:
I understand that. Thanks! Can I make an Addon for that that does not change with updates?

jgyates, 2022-04-17:
The Addons do not support adding any elements to the UI, however one thing you can do is create a new html file and put it in the /genmon/static folder. This should allow you to create a new web page that the flask web server will serve up.

      http://192.168.11.14:8000/custom.html

then put custom.html in /home/pi/genmon/static

you can look at the file /home/pi/genmon/static/internal.html as an example of a simple html that uses java script and a CSS file. (internal.html, internal.js, internal.css). On this page you could create buttons that do what ever you like. You can add files the to the genmon folder with no issue, but if you modify any they could be overwritten with an update.

cuncil1, 2022-04-17:
But if I can add an AddOn that allows me to activate something like:
@app.route('/turnon')
def turnon():
    GPIO.output(38, True)
    time.sleep(0.5)
    GPIO.output(38, False)
    return ServePage('index.html')

This would "press the button" everytime I add /turnon to the website
Is that possible for an AddOn? I tried it changing the genserv.py and it worked, but I want to add it in a way that does not change with updates.

jgyates, 2022-04-18:
add ons will not allow you to extend flask as the "app" variable available. 

I checked in a two line change that should allow you to achieve the above flask route:

genserv.py will look for a file in the genmon folder named genservext.py, if it is present it will import that file.

genservext.py should look something like this:

     from __future__ import print_function
     from __main__ import app
     from genserv import ServePage
     
     #-------------------------------------------------------------------------------
     @app.route('/testit', methods=['GET'])
     def testit():
         return ServePage("internal.html")

This will load the file internal.html if you use the http://192.168.1.10:8000/testit url

From here you can create a new html, js file and add your GPIO stuff, you can change internal.html to index.html.

jgyates, 2022-04-18:
I added this to the wiki here:

https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications#extend-genservpy

cuncil1, 2022-04-18:
Works great! Thank you very much!

cuncil1, 2022-04-19:
I was able to connect everything, very easy to connect because the board has testing points for the buttons; however, ON and MANUAL are normally open switches so you have to connect them in parallel, and OFF is a normally closed switch, having to place the switch in series. Now, I can turn on, off, and manually on without going to the roof! Thank you again!