jgyates/genmon #703: Error with SMS by text after upgrade to 1.18.11.

skipfire, 2022-04-20:
### Expected Behavior

SMS by email is sent

### Actual Behavior

SMS by email generates an error, I can forward you the email if you want.
`Your message contains invalid characters (bare line feed characters) which the email servers at vtext.com don't support.`

### Steps to Reproduce (including precondition)

Upgraded to 1.18.11 and onstart the error was generated, it worked yesterday when my weekly test ran.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submitted
- Genmon Version: 1.18.11

jgyates, 2022-04-20:
do you know the contents of the email that was sent? I do not see any errors in the log files. The email may help.

skipfire, 2022-04-20:
It would have been the standard 3 messages (I got 3 failures) that happen on start, one of them would be the on-ready. I have one I think is from a previous upgrade cycle.

```
(Generator Notice at Home) Time: 07:00:39 PM
Date: Mon 04-Apr-2022
Generator Ready [Attachment(s) removed]
```

Reading the email there is one additional bit that I saw that may help.

```
Reported error:	550 5.6.11 SMTPSEND.BareLinefeedsAreIllegal; message contains bare linefeeds, which cannot be sent via DATA and receiving system does not support BDAT 
```

jgyates, 2022-04-20:
I checked in an update that *might* fix the issue. Let me know how it goes.

Also, do you have outbound email working. Normally when genmon starts it sends 2 emails. Do you get those OK?

skipfire, 2022-04-20:
I do have the emails in my sent-mail, but it doesn't help.  I pasted one below, also tried forwarding it to my SMS vtext email address and it succeeded, but I just restarted genmon again and messages failed again.

```

Time: 07:51:20 PM
Date: Tue 19-Apr-2022
Generator Ready
```

skipfire, 2022-04-20:
I do have outbound email working and I am getting those messages.

skipfire, 2022-04-20:
It looks like it is a LF without a CR per a MS article. https://docs.microsoft.com/en-us/exchange/mail-flow-best-practices/non-delivery-reports-in-exchange-online/fix-error-code-550-5-6-11-in-exchange-online

I will browse the code some tomorrow and see if I can find anything.

jgyates, 2022-04-20:
yeah, that was my last checkin, not sure where another one is coming from. Just to make sure look at line 465 in /genmon/genmonlib/mymail.py

It should look like this:

      body = '\r\n' + 'Time: ' + tmstamp + '\r\n' + 'Date: ' + dtstamp + '\r\n' + msgstr

no this:

     body = '\n' + 'Time: ' + tmstamp + '\n' + 'Date: ' + dtstamp + '\n' + msgstr

jgyates, 2022-04-20:
I checked in one more update. Let me know how it goes.

skipfire, 2022-04-20:
That last update fixed it, thanks!

closed 2022-04-20 by skipfire