jgyates/genmon #704: Can't connect to Genmon after upgrade from 17.04 to 18.08 (similar to #504, #512)

bmg01907, 2022-05-02:
I was running v17.04 on a RPi 4B without any problems.  I just upgraded to v18.08 via the web interface.  After the upgrade finished I tried to access the Genmon web page (at 192.168.1.nnn:8000) and it just hung with the "Generator Monitor Loading" and rotating circle.  I rebooted the RPi and now Chrome reports the that site "refused to connect."  Firefox won't reach the site as well.  I've flushed the DNS Resolver cache and rebooted the router.  Neither of these actions had any effect.

I'd like to downgrade the Genmon software back to an older version of the software (16 or 17) and see if that solves the problem.  How can I do that?

Any other suggestions as to how to proceed?

(This problem seems similar to what's described in closed issues #504 and #512.  The fixes detailed in those issues didn't work for me.)

jgyates, 2022-05-02:
Since it was a while since you updated the upgrade process would likely take longer. The firs thing I would suggest is loging into your pi via ssh and type

cd genmon
 ./startgenmon.sh restart
The most note worthy change since V17 is the project now uses python 3 instead of python 2 since python 2 is deprecated and not included the raspbian anymore.

Also what is the output of this command on the pi console:

cat /etc/os-release 
This will let us know if you are able to install python3 or if your version of raspbian is too out of date. If it is then you can run genmon with python2 but upgrades are not recommended until you upgrade the pi operating system version

bmg01907, 2022-05-02:
Thanks for your quick reply.  I've attached screenshots of the responses to the "./startgenmon.sh restart" command and the "cat /etc/os-release" command.  Does it appear to you that I need to update the RPi OS?

[image]
[image]
.

jgyates, 2022-05-02:
That is good info. Now run this form the console:

      sudo apt-get --allow-releaseinfo-change update
      sudo apt-get install python3-pip

This should install the python3 version of pip. Once this is complete then run this again:

     ./startgenmon.sh

Let me know if you have any issues.

bmg01907, 2022-05-02:
Thanks.  I did as you instructed.  The RPi reported several errors (see the attached screenshots).  But once Genmon started I could connect via the web page from Chrome.

I'm planning to reconnect the RPi to the generator tomorrow.  At that point I expect to see the software behaving normally.

Do you believe that I should update the OS on the RPi?

Thanks very much for all of your help.

[image]
[image]

jgyates, 2022-05-02:
It looks like your pi is having problems reaching the raspberry pi foundations servers for installing software. Try this:

     sudo apt-get update
     sudo apt-get upgrade

Let me know if you have any errors.

jgyates, 2022-05-02:
if this fails, you may want to check that your name resolution is working on the pi, what is the output of this command

    cat /etc/resolv.conf

Also, can you ping a fully qualified domain name?

    ping www.google.com

Are you running any thing on your network like pi-hole or any other DNS related filtering software?

bmg01907, 2022-05-02:
I think it's working.  No errors reported from the "sudo apt-get update" command.  Only one error reported from the "sudo apt-get upgrade" command relating to being unable to delete a directory that wasn't empty.

I rebooted the RPi and took the attached screenshot of the response to the "cat /etc/resolv.conf" command and the ping of Google.

I'm able to access the RPi web page (192.168.1.nnn:8000) using Chrome.

[image]

bmg01907, 2022-05-02:
And I'm not running pi-hole or any other DNS related filtering software.

jgyates, 2022-05-02:
Can you run this

     sudo apt-get install python3-pip

Then

    ./startgenmon.sh restart

bmg01907, 2022-05-02:
Done.  Screenshot attached.  Also attached is the Genmon-About page.  It shows version 1.18.11.

It appears that I'll need to reenter all of the personalization settings that I had set in version 17.04.

[image]
[image]

jgyates, 2022-05-02:
FYI, there are thee main settings files:

/etc/genmon/genmon.conf - main settings for genmon and genserv (web server)
/etc/genmon/mymail.conf - settings for sending mail 
/etc/genmon/genloader.conf - determines which addons to load

then each add on has it's own conf file in the same folder. you can also backup all your conf files from the about page so if you ever need to restore you can.

jgyates, 2022-05-02:
I am going to close this now but feel free to post to this thread with follow up data or questions.

closed 2022-05-02 by jgyates

bmg01907, 2022-05-02:
Thanks for your help.

I do have the backup files from v1.17.04 (genmon_backup.tar.gz.tar).  What's the best way to restore them to the v1.18.11 running version?

jgyates, 2022-05-02:
You can find that info here:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#restoring-settings-from-a-backup

bmg01907, 2022-05-02:
Thanks.  What's the password for the pi user?  raspberry isn't working.

bmg01907, 2022-05-02:
I'm using WinSCP.

jgyates, 2022-05-02:
In your earlier screenshots you logged in as the pi user, use the same password you used there. If your pi has a different password than the default I do not know what it would be. 

The genmon software does not modify any system passwords. If you purchased your pi form a 3rd party that changed the default password then you may want to consult any documentation you received with that.

jgyates, 2022-05-02:
also, you can look on your pi in the /home/pi/genmon folder and the file genmon_backup.tar.gz is likely already present so if you are trying to transfer that zip file to your pi you might can skip that step if the file is still on your pi.

bmg01907, 2022-05-02:
Thanks for your help.  I reset the pi user password and was able to copy the backup file to the RPi using WinSCP and restore the configuration per your instructions in Appendix M.

bmg01907, 2022-05-03:
Genmon (v1.18.11) is working correctly.  No issues with accessing it using Chrome.  Thanks again.