jgyates/genmon #707: QUESTION: Please explain how to use a VPN to security access Genmon using VNC and the WEB interface.

yguy75, 2022-05-05:
I have read all the issues relating to VPN, but may have missed something.
https://github.com/jgyates/genmon/issues?q=is%3Aissue+vpn

I am new to networking and Linux, but like to dabble and learn new technology. I’m an old retired guy helping two other old retired guys run Genmon on their Generators. They are concerned about security

Currently, Genmon is connected directly to the internet thru the ISP modem/router with unique userid and passwords using SSL. 
We can remotely access the Pi using port forwarding to access VNC and the WEB interface on Genmon.
The ISP modem does not support VPN.

**THINGS TRIED:.**
**1. Installed PiVPN** connected directly to ISP modem on “same” subnet.
**Worked well**. Could securely and remotely connect using VNC and SSH. 
Used port forwarding for ports 5900 and 22. 
Could not connect to Genmon using a WEB browser from remote location, but could see the WEB interface using VNC.
 Port forwarded port 443, but did not seem to work.

**2. Installed Asus built-in OpenVPN server** on Asus AC68U router connected to ISP modem on a “different” subnet. 
Worked Well Was able to securely and remotely connect using OpenVPN client into VNC and SSH. 
Used port forwarding for port 5900 and 22 from the ISP modem to the Asus AC68U to Genmon on the Raspberry Pi.
Could not connect to Genmon using a WEB browser from remote location, but could see the WEB interface using VNC. 
Port forwarded port 443, but did not seem to work.

This has been fun to learn and I feel that I am getting closer but also getting confused. I must be missing something.

**Desired Capability Requested.**
To be able to access RPi and Genmon  securely and remotely primarily through the  WEB browser without having to open any ports, if possible.
We have access to NordVPN, PrivateVPN and SurfShark, if needed.

Do you have any recommendations and tips to help get me in the right direction?
Thank you very much.

jgyates, 2022-05-05:
The Asus router is going in the right direction. I would connect the pi to the wireless or wired connection of the asus router. Then you can use OpenVPN to connect to the subnet behind the asus router. If the pi is on the asus subnet then you should just be able to enter the same address you use on your local network (e.g. 192.168.1.20:8000) to access genmon in a browser. 

As long as your pi is connected directly to your ISP modem and not the Asus router you would have to use VNC to access the pi.

I hope this helps, let me know if you have additional questions.

jgyates, 2022-05-06:
Also, the above should work but another recommendation is that your ISP modem likely has a mode that will assign your internet facing IP address to your Asus router, this mode is typically called a pass through mode. Currently, it sounds like you has a subnet off the router and another off the modem. If the modem were in pass though mode there would only be on subnet off the router. This is more ideal as you are avoiding what is called a "double NAT". This page has more info on double NAT.

https://kb.netgear.com/30186/What-is-double-NAT-and-why-is-it-bad

To move your pi to the asus:
If your pi is connected via ethernet and setup for DHCP then just moving the ethernet cable to the router will do the trick.
If you pi is connected via wifi to your modem then you need to setup a new WiFi access point name on the asus router and change the wifi credentials on the pi. Again this assumes you have a DHCP address that would be assigned to the pi by the asus router one connected to the asus wifi.

If you have a static IP address then you would need to change the IP to be in the subnet IP range of the Asus router.

It may be easier to make these changes and test them with the pi connected to ethernet.

yguy75, 2022-05-06:
Thank you for getting back to me so quickly.
I'm in the process of reconnecting the Pi to the Asus router.
My Pi is not connected to a generator and to test WEB access, I turn on genmon on and  then turn it off when done testing. Otherwise Genmon fills up the modbus log files with errors.
**Question**. Is there a way to have Genmon running but not generate errors to the SD card.?

My ISP modem can be setup into Bridge Mode which then passes the external internet WAN address through to the Asus Router. When in Bridge Mode, the modem turns off all functionality and only has one ethernet output to Asus router. The Asus router then takes over DHCP function for all connected devices. I've installed Merlin on another Asus router and plan to test this. Merlin can selectively assign the VPN server to different devices. That way I **believe** I can have some devices go directly to the internet and other ones go thru a VPN. I'm not too familiar with Merlin. More fun and learning.

I don't have a data plan on a cell phone, so I have to go to a friends place to test remote OpenVPN connection. I try the best I can on my network by connecting to a different router/subnet,

My Pi is connected to Wi-Fi and has it's new IP address. such as 192.168.3.xxx. The modem is 192.168.0.1.
Unfortunately I assigned a static IP to the Pi while using the Pi connected to the modem IP address such as 192.168.0.33. 
I will have to reset the Ethernet on the Pi to remove the static IP.

I am able to connect to the Pi remotely by port forwarding from the modem to Asus router to Pi using the WAN address supplied by the modem, such as 192.168.0.yyy. This works as I can remotely connect to the PI using VNC.

I believe I am connected to OpenVPN to the subnet behind the Asus router. 
But I am confused by what you said to "enter the same address you use on your local network (e.g. 192.168.1.20:8000) to access genmon in a browser."  I'll need to do more research.

It's strange that I can remotely connect to VNC thru both PiVPN and the Asus router running OpenVPN, but not to the WEB interface. 
I guess I will have to sleep on this and start again tomorrow.
Thanks again for your advice.

jgyates, 2022-05-06:
Genmon will always utilize log files however you can setup your pi so log files will be written to a ram disk and only written periodicals to the SD card. This will save the life of your SD card by reducing the number of writes. See this entry for details:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#extending-the-life-of-your-sd-card

Yes, bridge mode is what you want.

I am guessing it is not working since you are using SSL (https). SSL has issues with double NAT.

yguy75, 2022-05-06:
**SUCCESS:**
1. Reinstalled Genmon on a new SD card. 
2. Connected Pi to Asus router one running OpenVPN Server
3. Regenerated ovpn file.
4. Modified ovpn file to add external WAN address and Port number of ISP Modem.
5. Reconfigured port forwarding on main ISP modem and Asus router two
6. Connected iPad to Asus Router two (Merlin) running VPN Client to UK
6. Connected iPad to new OpenVPN file.
7. Ran TraceRoute and WHOIS to very remote connection was thru UK.
8. Was able to remotely connect to VNC.
9. And also connected to Genmon WEB interface using HTTPS and  local IP address.. **(yippee)**

HTTPS  successfully connected to PI with double NAT thru OpenVPN and Router two connected VPN client.

### NOTE:
Could NOT run OpenVPN Server and VPN Client on Asus Router one.
Could only run OpenVPN Server by itself.
This router is needed for VPN client.

**CONCLUSION:**
This approach is too complex to implement and looks like it needs port forwarding.

**NEXT:**
1. Install ram disk on Pi

2. Will install and test OpenVPN server onto the Pi…
This may be the best solution as the Pi could be connected to the ISP Modem
Probably should have done this first, but it would mean installing and maintaining addition software on the Pi.

3. A last resort will be to try Bridge Mode.

Thanks you very much for your fast response and advice.
That's all for now.

zekyl314, 2022-05-06:
I was going to chime in here too.  Another few options could be the following.  I had planned on trying to write up this for Cloudflare Access and posting it to @jgyates, but I've not had the time yet.

https://tailscale.com/

or

https://www.cloudflare.com/products/zero-trust/access/

jgyates, 2022-05-06:
I am going to close this issue for now but feel free to continue to post to this thread with updates or questions.

closed 2022-05-06 by jgyates

zekyl314, 2022-05-10:
You're welcome @yguy75.  I've not deep dove into the Tailscale yet, but do use Cloudflare Access at work.  Tailscale looked interesting for sure.

bbrown-sc, 2022-05-14:
Tailscale seems to work and it or similar offerings, (e.g. Zerotier) are require for Starlink service. Starlink uses CGNAT and port forwarding and VPN routers don't work.

yguy75, 2022-05-15:
Thank you @bbrown-sc 

We are using TailScale which was very easy to install. 
I was able to connect to Genmon using SSH, VNC and the WEB access with no ports open.

PiVPN also worked on the same subnet as the ISP Modem but required port forwarding.. 

An Asus Router using VPN Server also worked when connected on different subnet then the ISP Modem but required double NAT and double port forwarding. This solution took too much effort.

TailScale was the best solution. I didn’t investigate other similar products.

lmamakos, 2022-06-27:
Just to chime in as another tailscale happy user.  Until recently, I've been using OpenVPN to the router at my home for remote access needs from random, unsecure locations.  That's no longer going to be a tenable solution for me as I've added in Starlink as a redundant and diverse ISP access, and there's no port forwarding available with their CGNAT-based network.  

The tailscale solution works in that CGNAT scenario, and I've been slowly adding it to various hosts I want to access remotely.  It works great, and was easy to install.  Their "free" offer works for up to 20 hosts in their "tailnet."   And even that's not a significant limitation since you can have on or more of those hosts act as a "subnet router" to reach other hosts on an attached LAN.

Even upgrading beyond that is pretty cost effective.   I think this might be a great solution for a managed environment of genmon installations since you can put access control lists in place to limit which nodes can talk to which others.  So you could prevent one customer from talking to another over tailscale, but both connecting back to some management host.

If you have a remote access problem, check it out.  It's nice also that it provides a stable (private) address that persists when one or both of the hosts involved switch from, e.g., the cellular WAN network to some Wi-Fi.  Or from one ISP to another.