jgyates/genmon #708: Email Notification Error - please run connect() first

bmg01907, 2022-05-08:
I recently upgraded to v1.18.11 with your help.  Everything seems to be running fine except that email notifications aren't working.  When I select the "Test Email Settings" button I receive an error that says, "please run connect() first" as shown in the screenshot below.

What's the fix for this?
[image]

jgyates, 2022-05-08:
I would need more info to give you anything other than a guess. 

1) Make sure you are either using "less secure apps" (which is going away with gmail at the end of the month) or for a longer term solution, use two factor authentication and "app specific passwords" (https://github.com/jgyates/genmon/issues/682)
2) look at /var/log/mymail.log and see if there are are errors and post them here. 
3) also, without knowing your settings like  SMTP Server Port, Use SSL Encryption, Disable TLS Encryption, Disable SMTP Auth it is difficult to say. Also know if this is gmail account or some other type would be helpful.
4) Are there any unusual non ASCII characters in your server address, email address or password (e.g. unicode characters)

bmg01907, 2022-05-08:
Thanks for your reply.  My answers:

1. I'm not using Gmail.  I'm sending email through my own domain name that's managed by Network Solutions.
2. I've attached the mymail.log file.
3. The attached screenshot shows the email settings in Genmon.  They're identical to what worked in the prior version of Genmon that I was running without any issues.  They're consistent to the settings that work in my email client (Outlook).
4. There are no unusual non ASCII characters in use.

[image]
[mymail.log](https://github.com/jgyates/genmon/files/8647677/mymail.log)

bmg01907, 2022-05-08:
Thanks for your reply.  My answers:

1. I'm not using Gmail.  I'm sending email through my domain name that Network Solutions manages.
2. I've attached the log file.
3. The attached screenshot shows the email settings.  They're identical to what I used in the prior version of Genmon that worked without any issues.  They're similar to the settings in my email client (Outlook).
4. There are no unusual non ASCII characters in use.

[mymail.log](https://github.com/jgyates/genmon/files/8647683/mymail.log)
[image]

jgyates, 2022-05-08:
The first error in your log file is this:

Error SMTP Init : SSL:<False>: [Errno -3] Temporary failure in name resolution : mymail.py:501

this occurs when the software is trying to connect to your SMTP server. Can you ping the server you specify in the SMTP Server setting form the pi console?

jgyates, 2022-05-08:
FYI the second error in your log file is:

Error SMTP sendmail: (451, b'See http://pobox.com/~djb/docs/smtplf.html.') : mymail.py:527

The "451" above denotes " Temporary local problem - please try later"

You can google "error 451 email" for details. Can you send email with another client using this account?

bmg01907, 2022-05-08:
I pinged the SMTP server from the console.  There doesn't appear to be any issue; see the attached screenshot.

I sent an email using Outlook from the "notifications" account without any problem.  It was received intact.

[image]

jgyates, 2022-05-08:
what is the output of this command:

     cat /etc/os-release

If you still have the operating system from your previous install then try this to give us an additional data point:

      cd genmon
       ./startgenmon.sh -p 2 restart

This should restart genmon forcing python 2 instead of python 3. Then see if your email is working. If this works then it will give more info as to the source of your issue. Your email settings are not really that common (no SSL, no TLS) so it is possible that this has not been tested with python3.

bmg01907, 2022-05-08:
Success.  Email is now working with Python 2.  See the attached screenshot.

Is there a fix for this so I can use Python 3?

[image]

jgyates, 2022-05-08:
No, this is an issue I have not seen before. I have an idea of how to fix it but implementing is tricky as I am concerned that this would break email for others. I will need to think about this and possibly get you to run some tests. 

Is this an email server that you maintain? If so can you provide a test email account that I can test with until I get a fix? If not then I would need you to test any possible fix.

jgyates, 2022-05-08:
I checked in a version that may fix this issue. First, restart using python 3

    cd genmon
    ./startgenmon restart

Then go to the About page in the web interface and update the software. When the software restarts after the update then let me know if this corrects the issue.

jgyates, 2022-05-08:
Also, the version number has not been updated.

bmg01907, 2022-05-08:
Done.  I restarted using Python 3 (see attached screenshot) and updated Genmon from the About page.  I then selected "Test Email Settings" and received the test email as well as the Genmon startup email.  It appears that you fixed it.  Much appreciated.

FYI - I can enable SSL and TLS if necessary.

Thanks for your help.  Let me know if you need me to do any testing for you.
[image]

jgyates, 2022-05-08:
If you could test with SSL and TLS, that would be helpful. Also send your logs on the about page to make sure everything is running as expected. Thanks

bmg01907, 2022-05-08:
Email stopped after turning on SSL and TLS.  Test Email Settings generates an error.  See the attached screenshots.

Logs sent from the About page.

[image]
[image]

jgyates, 2022-05-08:
Based on this I would say that your issue is that either your mail server is misconfigured (less likely) or your ssl libaries on your pi are messed up. What is the output of this command:

       pip3 freeze

and this command

       pip freeze

bmg01907, 2022-05-08:
pip3 and pip screenshots attached.

I don't manage the email server.  Network Solutions handles that.  Accordingly, I assume that it's configured correctly.

FYI - I turned off the SSL and TLS settings in Genmon.  The emails are working again.

[image]
[image]

jgyates, 2022-05-08:
Let's try updating your pip libraries that genmon uses. Type this:

    cd genmon
    ./genmonmain.sh -r

Have you installed any other software on the pi?

bmg01907, 2022-05-09:
I haven't installed any other software on the RPi.

Your command above generates an error; see the attached screenshot.

[image]

jgyates, 2022-05-09:
sorry for the typo:

       cd genmon
       ./genmonmaint.sh -r

bmg01907, 2022-05-09:
Done.  Three screenshots attached.

[image]
[image]
[image]

jgyates, 2022-05-09:
it is strange that you are getting the name resolution error like the one in your mymail.log file.

Did this allow you to send email with SSL or TLS?

If it did not then I would suggest a few other things:

1) Try upgrading your pi to Bullseye from Buster. You are currently running buster but bullseye is the latest. Instructions are here:

https://www.tomshardware.com/how-to/upgrade-raspberry-pi-os-to-bullseye-from-buster

2) reinstall a fresh copy of the operating system and reinstall genmon. Use this guide to install genmon:

    https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

you can restore your settings from your backup files.

I have seen this happen on rare occasions in the past (encryption libraries have a strange dependency problem that makes an SSL version error). Unfortunately there are quiet a few python modules that use these (smtplib, flask, basically anything that uses HTTPS or sends mail in python) and tracking this down can be time consuming so the easiest path is to just reinstall the OS.

Once you get everything backup an running test your email with your original settings. I am going to backout the last update as I am not convinced yet that this is the right change to make. If you still have an issue with non SSL and non TLS email I can put it back in.

Let me know how it goes of if you have questions.

bmg01907, 2022-05-09:
Thanks.  Emailing with SSL and TLS on won't work after the pip update; see the attached screenshot.  Emailing works with SSL and TLS off.  I'll leave them off for now.
[image]

I'll try the steps that you described above and report back in the next couple of days.

I appreciate your help.

closed 2022-05-18 by jgyates

bmg01907, 2022-05-19:
Thanks.  I haven't had a chance to upgrade the RPi OS and reinstall Genmon.  I'll post a report here about that as soon as I've done so.