jgyates/genmon #709: Service Journal / Maintenance Log - Suggested Improvements

lwbeam, 2022-05-14:
I recently had an issue with my generator which after some troubleshooting simply vanished. When I went to document this in the Service Journal I noticed a few things that you may want to consider:

1.  The text of the journal entry is currently _centre justified_. This looks fine for short entries, but looks odd for longer one. You might want to consider changing this to _left justified_.
2.   When entering text on the Service Journal page, the program would only save just over 600 characters. My entire entry was 1125 characters. I eventually edited the **maintlog.json** file directly. The whole entry is now displayed (see below).
3.  When entering text on the Service Journal page, you can enter _carriage returns_; however, once you've saved your entry they are not _displayed_. Instead of having several paragraphs, it's displayed as one big one, even though the _carriage returns_ (**\n**) are in the saved text in the **maintlog.json** file.
4.  If you edit an existing entry, the default **Type** (_Repair_) always displayed, so you have to remember to correct it before saving your changed.

Just a few thoughts. Many thanks.

jgyates, 2022-05-14:
These all sound like good suggestion. I will add these to the list. I will update. this thread once complete.

https://github.com/jgyates/genmon/blob/master/feature_requests.md

closed 2022-05-14 by jgyates

jgyates, 2022-05-28:
@lwbeam 

I have checked in the functionality you requested (thanks @buschauer for doing the bulk of the work, he handles the java script that is over my head).

Let me know once you have a chance to test.