jgyates/genmon #710: Smartgen based controller support

Buster14, 2022-05-17:
Generac isnt available in my country,
i've read https://github.com/jgyates/genmon/wiki/Appendix-N-Genmon-Supporting-Other-Controller-Types

So i was offered these combination of controller + wii module by a smartgen agent, would it work? 
Any would i still need the pi ?

http://www.smartgen-america.com/catalog/products/electric-generator-controllers/hgm93xx-series/smartgen-hgm9320mpu-generator-controller-schedule-function-real-time-clock-event-logs-sms-amf/
https://smartgen-america.com/catalog/products/internet-cloud-monitoring-modems/smartgen-cmm366-wifi-internet-access-adaptor/

Alternatively, he also offered LAN based controller DMS710/720 (ATS/AMF), cant find the manual, but it has ethernet port instead.

jgyates, 2022-05-17:
@Buster14 

Based on the link you provided, the following line indicates you can communicate with the controller via the modbus protocol, to this would allow comms over serial.

      RS485 communication port enabling remote control, remote measuring, remote communication via ModBus protocol (RS485 communication port is needed);

You would need to make sure your pi would need a RS-485 serial interface instead of the RS-232 used with most of the controllers currently supported in this project. It also indicates you can communicate via modbus TCP which would allow you to alternatively use genmon without a serial connection to your generator controller but rather over ethernet or wifi.

You would also need to have access to the modbus spec for this controller so you could populate the needed commands in the JSON file. I can provide some guidaance if needed if you have the spec sheet detailing the modbus commands.

You asked if you would still need the pi. You would need to have some type of linux platform that genmon can run on. A pi is preferred but others has used other platforms. Running on a non linux based platform would requires changes to the software before it would operate.

jgyates, 2022-05-18:
I am going to close this thread but feel free to post any follow up questions to this thread.

closed 2022-05-18 by jgyates

Buster14, 2022-05-18:
Thanks for the detailed explanation, 
So from my understanding, wifi / LAN on the controller isn't required, just modbus correct?
The ones i mentioned are pretty costly though, 
there are also other chinese alternatives, they dont have wifi / LAN but i will check if they have modbus or not.

Regarding the PI, i suppose running ubuntu / linux variant on vmware workstation should work as well?

jgyates, 2022-05-18:
genmon can communicate with the controller via the modbus protocol using either via the serial port (built in or USB) or TCP (network). If you use the serial port then the LAN/wifi add on for your controller is not needed.

As long as you can access the serial port from the VM then genmon can communicate with the controller via the serial port. To communicate with the outside world, genmon acts as a web server to serve up pages to a  browser. These are the only things that should be relevant to a VM, other than the linux file system.

ubuntu is known to work as others have done this successfully. I only test with the official Raspberry pi OS (formally known as Raspbian). 

As long as you have python3, a serial port and a network connection and a linux file system it should work.

Buster14, 2022-05-20:
Thanks, i think im going for this

https://www.deepseaelectronics.com/genset/manual-auto-start-control-modules/dse3110

Seems to be the cheapest with ats / amf
the ones with LAN costs 4-5 times

Now for connecting it to my network, Would something like this work?

https://www.tp-link.com/id/home-networking/computer-accessory/tl-wps510u/

it exposes printer through wifi using serial port.

jgyates, 2022-05-20:
@Buster14 

Genmon will only work with a controller that supports modbus. This controllers data sheet does not indicate that it does. the higher end DeepSea controllers do appear to support modbus and I know of one person using a deepsea controller with genmon.

Also, if the controller is not currently support it would need to be added and a modbus registers spec for that controller would be needed (e.g. I don't have that spec for smartgen so it would need to be provided). I have this spec for DeepSea controllers but only the ones that support modbus.

For network connectivity you have two choices:

1) connect the raspberry pi serial port directly to the controller and then connect the pi's network (ethernet or wifi) to your network
2) use a serial to IP converter like this: https://www.pusr.com/products/rs232/rs485-to-wifi-converters-usr-w610.html. Detailed info on serial over IP with genmon is located here: https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

Buster14, 2022-05-21:
Which deepsea controllers specifically?

What about dse7320? (Not the mkII version) this is also available here

https://drive.google.com/file/d/1MhS42Qyz0yN6_Ky4eHyK2-gKdm0UYCrH/view?usp=sharing

but some people seems to have problem finding the ddf file for that controller, would this be an issue?

https://community.exchange.se.com/t5/EcoStruxure-IT-forum/Need-help-in-get-updated-version-modbus-ddf-file-for-genset-DSE/td-p/209271

jgyates, 2022-05-21:
I know it works with 7320MKII but I feel sure that it would likely work with others that also support modbus.

This is the thread that originally kicked off the Deep Sea support:

https://github.com/jgyates/genmon/issues/580

This is the doc that I used, it mentions model numbers:

[ Deep Sea Electronics controller - gencomm.pdf](https://github.com/jgyates/genmon/files/8746973/Deep.Sea.Electronics.controller.-.gencomm.pdf)

Buster14, 2022-05-21:
I see, so is using modbus-based controler the only way to connect a honda genset to home assistant?
I'm planning to use it first at my home if it works properly,
So i'm wondering if there's any alternative inexpensive solution for home usage

Smartgen also has similar variant with modbus support but they're about the same price as deepsea

jgyates, 2022-05-21:
Yes, with this project. This project only uses modbus to get controller data.

Buster14, 2022-05-22:
Hmm the home assistant part that i need is to 
1. monitor + warn generator condition e.g oil, fuel, engine, e.g things that need repairing / refueling.
2. autostart perhaps every 2 weeks so that the battery won't go bad.
3. Monitor load and reduce load if too high
I suppose the second part can be done with IR controller but that's out of topic i guess.
I will explore IR + home assistant if it's not good enough i'll go with DSE controller and continue here. 
thanks for the explanations.