jgyates/genmon #713: gentankdiy.py not starting after a restore

davisgoodman, 2022-05-26:
Had a power outage and shortage of propane so had to shutdown everything.

When restarted the generator, I could no longer access Genmon. 

Did a complete reinstall and restored recent backup files.

Now when trying to start my tank gauges I get this log.

`pi@generac:~/genmon $ sudo /usr/bin/python3 /home/pi/genmon/gentankdiy.py -c /etc/genmon
Traceback (most recent call last):
  File "/home/pi/genmon/gentankdiy.py", line 14, in <module>
    import atexit, getopt, requests
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 118, in <module>
    from . import utils
  File "/usr/lib/python3/dist-packages/requests/utils.py", line 25, in <module>
    from . import certs
  File "/usr/lib/python3/dist-packages/requests/certs.py", line 15, in <module>
    from certifi import where
ImportError: cannot import name 'where' from 'certifi' (unknown location)
`

### Your Environment

- Generator Model: G0070380
- Genmon Logs: Submitted
- Genmon Version: 1.18.11

davisgoodman, 2022-05-26:
Did a bash ./genmonmaint.sh -r -p 3 and was getting an error in one of the required librairies

Requirement already satisfied: pyowm==2.10.0 in /usr/local/lib/python3.9/dist-packages (2.10.0)
Collecting pyowm==2.10.0
  Using cached https://www.piwheels.org/simple/pyowm/pyowm-2.10.0-py3-none-any.whl (3.7 MB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    pyowm==2.10.0 from https://www.piwheels.org/simple/pyowm/pyowm-2.10.0-py3-none-any.whl#sha256=2ac88777565518b9e9aa3fc172c543ada4cbfc667d83775cd17c0e52ea6ccc86:
        Expected sha256 2ac88777565518b9e9aa3fc172c543ada4cbfc667d83775cd17c0e52ea6ccc86
             Got        53751bdb2f936eadf9446364b9ffa3ba32b42e5d15bee8b4165330cdbebafd07

Had todo:
 sudo pip3 install --force-reinstall --no-cache-dir  pyowm==2.10.0 

at which point at the end mentioned:
Installing collected packages: urllib3, idna, charset-normalizer, certifi, requests, geojson, pyowm
  Attempting uninstall: urllib3
    Found existing installation: urllib3 1.26.5
    Not uninstalling urllib3 at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'urllib3'. No files were found to uninstall.
  Attempting uninstall: idna
    Found existing installation: idna 2.10
    Not uninstalling idna at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'idna'. No files were found to uninstall.
  Attempting uninstall: certifi
    Found existing installation: certifi 2020.6.20
    Not uninstalling certifi at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'certifi'. No files were found to uninstall.
  Attempting uninstall: requests
    Found existing installation: requests 2.25.1
    Not uninstalling requests at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'requests'. No files were found to uninstall.
  Attempting uninstall: geojson
    Found existing installation: geojson 2.5.0
    Uninstalling geojson-2.5.0:
      Successfully uninstalled geojson-2.5.0
  Attempting uninstall: pyowm
    Found existing installation: pyowm 2.10.0
    Uninstalling pyowm-2.10.0:
      Successfully uninstalled pyowm-2.10.0
Successfully installed certifi-2022.5.18.1 charset-normalizer-2.0.12 geojson-2.5.0 idna-3.3 pyowm-2.10.0 requests-2.27.1 urllib3-1.26.9

This solved the problem and also the problem I was having getting a lets Encrypt certificate

We can close the case

davisgoodman, 2022-05-26:
Actually something is quite strange but as this is my first refuelling since installing the tank gauges it could normal.

I have 3 tanks which each hold 400 lbs or 363 liters of propane. They just refueled with 901 liters and now all my gauges are having exactly the same readings 297.7 and not moving at all. 

If I look at the value of each register on I2c they all read differently and in the /var/log/gentankdyi.log I get this:

2022-05-26 13:58:20,454 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 2.6, "Percentage2": 11.0, "Percentage3": 11.0}
2022-05-26 13:58:20,455 : OK
2022-05-26 13:58:49,293 : Num Tanks: 3
2022-05-26 13:58:49,294 : Gauge Type: 1
2022-05-26 13:58:49,365 : I2C Init complete: success
2022-05-26 13:58:51,383 : [4, 4]
2022-05-26 13:58:51,384 : Reading Gauge Data: 2.60%
2022-05-26 13:58:52,386 : [16, 251]
2022-05-26 13:58:52,387 : Reading Gauge Data: 10.80%
2022-05-26 13:58:53,389 : [17, 39]
2022-05-26 13:58:53,390 : Reading Gauge Data: 11.00%
2022-05-26 13:58:53,393 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 2.6, "Percentage2": 10.8, "Percentage3": 11.0}
2022-05-26 13:58:53,394 : OK
2022-05-26 14:21:17,551 : Num Tanks: 3
2022-05-26 14:21:17,553 : Gauge Type: 1
2022-05-26 14:21:17,601 : I2C Init complete: success
2022-05-26 14:21:19,612 : [127, 255]
2022-05-26 14:21:19,613 : Reading Gauge Data: 82.00%
2022-05-26 14:21:20,616 : [127, 255]
2022-05-26 14:21:20,616 : Reading Gauge Data: 82.00%
2022-05-26 14:21:21,619 : [127, 255]
2022-05-26 14:21:21,620 : Reading Gauge Data: 82.00%
2022-05-26 14:21:21,635 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 82.0, "Percentage2": 82.0, "Percentage3": 82.0}
2022-05-26 14:21:21,635 : OK
2022-05-26 14:31:22,638 : [127, 255]
2022-05-26 14:31:22,640 : Reading Gauge Data: 82.00%
2022-05-26 14:31:23,643 : [127, 255]
2022-05-26 14:31:23,644 : Reading Gauge Data: 82.00%
2022-05-26 14:31:24,647 : [127, 255]
2022-05-26 14:31:24,648 : Reading Gauge Data: 82.00%
2022-05-26 14:31:24,650 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 82.0, "Percentage2": 82.0, "Percentage3": 82.0}
2022-05-26 14:31:24,650 : OK
2022-05-26 14:32:44,173 : Num Tanks: 3
2022-05-26 14:32:44,175 : Gauge Type: 1
2022-05-26 14:32:44,248 : I2C Init complete: success
2022-05-26 14:32:46,262 : [127, 255]
2022-05-26 14:32:46,263 : Reading Gauge Data: 82.00%
2022-05-26 14:32:47,265 : [127, 255]
2022-05-26 14:32:47,266 : Reading Gauge Data: 82.00%
2022-05-26 14:32:48,269 : [127, 255]
2022-05-26 14:32:48,269 : Reading Gauge Data: 82.00%
2022-05-26 14:32:48,278 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 82.0, "Percentage2": 82.0, "Percentage3": 82.0}
2022-05-26 14:32:48,279 : OK
2022-05-26 14:42:49,282 : [127, 255]
2022-05-26 14:42:49,283 : Reading Gauge Data: 82.00%
2022-05-26 14:42:50,285 : [127, 255]
2022-05-26 14:42:50,286 : Reading Gauge Data: 82.00%
2022-05-26 14:42:51,289 : [127, 255]
2022-05-26 14:42:51,289 : Reading Gauge Data: 82.00%
2022-05-26 14:42:51,291 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 82.0, "Percentage2": 82.0, "Percentage3": 82.0}
2022-05-26 14:42:51,292 : OK
2022-05-26 14:52:48,326 : Num Tanks: 3
2022-05-26 14:52:48,327 : Gauge Type: 1
2022-05-26 14:52:48,392 : I2C Init complete: success
2022-05-26 14:52:50,406 : [127, 255]
2022-05-26 14:52:50,407 : Reading Gauge Data: 82.00%
2022-05-26 14:52:51,409 : [127, 255]
2022-05-26 14:52:51,410 : Reading Gauge Data: 82.00%
2022-05-26 14:52:52,412 : [127, 255]
2022-05-26 14:52:52,413 : Reading Gauge Data: 82.00%
2022-05-26 14:52:52,416 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 82.0, "Percentage2": 82.0, "Percentage3": 82.0}
2022-05-26 14:52:52,417 : OK
2022-05-26 15:02:53,420 : [127, 255]
2022-05-26 15:02:53,420 : Reading Gauge Data: 82.00%
2022-05-26 15:02:54,423 : [127, 255]
2022-05-26 15:02:54,423 : Reading Gauge Data: 82.00%
2022-05-26 15:02:55,426 : [127, 255]
2022-05-26 15:02:55,427 : Reading Gauge Data: 82.00%
2022-05-26 15:02:55,428 : {"Tank Name": "External Tank", "Capacity": 0, "Percentage": 82.0, "Percentage2": 82.0, "Percentage3": 82.0}
2
Seems the readers are at max. Does it make sense?

jgyates, 2022-05-27:
@davisgoodman 

I am going to close this issue as you found a solution to your first questions and your second questions is being addressed [here](https://github.com/jgyates/genmon/issues/548#issuecomment-1139068935).

closed 2022-05-27 by jgyates