jgyates/genmon #714: Generator Always Running - Utility Voltage 125V

StuartRothrock, 2022-05-29:
### Expected Behavior
Generator runs only during an outage.
### Actual Behavior
Due to "Utility Voltage" showing 125V, the generator runs all the time.
### Steps to Reproduce (including precondition)
Unsure
### Screenshot or Pictures relating to the problem (if possible)
    Line :
        Transfer Switch State : Utility
        Utility Voltage : 125 V
        Utility Max Voltage : 0 V
        Utility Min Voltage : 0 V
        Utility Threshold Voltage : 144 V
        Utility Pickup Voltage : 192 V
        Set Output Voltage : 240 V
### Your Environment
- Generator Model: Generac RG032
- Genmon Logs: Submitted Logs and Registers
- Genmon Version: V1.18.11

jgyates, 2022-05-29:
I received your logs. Based on the info I have I would guess that this is an issue with your generator. Has it every run normally in the past or is this a new installation? The Threshold and Pickup voltages are the default values.  There are two lines that are monitored by the controller, each around 120V (+/- ~5%). These two lines add up to 240V for a normal utility voltage.  It looks like the controller is only reporting one of these lines, hence the 125V utility reading. Unless you are comfortable troubleshooting this, I would contact a company that services generac generators. It is possible that a connection is loose or worst case you have a bad controller.

Also, the genmon software does not control any of this, it only reports what is happening as details by the modbus communication link.

StuartRothrock, 2022-05-29:
Thanks for the quick reply and awesome product. It worked until I plugged in the pintsize.me's HAT/RPI4b/POE. I am happy trace down the two utility inputs. Any clue for the Evolution controller? Is there any Evo controller docs on the net to download to help?

jgyates, 2022-05-29:
Do you see the same behavior if you unplug the pintsize.me board? There may be a short on the board or even the connector on the controller may have an issue that only shows up when you try to use the serial port.

StuartRothrock, 2022-05-29:
Same with or without the serial plugged in. Interesting enough, it didn't happen when I first plugged it in. It worked properly for an hour or two. The gen didn't start for an hour or two.

jgyates, 2022-05-29:
one easy thing to try is to power cycle the controller. I have seen that solve strange issues, but it is a long shot:

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

StuartRothrock, 2022-05-29:
I checked N1/N2 (pins 3&6 on J13) for 240v and the 240v was present. I did the power cycle as suggested and it brought up the setup as predicted. The controller still only sees 125V so it still runs. At this point, we are done troubleshooting unless you have any more suggestions. I can't imagine there is a writable register that would mask the utility voltages.

jgyates, 2022-05-29:
At this point you have reached my limit of suggestions, there may be some others who might have some ideas so I will let them chime in.

Nothing modbus related that I am aware of that could cause this.

StuartRothrock, 2022-05-29:
Thanks again for your timely assistance and great product. You can imagine the worry I had not knowing if I would need to replace a $1000 part or not. At least now I know.. 
If anyone knows how to set the " Utility Threshold Voltage" to a voltage from 144V to say 100V, that would be an temporary solution.

jgyates, 2022-05-29:
you can set that via the dealer menu if memory serves

To access Dealer Menu on the Evolution controller, from the top menu selection (SYSTEM, DATE/TIME,BATTERY, SUB-MENUS) enter (for air cooled models) UP UP ESC DOWN UP ESC UP, then go to the dealer menu and press enter. For liquid cooled models a level 2 dealer code can be entered, ESC UP UP DOWN DOWN ESC ESC, then navigate to the dealer menu and press enter. Passcode for Nexus controller is ESC, UP, UP ESC, DOWN, UP, ESC, UP, UP, ENTER."

StuartRothrock, 2022-05-29:
Thanks. I'll try that tonight when I get back home. I'll leave this onen, for a short time, in case anyone else has a suggestion.

liltux, 2022-05-29:
@StuartRothrock the lowest setting for drop out is like upper 140 or lower 150 and Pickup is 180 something.

StuartRothrock, 2022-05-30:
Thanks @liltux - Next option - to use an automation to send start/stop commands...

davisgoodman, 2022-05-31:
@StuartRothrock 

That happened to me once but not sure it would match your situation. There was a  blown fuse in my Smart Transfer box. But would believe it was one of the fuse feeding N1 or N2 to the generator therefore it would only have 125v. Seems you have 240v. 

Just in case it puts you on a track

closed 2022-06-05 by jgyates

davehawthorneGitHub, 2022-06-06:
These guys are really good for helping solve issues with generators. Maybe pose your question there since this is leaning towards a hardware issue. Chris and others are really knowledgeable plus you can get the right parts if you need them. I've used them a few times:
https://gentekpower.com/forums/

StuartRothrock, 2022-06-06:
Will do - thanks @davehawthorneGitHub