jgyates/genmon #715: Genmon won't connect to generator after RPI update

wrvolz, 2022-06-01:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here} Genmon should connect to generator

### Actual Behavior

{Please write here} Genmon won't communication with generator

### Steps to Reproduce (including precondition)

{Please write here} Was updating RPI 3B+ Buster using apt-get dist-upgrade. While doing that I did an update of Genmon but it was the same version -did it anyway. After rebooting, genmon would not connect to generator. As far as I know, it was connecting before the update (was not getting any connection emails from Genmon). Rebooted RPI a couple of times. I looked through the update logs but didn't see anything that might cause this except SSL. It will send me test email but don't know if SSL is used anywhere else. 

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here} Generator 22K G0062500
- Genmon Logs: Submit via the About page submitted
- Genmon Version: {Please write here} 1.18.11

jgyates, 2022-06-01:
I received your logs. Looks like some type of issue with the serial port based on the data so far. What is the output of this command:

      ls -la /dev/ser*

You should see this:

     lrwxrwxrwx 1 root root 7 May 23 15:17 /dev/serial0 -> ttyAMA0
     lrwxrwxrwx 1 root root 5 May 23 15:17 /dev/serial1 -> ttyS0

you could run through the serial troubleshooting steps listed here: 

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Your /var/log/myserial.log has this error many times:

     2022-03-01 18:39:49,207 : Resetting SerialDevice:ReadThread Error: /dev/serial0:read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:129

This indicates that genmon can access the port but something else may be trying to use the port. Was any other software installed while updating? or was anything updated that might use the serial port?

wrvolz, 2022-06-01:
Thanks, I'll keep this for future troubleshooting.

I thought it might be something with the ground wire - I had an issue with that before being pinched in the lid for the enclosure and eventually breaking the wire. I thought I had solved that so took the Raspberry off and checked the continuity of the ground wire and it's ok. When I reconnected the Raspberry it started to work again. Perhaps the RPI needed a power cycle not just a reboot. 

All's ok again and you can close this issue.

Thanks for your help - great product.

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting [https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting]
[43e21afe-d0fe-499a-a006-91a699c54558]

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jgyates, 2022-06-01:
Great! I am glad it is working for you.

closed 2022-06-01 by jgyates

wrvolz, 2022-06-01:
I looked into that issue with myserial.log showing more than one accessing the serial port. I can't find who's accessing /dev/sereial0 or /dev/ttsAMA0. I've tried lsof, ps -lf |grep dev and fuser -k /dev/ttyAMA0 and /dev/serial0 and nothing is returned. It doesn't even find that genmon is accessing that but I suspect it's too quick to see. Any other ideas to find who's accessing that port? Thanks

[20af6198-3baf-42ab-9ad0-5902a145280c]

https://www.avast.com/antivirus

wrvolz, 2022-06-01:
> Try this to validate the Bluetooth is not accessing:
> cd genmon cd OtherApps sudo python3 serialconfig.py -c
> The output should look like this:
> Checking : Is enable UART in boot config: OK Checking : Is BT overlay disabled: OK Checking : Serial console command line removed: OK Checking : Serial console service disabled: OK Checking : BT service disabled: OK Serial port settings are OK. A reboot is needed if changes were made.
> [6fc01154-be88-4286-b26e-18f35d231c9e]

I got this:

 sudo python3 serialconfig.py -c

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

Which I think is the same as you sent so it's not BT.

I'm not sure this is a problem anymore. There were a bunch of those errors on 3/1, one today and none since. Looks like it's a non-issue. The ones I had on 3/1 might be related to a bad ground wire and the one today occurred before I reconnected the ground wire after putting the RPI back in service.

Thanks for you help

https://www.avast.com/antivirus