jgyates/genmon #716: SNMP Incorrect Data Type

jf5264835, 2022-06-02:
All SNMP data fields polled seem to be returned as type string, regardless of the data actually present. This seems to occur in both standard as well as in "Numerics Only" mode.

### Expected Behavior

Numeric data should return as [ASN_INTEGER] type for numeric data such as all controller specific values (Evo 2.0) as well as information from the Genmon stats, specifically from the serial stats and elsewhere where applicable.

### Actual Behavior

All data is returned as [ASN_OCTET_STR] EXCEPT for the extended OID and uptime ticks, which both correctly return values as [ASN_OBJECT_ID] and [ASN_TIMETICKS] respectively. **Attaching SNMP walk dump from Paessler SNMP Tester as txt file**

### Steps to Reproduce (including precondition)

Have genmon running, and enable SNMP support. Then use SNMP walk tool such as Paessler SNMP Tester to view datatypes being presented with values.

### Screenshot or Pictures relating to the problem (if possible)

Attaching txt file with snippet from other SNMP device showing correct datatype being presented.

### Your Environment

- Generator Model: G0070433
- Genmon Logs: Submit via the About page **genmon log complaining about /etc/genmon/kwlog.txt not being a valid file or directory, running config from a different directory, did not update that line to a new location.**
- Genmon Version: v1.18.11 (Updated with SNMP port config support)
[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/8825104/genmon_logs.tar.gz)
[genmon SNMP dump.txt](https://github.com/jgyates/genmon/files/8825160/genmon.dump.txt)
[SNMP snippet.txt](https://github.com/jgyates/genmon/files/8825176/SNMP.snippet.txt)

jgyates, 2022-06-02:
@jf5264835 

The wiki entry for this add on mentions that all OID values specific to genmon are of type string.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gensnmppy-optional

There is a lot of info in the wiki so this can get easily overlooked. The "Numerics Only" option will strip the units off of selected values so they can be parsed easier.

In your attached logs and initial question it look as thought you may have an issue with specifying an alternate location of the config files. How are you specifying the non default location? Typically this is done via the command line of startgenmon.sh like this from the genmon directory:

         ./startgenmon.sh -c /home/username/genmondata restart

This will set the config data path for all loaded modules.

Let me know if you have any questions or feedback.

jgyates, 2022-06-02:
The kwlog file path only needs to be update if the default value in the genmon.conf file is overridden. By default the entry in genmon.conf is commented out. In this instance the specified directory on the command line is used (or /etc/genmon/ is nothing is specified on the command line).

I am going to close this issue but feel free to post to this thread with follow up questions or comments.

closed 2022-06-05 by jgyates

pcbron, 2023-07-07:
I concur with the original request and ask that you consider adding it to your feature request list.  Numeric data should be returned using numeric data types. 

We're using PRTG to monitor Genmon, but PRTG is limited in its ability to parse strings, especially using the "advanced" snmp sensor that can pull up to 10 values at once.  This sensor parses integers correctly but converts floats to bogus values.   "24.25" might become something like "9.15134243".    A workaround is to use the SNMP Custom String sensor and regex, but that uses 1 sensor license per SNMP item.   This is definitely a PRTG limitation/flaw, but we don't normally run into this issue because other devices return the expected data types using SNMP.

elias314, 2023-07-08:
I ran into this too with PRTG.  I was tempted to try to fix the datatypes Genmon outputs, but I found an easier and possibly even better way.  Genmon outputs a number of values via JSON at the URL: `/cmd/status_num_json` and so I use a PRTG "Rest Custom" sensor type with templates to parse these values.  I use 3 sensors: one for "utility", one for "generator", and one for "output", but you could easily combine these into one sensor if you want.  Here's my "output" template:

```
{
    "prtg": {
        "description": {
            "device": "https://genmon",
            "query": "/cmd/status_num_json",
            "comment": "Generator Output"
        },
        "result": [
            {
                "channel": "Frequency",
                "CustomUnit": "hz",
                "value": $["Status"][0]["Engine"][4]["Frequency"]["value"]
            },
            {
                "channel": "Output Voltage",
                "CustomUnit": "V",
                "value": $["Status"][0]["Engine"][5]["Output Voltage"]["value"]
            },
            {
                "channel": "Output Current",
                "CustomUnit": "A",
                "value": $["Status"][0]["Engine"][6]["Output Current"]["value"]
            },
            {
                "channel": "Output Power",
                "CustomUnit": "kW",
                "value": $["Status"][0]["Engine"][7]["Output Power (Single Phase)"]["value"]
            }
        ]
    }
}
```
My "utility" template just grabs the utility voltage, and my "generator" template reads battery voltage and engine RPM.

jgyates, 2023-07-08:
I have added this to the feature list: https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will update this thread once it is complete.

jgyates, 2023-07-12:
I have updated gensnmp.py to support the optional returning of integers. A new option is available on the gensnmp add on called "Force Integers". If the value can be converted to an integer then it will be. Floating point data is still a string. Let me know if you have any issues.