jgyates/genmon #717: Twilio max duration retry stops all future messages, not just the current message

dieseld23, 2022-06-05:
### Expected Behavior

After powerloss, and unable to communicate with internet/twilio, system will give up after 600 seconds (set under twilios settings, tries every 120 seconds).  When the generator exercised a week later, it should have notified via twilio.

### Actual Behavior
A week later, the generator exercised but did not notify, all exercises thereafter were not sent via twilio. Seems that once the timeout occurs, all messages from then on are disabled, not just that specific message. 

### Steps to Reproduce (including precondition)
Have a power loss with no twilio communication and times out. Then later try to exercise it manually and it should send a message but does not. Not until a reboot of the pi will it again start to communicate.

### Your Environment

- Generator Model: Evolution, Air Cooled 20kW
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.07

jgyates, 2022-06-05:
I have not received logs on this issue

jgyates, 2022-06-05:
If your router lost power, but your pi did not, it is possible that you have a network issue between your pi and your router. This can happen for a few reasons. The fact that I did not receive your logs may be an indicator of a network issue.  If you have not rebooted your pi since your router power cycled then your pi may still have this issue today. If this is the case, can you send a test email from the settings page successfully? This will let us know if networking in general is working. It is possible that your pi can talk on your local network, but can not communicate beyond your router. 

With the data I currently have it don't see and issue that can not be explained by a local networking issue.

jgyates, 2022-06-08:
I am going to close this issue but feel free to post to this thread with follow up questions or status.

closed 2022-06-08 by jgyates