jgyates/genmon #720: (non-issue) non-minimized libraries.js

StuartRothrock, 2022-06-08:
Is there a chance of getting the non-minimized versions of the libraries.min.js and libraries.min.css files? Thanks!

closed 2022-06-08 by jgyates

StuartRothrock, 2022-06-11:
Thank you @jgyates - I should have assumed all was included and just look a little further before posting. I am impressed with your coding style and python knowledge. Is your bio posted anywhere? I am curious about your background. I have been in enterprise IT for 30 years and have seen the good and the bad.

jgyates, 2022-06-11:
Hi @StuartRothrock, Thanks for the kind words. No bio, but I started coding around 1990 with x86 assembly, then moved on to C around 1992. I started using python for projects around the house around 2011. Most of my experience is centered around low level interfacing (device drivers, firmware and embedded systems).

StuartRothrock, 2022-06-11:
Hey me too on the x86 assembler - had the ibm pc hardware reference and of
course Peter Norton's book. Back in the ugly TSR days. What was it 0 I may
have forgotten the handiest of all - Bios Int 21h  - Thanks for sharing.

jgyates, 2022-06-11:
Ah, memories...Ray Duncan's Advanced MS-DOS Programming was the book always in arms length back then, i think.

https://www.amazon.com/Advanced-MS-DOS-Programming-Microsoft-Programmers/dp/1556151578